tsduck/tsduck #874: How make CBR stream from VBR stream ?

Uinserte, 2021-09-27:
I have VBR SRT stream, that need convert in UDP (multicast or unicast) CBR stream. How I can do it?

Maybe I'm doing something wrong, but   --target-bitrate   command is not work correctly

Details in attachment along with bitrate graphs

I would be grateful for a hint

[I have VBR stream.pdf](https://github.com/tsduck/tsduck/files/7239682/I.have.VBR.stream.pdf)

lelegard, 2021-09-28:
The plugin `reduce` is probably  reducing to an _average_ of 10 Mb/s but very bursty, according to your diagram. You may try some tuning using options `--time-window` or `--pcr-based` but it depends on the measurement granularity you target.

A common misunderstanding with `tsp` is to consider it as a pure real-time system, synchronised on the wall clock time. In fact, `tsp` is designed to work independently from the wall clock time. Apart from a few plugins such as `regulate`, it completely ignores the wall clock time and gets all synchronization and timing information from the content of the stream (PCR, number of packets according to a supposed bitrate). The purpose is to have an identical result with a real-time stream and an offline processing of a file containing the same stream. The counterpart is that the reaction time cannot be smaller than the interval between two PCR's. It you need a real-time adjustment based on the number of packets actually arriving per millisecond (wall clock time), then TSDuck is probably not the right tool. This is a generic framework for lab experimentation and testing, not an accurate real-time production tool.

Uinserte, 2021-09-28:
Hi,

Nothing helped. Details in the attachment.

[CBR output.pdf](https://github.com/tsduck/tsduck/files/7247382/CBR.output.pdf)