tsduck/tsduck #876: [Q] Merge two VBR files with related PCRs

lyncher0, 2021-09-28:
Hi @lelegard,
I'm trying to merge two files which are related by PCR. The main file (AV) isn't CBR so I'm adding stuffing with --add-input-stuffing and then I'm using merge with --no-pcr-restamp  to merge the second file.

Merge is inserting the packets as soon as there are NULL pkts available without regarding to the stream PCR. This means that if I add-input-stuffing in the rate 1/1 the merged files gets merged into the beginning to the output file but if I reduce that value to 1/60 there's a more even distribution along the file but the streams are not in sync.

I was trying to avoid using regulate (by PCR).... since I'm dealing with files.

What is the recommended strategy to merge to VBR files which streams are related by PCR?
Thank you

lyncher0, 2021-09-29:
Just sharing my current findings:
 - merge is for live, mux is for files (what I'm using to test)
 - tsstuff adds NULL pkts to target file
 - interleave option of file input exists!

Using the above concepts I ended with two different non-working results:

- Interleaving the files to mux using file input option --interleave. Used tsstuff to increase bitrate of all input files to the same value.
For a 10 minute file this results in a playable file but I'm guessing that for a longer file the PTSs of the inserted stream will diverge from PCRs.
[image]

Zoom from start if the resulting file:
[image]

Zoom at 10 minutes:
[image]

- Muxing the second file in AV stream with mux plugin: Only AV Stream had stuffing added with tsstuff. The secondary file had PAT, PMT and NULL removed to avoid mux conflicts. Bitrate of the secondary file was calculated with tsanalyse and used as mux --bitrate option.

After a while PTSs start diverging from PCRs:
[image]

*Conclusions*:
File interleave might no be very useful in this scenario because it doesn't look into contents only to N-packets chunks.

Mux could be a viable solution if insertion was delayed/accelerated until the right PCR appears in the main stream. Right now when a packet is read (even if it's a NULL pkt) it's inserted in the next available NULL pkt of the main stream (if it min_pts, max_pts, inter_time are not specified): https://github.com/tsduck/tsduck/blob/v3.28-2551/src/tsplugins/tsplugin_mux.cpp#L377.
Maybe mux could support a concept like: mux a stream from another file (read PCRs and insert them aligned into current packet flow).

Just my 2 cents.

lars18th, 2021-10-08:
Hi @lyncher0 ,

Here my experience with the muxing functionality of the TSDuck:
- Don't try to mux VBR streams. It's a pain! In fact, a lot of professional devices and tools overcome this. They convert previously any VBR streams to CBR and then they do the mux.
- I'm muxing without troubles using files. The trick is to convert all input streams to CBR with idential bitrate, then "mux" them with interleaving, and finally reduce the output stream to the target bitrate. The result then is nearly perfect after you reprocess the PCR values.

I hope that in the near future @lelegard will want to implement this technique for realtime processing.
Regards.