tsduck/tsduck #877: Run TSP as a service under debian/linux[Q]

kovbal1981, 2021-10-01:
Dear all,

I'm struggling creating a service for tsduck with either supervisord or with systemd. Spawning in background is problematic. Tried with creating a simple shell script that has & at the end, tried without it. 

I'd really love to see a systemd service method to run tsp with various configurations. 

I hope someone can turn on the lights for me :)

Thanks,
Balazs

cbernander, 2021-10-02:
Hi!

With systemd, a [simple service](https://medium.com/the-code-vault/systemd-simple-service-examples-ce586afeed27) (`Type=simple`) should do the trick.

Here is a trivial (and quite useless) implementation:

```
[Service]
Type=simple
ExecStart=/usr/bin/tsp -I null -P reduce -i 1000000 -t 100 -O file /tmp/null.ts
```

When a tsp one-liner won't suffice, create a shell script and reference that with the `ExecStart` directive, e.g:

```
[Service]
Type=simple
ExecStart=/usr/local/bin/my-tsp-script.sh
```

Hope this helps!

lelegard, 2021-10-06:
>  I think this fail due to the tsp output messages

Could you explain that? Is there anything that tsp could do to help using it as a service?

cbernander, 2021-10-06:
This works fine for me:

```
$ cat /usr/local/bin/duck.sh 
#!/bin/bash

/usr/bin/tsp -I null -P reduce -i 1000000 -t 100 -P bitrate_monitor -p 5 -O drop
```

```
$ cat /etc/systemd/system/duck.service 
[Service]
User=ubuntu
Type=simple
ExecStart=/usr/local/bin/duck.sh
```

```
$ sudo journalctl -u duck
-- Logs begin at Fri 2020-11-20 15:46:44 CET, end at Wed 2021-10-06 14:25:36 CEST. --
okt 06 14:25:19 xps-9380 systemd[1]: Started duck.service.
okt 06 14:25:29 xps-9380 duck.sh[25390]: * bitrate_monitor: 2021/10/06 14:25:29, TS bitrate: 1,871,578 bits/s
okt 06 14:25:34 xps-9380 duck.sh[25390]: * bitrate_monitor: 2021/10/06 14:25:34, TS bitrate: 1,865,862 bits/s
```