tsduck/tsduck #878: [Q] Possible to build tsduck without openjdk dependency?

dbussert, 2021-10-08:
I'm trying to build tsduck on Alpine for `arm/v7` architecture (Raspberry Pi). Alpine does not have an `openjdk11` package for `arm/v7`, probably due to Java 11 dropping 32-bit support.

Is it possible to build tsduck without Java bindings?

lelegard, 2021-10-09:
> Is it possible to build tsduck without Java bindings?

Not easily right now. I will update the makefiles to handle something like `make NOJAVA=1` or to automatically detect the absence of Java or both.

> Alpine does not have an openjdk11 package for arm/v7, probably due to Java 11 dropping 32-bit support.

That is strange. On a 32-bit Raspbian distro, I have this:

~~~
$ uname -a
Linux zozoberry 5.10.52-v7+ #1441 SMP Tue Aug 3 18:10:09 BST 2021 armv7l GNU/Linux
$ java --version
openjdk 11.0.12 2021-07-20
OpenJDK Runtime Environment (build 11.0.12+7-post-Raspbian-2)
OpenJDK Server VM (build 11.0.12+7-post-Raspbian-2, mixed mode)
$ dpkg -l '*jdk*'
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                          Version      Architecture Description
+++-=============================-============-============-===========================================================
ii  default-jdk                   2:1.11-72+b4 armhf        Standard Java or Java compatible Development Kit
ii  default-jdk-headless          2:1.11-72+b4 armhf        Standard Java or Java compatible Development Kit (headless)
un  openjdk-11-demo               <none>       <none>       (no description available)
ii  openjdk-11-jdk:armhf          11.0.12+7-2  armhf        OpenJDK Development Kit (JDK)
ii  openjdk-11-jdk-headless:armhf 11.0.12+7-2  armhf        OpenJDK Development Kit (JDK) (headless)
ii  openjdk-11-jre:armhf          11.0.12+7-2  armhf        OpenJDK Java runtime, using Hotspot JIT
ii  openjdk-11-jre-headless:armhf 11.0.12+7-2  armhf        OpenJDK Java runtime, using Hotspot JIT (headless)
un  openjdk-11-source             <none>       <none>       (no description available)
un  openjdk-6-jre                 <none>       <none>       (no description available)
un  openjdk-6-jre-headless        <none>       <none>       (no description available)
un  openjdk-7-jre                 <none>       <none>       (no description available)
un  openjdk-8-jre                 <none>       <none>       (no description available)
ii  oracle-java8-jdk              8u65         armhf        Java™ Platform, Standard Edition 8 Development Kit
$ 
~~~
My only Alpine system is a 64-bit virtual machine hosted by VMware on a Mac and openjdk packages are available.

Did you try `apk search '*jdk*'`?

dbussert, 2021-10-09:
> Sorry, I was wrong in my previous message. After checking, the absence of Java is already implemented in the makefiles and TSDuck is automatically built without Java support.

I didn't get fair enough to build as the prerequisites failed on openjdk11, so I didn't actually move past that to try and build, I assumed it was a prereq. I will try and see if it will build.