tsduck/tsduck #890: [Q]

EnoSat, 2021-11-03:
Hi,
analyze this stream https://mega.nz/file/YIEWmTKR#U293oku9jhmhfX-hnLkyq3smAV2Tq9G3v9Rmm4GL_D8
verbose output show problem with **TID 114 (0x72)** on PID 16 (0x0010),
[image]
next TID without attention on PID 18 (0x0012)
[image]
Personally, I haven't encountered **Table_ID 0x72** yet, what is it actually used for?

lelegard, 2021-11-03:
Hi @EnoSat 

This is the DVB-defined "Stuffing Table" (ST). See ETSI EN 300 468, section 5.2.8. This is rarely used, typically to overwrite an existing section in some retransmitter. These sections are just stuffing, they carry no particular semantics and shall be ignored.

The "unexpected TID" warning is issued by the TSDuck `psi` plugin because the only expected PSI/SI on PID 16 is a NIT. This is just a warning to draw your attention to this unexpected table.

Your TS must have gone through some retransmitter, transrater or transmux which, for some reason, decided to erase the input NIT and replace it with stuffing sections.

lelegard, 2021-11-04:
As a side note, unlike a few applications, `tsp -vvv` won't be more verbose than `tsp -v`. More verbosity is activated though option `--debug` or `-d` with optional higher levels such as `-d2`, `--debug=10`, etc. The default debug level is 1, the lowest.

closed 2021-11-20 by EnoSat