tsduck/tsduck #893: [FEAT] tspcontrol: reload command & HTTP input GET update

lars18th, 2021-11-10:
#### Is your feature request related to a problem?
Here my use case:

- I want to drive the "pid filtering" of a SAT>IP server using the HTTP protocol. With the RTSP protocol it's easy to change the pid filtering in the server, as you only need to send a new PLAY command with the new pid list. However, using the HTTP this is more complex. With a continous streaming after the initial HTTP header `Content-Type: video/mp2t`, you're receiving all the TS packets. And in orther to change the pid filtering, it will be sufficient to resend the GET command with the same URL ending with the new pid list. Example: `GET http://satip.server/?...&pids=0,16,18,20,100,101,102`. This will work if the server accepts to process the new GET command without interrupting the streaming.

So, what I request is that the TSDuck tsp command supports a "remote control" passing the new URL to the HTTP input plugin for sending a new GET command, without any interruption.

#### Proposed solution:
To implement this the TSDuck tsp command requires to have support in the `tspcontrol` plugin for a new command: `reload`. This will be similar to the `restart` but without reinitializing the plugin. And the plugin `http (input)` will need to implement this new transfer mode that implies to send a new request while receiving the stream, even without any check. Then if the server is compatible and the new URL is valid, the stream continues and the server will change the pid filtering.

#### Alternatives:
I don't have any alternative to this. My interest is to mix the virtual tuner of the TSDuck with a switching matrix that can receive TS feeds from SAT>IP servers, but with _server_ pid filtering support.

#### Additional information:
I feel the request is clear, even perhaps quite creazy. What you think @lelegard ?

lelegard, 2021-11-10:
Resending a URL is restarting a new HTTP session. So, there won't be any noticeable difference with a restart of the input plugin. 

I would recommend a client-side PID filtering. Use an input URL without PID filtering. Add a `filter` plugin immediately after the input plugin. Restarting the `filter` plugin with new arguments using `tspcontrol` will change the PID's without breaking the input flow.

The only drawback is the input TCP bandwidth which carries all PID's. If this matters and if you know in advance the set of all PID's you are interested in, specify that superset in the URL to reduce the overall input bandwidth.

lars18th, 2021-11-11:
Hi @lelegard ,

Thank you for your response. However, I'm aware of all you've pointed. But, please note this:

- Not all SAT>IP servers have support for `pids=all`. This limitation could be created by multiple factors, so for this reason this protocol has support for _server-side pid filtering_.
- Futhermore, my objective is to target on-the-fly tunning. So in this scenario I can't define in advance the used pids.
- Finally, my request about the re-GET support in the HTTP input plugin it's different for a mere URL restarting with a new HTTP session. The session will continue to be the same, and the streaming as well. This is not the standard behaviour of a HTTP 1.0 communication. But some servers have support for it. The difference is that the server continues reading the TCP socket for new GET commands while writing the streaming to it. Then if it receives a new GET it process it only to change some parameter of the streaming, but without sending any response that interfere with the streaming.

I know that this is a very specific use case. However, I feel it's easy to understand. Then I hope you want to provide some support. Or almost provide a solution for a _server-side pid filtering_ support in TSDuck.

Thank you for listening.
Best regards.

lelegard, 2021-11-11:
> I know that this is a very specific use case. However, I feel it's easy to understand. 

TSDuck does not directly implement the HTTP protocol and other protocols which are encoded in a URL. Instead, it uses the "natural system library" for that (WinInet on Windows, libcurl on macOS and Linux).

Resending a new GET on the same connection without waiting for the end of the previous input request is not something natural. An HTTP response starts with a header, followed by the body (here, a stream of TS packets). Starting with HTTP 1.1, a TCP connection can remain alive and send a new GET. However, this works only after the completion of the previous response, when the client knows that the new expected data is an HTTP response header. In your case, you have an endless stream of TS packets from request 1. If you do not want to break the connection and create a new one with a new URL, you need to send the URL asynchronously, while TS packets from the previous request are still flowing. At some point, _asynchronously_ again, this previous flow will stop and a new HTTP response header will arrive in the same connection stream. How can the client detect the transition? Parsing incoming data is not a really deterministic method. I do not know how your server handles this but it does not like standard to me. The WinInet and libcurl libraries do not seem to provide an API to do that.

So, this won't be implemented. If you want to change the server-side PID filtering, you need to restart the `http` input plugin with a new URL.

> Or almost provide a solution for a server-side pid filtering support in TSDuck.

If the PID filtering options are present in the URL, then it this is already supported. The `http` plugin transparently passes the URL to the HTTP library.

lars18th, 2021-11-11:
Hi @lelegard ,

Thank you another time to respond to my request. I know that this HTTP re-GET mode is _not standard_. But some servers support it, mainly because it's simple and trivial to implement. In order to not corrupt the TS stream the server discards the HTTP header response (and someone encapsulates the HTTP response inside a NULL TS packet). I'm sure you can understand that this asynchronous manipulation of the pid filtering is not a problem, as with SAT>IP the streaming (transpor plane) is done in RTP _asynchrounsly of the control plane_ provided by the RTSP commands. So the only difference between the the HTTP control/transport over the RTSP mode is this capacity to control de pid filtering (as any other change in the tunning values implies a retune that will restart the streaming).

Focusing on standard solutions, we have two: **RTSP interlaved** and **HTTP chunked transfer**. The first consists on transmiting the streaming over the same TCP control socket, but the drawback is that it continues to be RTSP and the client requires to manage the server with multiple commands (in contrast with the HTTP syntax that is "one-shot-tuning-mode"). The second solution is the `Transfer-Encoding: chunked` encapsulation (see https://en.wikipedia.org/wiki/Chunked_transfer_encoding ). This mode is standard, and provides the same functionality of the re-GET commands. However it consumes more bandwith and the server/client requires to implement it (that isn't trivial as with the plain re-GET). And fortunately, the `libcurl` supports this mode.

So, if you want to support the "HTTP chunked transfer mode with multiple GETs and without streaming interruption" in the `HTTP (input)` plugin, then perhaps you can do it almost for Linux and macOS with the help of the library.

What you think about this last suggestion?

closed 2026-04-26 by github-actions[bot]