tsduck/tsduck #894: [Q] Generate an endless (live) HLS stream looping over some .ts files

jbglaw, 2021-11-11:
Hi!

I'd like to generate a HLS live stream from a few files. These shall just be looped.

I tried this way:

```
tsswitch -v --infinite -I file 1.ts 2.ts | \
     tsp -v --realtime -I file -  -O hls --duration 10 --live 8 --playlist stream.m3 stream.ts
```
However, `tsswitch` processes its input files as fast as possible, so does the later `tsp` call for writing the output stream. How can I slow down this combination to basically present a live stream from a set of infinitely looped files?

Thanks a lot,
  Jan-Benedict

lelegard, 2021-11-12:
Hi @jbglaw 

To slow down the processing speed to the playout / rendering speed, add a `regulate` plugin. 

To get familiar with the various side effects of transport stream processing, I suggest to read the reference section for `tsp` in chapter 3 of the TSDuck user's manual. The chapter 5 also contains a lot of useful examples.

jbglaw, 2021-11-16:
I think I've got quite some more to read. I can actually easily make sure that all input streams are the same wrt. video/audio codecs, but PIDs are different.

With the `regulate` plugin, the writing speed seems to be just fine. But things break hard when the next input file starts. Maybe I'd just pre-process them all and remap the PIDs to a common scheme?

lelegard, 2021-11-16:
> Maybe I'd just pre-process them all and remap the PIDs to a common scheme?

Probably, yes. There are all necessary plugins to rename TS, services, remap PID's, etc.

If you want fluid rendering, you must also ensure that A/V codecs are identical in all files and files are truncated at intra-images.

lars18th, 2021-11-17:
Hi @jbglaw ,

> If you want fluid rendering, you must also ensure that A/V codecs are identical in all files and files are truncated at intra-images.

This last sentence it's very important. I do something similar in the past, and that's a point regulary missing when creating loops. My suggestion is to use FFmpeg to create the loopable video. You can do it recompressing the audio and video, and configure the output to use "closed GOPs", and obviously to write to a mpegts container file. After that you can use the TSDuck toolkit to remap the pids and use the same in every file, change PMT, etc... to prepare your files to loop them with the tsswitch+tsp.

I hope this helps you.
Regards.