tsduck/tsduck #895: [BUG] zombie process after tsswitch --event-command

ASKazak, 2021-11-12:
#### Bug description:
After work tsswitch --event-command "some_command"
the some_command does not die cleanly and remains \ _ [sh] <defunct>

#### How to reproduce:
tsswitch --infinite --receive-timeout 500 --timed-log --event-command "echo test" \
 -I fork 'tsp --realtime --timed-log -I ip --receive-timeout 1000 233.1.1.1:1234 \
 -I fork 'tsp --realtime --timed-log -I ip --receive-timeout 1000 233.1.1.2:1234 \
 -O drop

#### Expected behavior:
do not spawn zombie processes

#### Errors and logs:

#### Environment:
 - OS: Linux
 - OS version: Ubuntu 20.04
 - TSDuck full version: 3.26-2349
 - Installation type: official binary tsduck_3.26-2349.ubuntu20_amd64.deb

#### Additional information:
This all happens when the incoming streams disappear and the command starts to be executed, and if you do not kill the tsswitch, all the child defunct remains.

lelegard, 2021-11-12:
Hi @ASKazak 

This is a known issue which has been fixed a while ago. Please try the latest version, ie. 3.28, or even the latest nightly builds (non-final 3.29).

ASKazak, 2021-11-12:
Thank you, I didn't find that the problem was fixed earlier. Unfortunately, I could check on the latest version, since we only use the LTS version of Ubuntu, and for it the latest suitable deb package with version 3.26. As soon as I can build the package for Ubuntu 20.04, I'll check to see if the problem still exists.

cbernander, 2021-11-12:
FYI @ASKazak - this is how I build TSDuck deb packages for Ubuntu 20.04 using Docker BuildKit:

Contents of _Dockerfile_:

```
FROM ubuntu:20.04 AS build

ENV DEBIAN_FRONTEND=noninteractive LANG=en_US.UTF-8 LANGUAGE=en_US:en LC_ALL=en_US.UTF-8

COPY tsduck /srv/tsduck

RUN apt-get -y update && \
  apt-get -y upgrade && \
  apt-get --no-install-recommends -y install \
  lsb-release sudo locales tzdata && \
  apt-get -y clean && \
  sed -i '/en_US.UTF-8/s/^# //g' /etc/locale.gen && locale-gen

WORKDIR /srv/tsduck

RUN scripts/install-prerequisites.sh

# no support for Dektec / smartcard
RUN make NOGITHUB=1 NOPCSC=1 NODEKTEC=1 installer

FROM scratch as export
COPY --from=build /srv/tsduck/installers/*.deb ./
```

I do a `git clone` of tsduck in the same directory as the Dockerfile, and then I run the following command to create the deb packages:

```
DOCKER_BUILDKIT=1 BUILDKIT_PROGRESS=plain docker build --target export . --output .
```

lelegard, 2021-11-12:
And if you do not want to mess with containers, on any Ubuntu system:
~~~
git clone https://github.com/tsduck/tsduck.git
cd tsduck
scripts/install-prerequisites.sh
make installer
~~~
All details here: https://tsduck.io/doxy/building.html

closed 2022-09-23 by lelegard