tsduck/tsduck #896: [FEAT] Ability to specify an additional parameter for tsswitch --event-udp

ASKazak, 2021-11-12:
#### Is your feature request related to a problem?
Ability to specify an additional parameter for tsswitch --event-udp

#### Proposed solution:
like this:
tsswitch --event-udp address:port[:"message"]
or 
tsswitch --event-udp address:port "message"

#### Alternatives:

#### Additional information:
This is to work around the problem https://github.com/tsduck/tsduck/issues/895
Because in the case of --event-udp it doesn't spawn zombies

lelegard, 2021-11-12:
> Ability to specify an additional parameter for tsswitch --event-udp

Could be done but the UDP message essentially contains a technical description of the event in JSON format.

> tsswitch --event-message "message" --event-udp address:port
> in the latter version, the message can be used for --event-command too

The `--event-command` options specifies a command. If you need to pass a fixed message to that command, it shall be part of the specified command.

> This is to work around the problem #895
> Because in the case of --event-udp it doesn't spawn zombies

As indicated in #895, this problem has already been fixed. You use an outdated version.

ASKazak, 2021-11-12:
> > Ability to specify an additional parameter for tsswitch --event-udp
> 
> Could be done but the UDP message essentially contains a technical description of the event in JSON format.
Let's say we have a description like this:
`{
 "command": "tsswitch",
 "event": "newinput",
 "new-input": 1,
 "origin": "tsduck",
 "previous-input": 0,
 "timestamp": "2021/03/13 19:33:47.688"
}`
If we have some kind of monitoring system on the port indicated by --event-udp, how will it understand from this message which channel we are talking about?
By this description, you can identify where the alert came from:
`{
 "message": "channel_disney",
 "command": "tsswitch",
 "event": "newinput",
 "new-input": 1,
 "origin": "tsduck",
 "previous-input": 0,
 "timestamp": "2021/03/13 19:33:47.688"
}`
> 
> > tsswitch --event-message "message" --event-udp address:port
> > in the latter version, the message can be used for --event-command too
> 
> The `--event-command` options specifies a command. If you need to pass a fixed message to that command, it shall be part of the specified command.
>
Actually, I did just that, but if you add the --event-message option, you can also add this message to the beginning of the service message.
But this is just a suggestion for convenience.

lelegard, 2021-11-13:
I have added a parameter named `--event-user-data` which does this, passed in event command and UDP JSON structure. Please RTFM for the details. You have to rebuild from the latest state of the repo or, if you use Ubuntu 20 as indicated in #895, you may wait until tomorrow and get the latest [nightly builds here](https://tsduck.io/download/prerelease/) (version 3.29-2631 or higher).

closed 2022-09-23 by lelegard