tsduck/tsduck #897: [FEAT] New filter to inject/replace standard PSI/SI tables based on delivery standard

lars18th, 2021-11-17:
#### Is your feature request related to a problem?
I prefer to describe the _general_ use case:

- You are using some A/V tool that generates a MPEG-TS stream (FFmepg for example, or any hardware encoder). This stream typically doesn't have the full PSI/SI tables to be DVB/ATSC/ISDB/DTMB conformant. But you want to broadcast it (for testing) using a modulator. More or less the pipe is `streamer | tsduck >> DTV modulator`.

Now, you can manually create/define all tables using the TSDuck toolkit and add the corresponding plugins inside the `tsp` chain to add/replace them. But the problem is that this could be quite complex... only for a simple broadcasting test.

So the idea is to have something like this: `source | tsp -P gentabs --standard dvb-t -O Hides -f 474000000`
And then "automatically" all incoming tables from the source are discarded/replaced by the required minimum tables to be conformant with the specific standard selected.

#### Proposed solution:

The proposed solution is to create a new filter `-P gentabs` that it's sufficient smart to read the incoming tables and use the provided information to generate a minimum set of the required tables for the target standard. The main target is a simple broadcast testing, but it could also used to do some transmodulation, even this is not the main objective.

This filter `-P gentabs` could have these options:
- `--standard dvbt|dvbs|dvbc|atsc|isvbt|...`: selects the target standard
- `--sid-mapping auto|100:1|0x101:10,0x201:11...`: you can put here some specific mapping (PMT pid to SID) of source services (mainly the idea is one service, but it could be an MPTS), or use by default and "auto" mapping.
- `--update yes|no`: if some input table is updated (for example the PMT of the service), then update the output tables or not.

Please, take note that this plugin only _generates_ the tables with default values. If you want to tweak something (for example, the broadcast name, the TSID, etc.) then you will need to add the corresponding plugin to change the value. I think this will simplifies this new plugin and it doesn't compromise the KISS model.

#### Alternatives:

The current alternative is to manually write the corresponding XML files for each standard and inject them. However, this is a common task and this functionality will simplifies a lot the work to do fast tests.

#### Additional information:

I hope you want to consider this.
Thank you @lelegard !

lelegard, 2021-11-17:
I understand the general idea but, as usual, the devil is in the details. So let's go into the details.

What is the exact list of tables you would like to generate? Some of them must be in the input stream such as PAT and PMT's. Without them, you have no idea of what the PID's are used for. Without PMT, you do not even know which audio / video codec to use. They must have been set by the encoder.

So, assuming that PAT and PMT's already exist, what would you like to generate? NIT, SDT, CAT? There is only one ("actual" version) per TS and can be easily generated when non existent (option `--create` in plugins `nit` and the like).

If you assume that PAT or PMT do not exist, then there is no generic way to guess the structure. You can add the tables but you need to explicitly specify all the details.

Additionally, why do you separately mention DVB-T, DVB-S, DVB-C? What can we automatically generate in different ways between them? The main differences are the delivery descriptors in the NIT but this is not something that can be automatically generated. You need to provide the details of the modulations.

lars18th, 2021-11-17:
Hi @lelegard ,

I'm satisfied if you understand that the general idea makes sense. Is this so?

Then, let me to discuss the details:

- The input _requires_ to be a valid MPEG-TS stream. So it has almost one PAT, one PMT and one service. That's the minimum to understand the source. And if this is not true, then the user has the tools to add these tables manually with TSDuck, or instruct the source to generate them. So in reference to this point I feel that you agree is not a problem.
- Another point is to distinguish between two grups of the required data: the input _dependent_, and the input _independent_. For example, if you want to broadcast something the TDT table it's independent of the source. But all of the service description (pids, pes types, etc.) is dependent. So the idea is to start with _empty_ independent tables and then as some data comes from the source, then read the data and add the corresponding data to the output tables. As the TSDuck model is oriented to realtime processing, we can't consider a problem that until some starting time the output tables will be filled out entirely.
- You can view this plugin as one shortcut to construct a DTV conformant stream. You don't need to care about the specific values of some data, as just a default value requires to exist. Then the user can using other plugins to change/adapt the values.

An alternative to this could be to provide a new tool that generates empty MPTS streams with a fixed bitstream with all conformant tables. This could be easy to implement using some static XML tables. However, this implies that if you want to "translate" the input then you need to merge/mux it inside the blank MPTS. And that's a lot of effort (in CPU). So I feel it's more easy to provide this _smart_ plugin.

Anyway, perhaps if you feel that this has almost a bit of sense, then you could implement a minimum version. And if the result will be useful, then improve it with more functionality (in fact that's your regular development path, but I want to emphasize here that something easy will be implemented).

I hope that you almost want to think on this idea.
Thank you for provide this amazing toolkit! 😉

lars18th, 2021-11-17:
> Additionally, why do you separately mention DVB-T, DVB-S, DVB-C? What can we automatically generate in different ways between them? The main differences are the delivery descriptors in the NIT but [...]

Why not? See the example: `source | tsp -P gentabs --standard dvb-t -O Hides -f 474000000`
We have sufficient information here to know that the output is DVB-T in the channel 21 UHF, so it's easy to fill the NIT table with the corresponding Terrestrial Delivery System descriptor describing the used modulation.

Remember that the use case is to make more easy the generation of a conformant _broadcast_ for testing.
Then we can do something like this whith a SPTS multicast encoded by some hardware:
```
tsp -I ip 226.2.2.2:1234 -P gentabs --standard dvb-t -P tsrename -o 8101 -t 8120 -P svrename -n "test-ch" -p "testing" -r 4 -O Hides -f 474000000
```

lelegard, 2021-11-24:
> source | tsp -P gentabs --standard dvb-t -O Hides -f 474000000
> We have sufficient information here to know that the output is DVB-T in the channel 21 UHF, so it's easy to fill the NIT table with the corresponding Terrestrial Delivery System descriptor describing the used modulation.

Nope. All plugins are fully independent. The hypothetical `gentabs` plugin does not know that there is an output plugin and enven less knows what are its parameters and how to interpret them.

lars18th, 2021-11-24:
> Nope. All plugins are fully independent. The hypothetical `gentabs` plugin does not know that there is an output plugin and enven less knows what are its parameters and how to interpret them.

Ok. Then almost it can be useful to have something like `-P gentabs --standard dvb-t` that uses some default values for DVB-T. For example, it could use 474MHz, 8MHz, 3/4, 1/32, etc. And the user after this could change the values with other plugins.

My idea is more or less this: create a list of _default_ files XML representing the mandatory tables for each DTV standard. Then the `-P gentabs` plugin could use these default files to inject the tables to the stream. But with sufficient intelligence to adapt some data into the tables to the incoming stream.

I feel almost you think this has sense. I'm wrong?

SonnyWalkman, 2023-10-05:
I'd support @lars18th observation.  

As newcomer, learning TSduck isn't intuitive without hours of reading this forum and the guide 10 times and adapting what works for you particular application.   

I maybe dumber than many using it. However, as a hobbyist, its the ONLY tool other than open caster available open source.

Therefore:-

The `gentabs` "hypothetical" is a great idea to create either DVB-T/T2 /S/S2/C compliant stream as a first run. and have it generate the xml tables with default values and creating a tables directory.  
I've spent months and just beginning to get close to DVB except EPG working using Tsduck.

lars18th, 2023-10-05:
Hi @SonnyWalkman,

> I'd support @lars18th observation.

Thank you.

I'm an engineer, but IMHO the `gentabs` will introduce a new grade of usability. Futhermore, the implementation doesn't requires to be very complex: a simple list of "default" XML files (one for each standard) and the code to call to other plugins with "default" values, all of this will produce one conformant DVB stream with just a few list of commands. Imagine if the "ffmpegs ... -t mpegts" command will require a lot of lines of parameters only to add the PAT and PMT tables. Then nobody would use it.

I hope @lelegard will consider this idea in the future when he has time.
Regards.

SonnyWalkman, 2023-10-05:
Hi @lars18th 

Yes if ffmpeg didn't have PAT, SDT it be useless. I believe in can add NIT however OBS studio is unable to add it via the ffmpeg API. 

It would be so easy to do using template XML for each Table with the defaults.

I'm tackling getting EIT.xml write. I'm working on adding TDSTxml and looking at the DVB spec on what's needed??

Sonny..

lars18th, 2023-10-05:
Hi @SonnyWalkman ,

It's nice if you almost start to provide something. I'll try to help you reviewing your XML templates. I suggest to comment them. I'm sure that other users (and @lelegard ) will help with comments about the standards, as they know them very well.

Regards.

lelegard, 2023-10-05:
To all, please be careful about so-called "default values".

I have seen many times odd systems, almost working but not fully, where people spent days or weeks trying to find the root cause of the intermittent issues, ending up with some deeply buried constant in the signalization (or, worse, in the code). When the guy who set that constant was identified and asked "why this value", the answer was always something like "I copied it from network X", or "from country Y", or "it was the default value in the tool", etc.

Default values are good excuses for not thinking. And the few seconds you initially spare, thinking about the right value to set and setting it, is paid a high price by other people who will have to identify and clean the sh** you created by not thinking in the first place.

Default values are fine for things you do several times a day and you don't want to get bothered with. This is why there is the concept of per-user configuration file in TSDuck to set options for commands.

Configuring a network or a transport stream is not something you do several times a day. I do recommend to think about what you are doing before using "default" values, some of them probably being incorrect for your case.

TSDuck has always been designed for people who care about the details and want control over everything. This is why it is hard to use (is it harder than ffmpeg?). But this is what lab, research and design engineers need. I understand that some people, mainly on the front line of production, would prefer to use higher-level tools where everything is done under the hood, without caring too much about the details. Such tools exist. Maybe not in open source. Maybe with some fee. Maybe "not thinking" is more expensive than "thinking" after all...

lars18th, 2023-10-05:
Hi @lelegard ,

I agree with your comment. However, perhaps you forgot this point:
- To provide a **conformant** DVB Transport Stream, several tables are required to be present. And the info **needs** to include some data based on the used distribution network.

Therefore, my proposal is not focused on which _default_ values the tool provides; but in which data is _required_. So, if one single plugin can _add_ the data, then we can _after_ change to some right values. If the default values could be risky, then this `gentabs` plugin could print a Warning message. And it can list inside the warning message the default values used, with a comment about the implications. For example, the Network ID could have a "non-registrered" identifier by default. And the plugin could say: "this is a testing value, please use a valid value".

More or less this is the way that some tools that generate MPEG-TS streams (i.e. FFmepg) are doing. The tool provides a method to change the default values, but _some_ default values are necessary.

The key point is to have an easy method to add all necessary tables and info. Then we will adjust the values with other filters.

I hope this comment doesn't bother you. 😉 
Regards.

lelegard, 2023-10-05:
> For example, the Network ID could have a "non-registrered" identifier by default. And the plugin could say: "this is a testing value, please use a valid value".

That is one exact example I experienced in the past. And the guy who set the value did not care and left an incorrect network id.

By forcing knowledgeable people to handle the issue, you lower the risks. By letting "less skilled" people handle the configuration, you create problems in the details.

lars18th, 2023-10-05:
> > For example, the Network ID could have a "non-registrered" identifier by default. And the plugin could say: "this is a testing value, please use a valid value".
> 
> That is one exact example I experienced in the past. And the guy who set the value did not care and left an incorrect network id.
> 

That's my feeling! 😆 

> By forcing knowledgeable people to handle the issue, you lower the risks. By letting "less skilled" people handle the configuration, you create problems in the details.

Leaving aside the question of risks.... Don't you find _useful_ the idea of a simple plugin that takes care of _ALL_ tables and data pertinent for a MPTS?