tsduck/tsduck #899: [BUG] Continuity Errors Count skyrocketing when two TSP instances with same multicast input are launched

moving-digital, 2021-11-22:
Hi @lelegard 

I thought I opened a bug ticket mistakenly earlier on today, therefore I closed it. 
But it seems I may actually have found a bug, so I am re-opening one. Please see below the explanations.

#### Bug description:
- When two TSP instances with the same multicast input are launched on the same server, the number of Continuity Errors are skyrocketing. We will call these instances TSP1 and TSP2.
- If only TS1 is launched, not a single CC error is detected.
- Likewise, if only TS2 is launched, not a single CC error is detected either.
- However, as soon as both TSP1 and TSP2 are running at the same time, something I can't explain make the Continuity Errors Count skyrocket like crazy, from the very first second the two TSP process run concurrently.

#### How to reproduce:

Equipment involved to easily reproduce the problem:
Option 1: 
- One MPEG2-TS playout/streamer, aka "sender",  with two different physical Ethernet ports used as TS output. We will call them NIC1 and NIC2.
- One Linux server with two different physical Ethernet ports used as TS input. We will call them NIC1 and NIC2.
Connection is simple: Playout-NIC1 to TSDuck-NIC1, and Playout-NIC2 to TSDuck-NIC2.

Option 2: 
- One MPEG2-TS playout/streamer, aka "sender",  with one physical Ethernet ports used as TS output. We will call it NIC1.
- One Linux server with two different physical Ethernet ports used as TS input. We will call them NIC1 and NIC2.
Connection will involve a switch: Playout-NIC1 to Switch. Switch to TSDuck-NIC1, and Switch to TSDuck-NIC2.

#### Expected behavior:
The number of CC errors must remain = 0, even if two instances of TSP are running at the same time on the same server, with the same multicast input.

#### Errors and logs:
TSP1
~~~
tsp --verbose --bitrate 50000000 --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7 -I ip 225.1.1.1:11111 -l 10.168.1.153 -O ip 239.1.1.1:4001 -l 10.168.1.153 --packet-burst 7 --enforce-burst
~~~
TSP2
~~~
tsp --verbose --bitrate 50000000 --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7 -I ip 225.1.1.1:11111 -l 10.168.1.154 -O ip 239.1.1.1:4002 -l 10.168.1.154 --packet-burst 7 --enforce-burst
~~~

#### Environment:
 - OS: Linux
 - OS version: CentOS 7.9
 - TSDuck full version: 3.29-2627
 - Installation type: Official binary

#### Additional information:

lelegard, 2021-11-22:
> But using Dektec, TSReader and TSDuck with continuity plugin will show a rising CC errors, as follows:

You did not explain what you measured. CC errors on which stream? Using which command on which system on the network?

The way you explain the problem, it looks like the two streams from 239.1.1.1:4001 and 239.1.1.1:4002 are getting mixed while they are in fact two different transport streams. Is it possible that what you read to evaluate CC errors is in fact a mixture of all packets on address 239.1.1.1, ports 4001 and 4002 mixed?

moving-digital, 2021-11-22:
Hi @lelegard 

Sorry for missing important details. I'm measuring the CC errors at the output of each TSP processes:
- CC errors at the output of TSP1: 239.1.1.1:4001
- CC errors at the output of TSP2: 239.1.1.1:4002

> The way you explain the problem, it looks like the two streams from 239.1.1.1:4001 and 239.1.1.1:4002 are getting mixed while they are in fact two different transport streams. 

Indeed, that's also what I believe based on my findings.

> Is it possible that what you read to evaluate CC errors is in fact a mixture of all packets on address 239.1.1.1, ports 4001 and 4002 mixed?

No, or at least, the command I'm using is also taking into account the port, as follows:
TSP1
`tsp --verbose --bitrate 50000000 -I ip 239.1.1.1:4001 -l 10.168.1.153 -P continuity -O drop`

TSP2
`tsp --verbose --bitrate 50000000 -I ip 239.1.1.1:4002 -l 10.168.1.154 -P continuity -O drop`

If TSP1 runs alone, no issue. If TSP2 runs alone, no issue either. But if both runs together, there is a conflict.

Is it possible that something is wrong with the way the <multicast:port> output address is handled by TSP output plugin?

moving-digital, 2021-11-22:
> But you mentioned that you noticed the same CC error when checking with TSReader or Dektec software. Is this really the case?

Yes, it is the case so I am still thinking about a TSDuck Output plugin issue.

>  On the same receiver system? If yes, this must be a system issue rather than a TSDuck issue.

No, on 3 different systems that have nothing to do with the two equipment used for my test:
- Dektec Stream Xpert on a Windows Server
- TS Reader on a Windows 10 Laptop
- TSP plugin with a CentOS 8.4 Server

Would you be able to look into this matter as it is impossible to always maintain a different multicast address as the users are not always communicating together?

lelegard, 2021-11-22:
> Yes, it is the case so I am still thinking about a TSDuck Output plugin issue.

I would be quite surprised if that would be the case. You will need to produce strong evidences to convince me. Since the two output processes use distinct local addresses, you should be able to use Wireshark to check if all output packets on each interface have the right port, meaning 4001 on 10.168.1.153 and 4002 on 10.168.1.154.

That being said, 10.168.1.153 and 10.168.1.154 are on the same subnet. Why did you configure two interfaces of the same system on the same subnet? Mixing that with multicast routing is clearly looking for trouble. One hypothesis is that the Linux system duplicates all packets from the two streams on each interface when the two interfaces are used in multicasting. Since the two interfaces are on the same subnet, the net result is that all packets are duplicated. You should be able to confirm that or not using wireshark.

> Would you be able to look into this matter 

No I can't. This is your network configuration, a bit unusual. You need to do some network investigations on this configuration.

It would be interesting to use only one output interface and see if the problem is the same.

> as it is impossible to always maintain a different multicast address as the users are not always communicating together?

I do not see the relationship between "impossible to always maintain a different multicast address" and "the users are not always communicating together". Usual large telco multicast networks (live TV for instance) use different multicast addresses for different transport streams.

moving-digital, 2021-11-22:
Hi @lelegard 

Thanks, I will check with Wireshark tomorrow.

In the meantime, I have tried with one interface only (Using .153 and stopped .154).
The problem is still happening: no CC errors when only one TSP is running, a lot of CC errors when two TSP are running.

My commands are:

TSP1
`tsp --verbose --bitrate 50000000 --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7 -I ip 225.1.1.1:11111 -l 10.168.1.153 -O ip 239.1.1.1:4001 -l 10.168.1.153 --packet-burst 7 --enforce-burst`

TSP2
`tsp --verbose --bitrate 50000000 --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7 -I ip 225.1.1.1:11111 -l 10.168.1.153 -O ip 239.1.1.1:4002 -l 10.168.1.153 --packet-burst 7 --enforce-burst`

The CC errors check commands are:

TSP1
`tsp --verbose --bitrate 50000000 -I ip 239.1.1.1:4001 -l 10.168.1.153 -P continuity -O drop`

TSP2
`tsp --verbose --bitrate 50000000 -I ip 239.1.1.1:4002 -l 10.168.1.153 -P continuity -O drop`

Output is:

```
* continuity: packet index: 1,253, PID: 0x02D0, missing 12 packets
* continuity: packet index: 1,254, PID: 0x026C, missing 13 packets
* continuity: packet index: 1,281, PID: 0x026C, missing 6 packets
* continuity: packet index: 1,282, PID: 0x02D0, missing 6 packets
* continuity: packet index: 1,288, PID: 0x02D0, missing 7 packets
* continuity: packet index: 1,289, PID: 0x026C, missing 6 packets
* continuity: packet index: 1,295, PID: 0x02D0, missing 6 packets
* continuity: packet index: 1,296, PID: 0x026C, missing 7 packets
* continuity: packet index: 1,302, PID: 0x02D0, missing 6 packets
* continuity: packet index: 1,304, PID: 0x026C, missing 7 packets
* continuity: packet index: 1,316, PID: 0x026C, missing 3 packets
* continuity: packet index: 1,317, PID: 0x02D0, missing 3 packets
* continuity: packet index: 1,323, PID: 0x026C, missing 9 packets
```

To me, there is still a wrong multicast management by TSP which is not able to make the difference between two independent multicast address+ports and outputs both multicast as one, by aggregating the ports linked to the same address.

lars18th, 2021-11-22:
Hi,

My two cents:

> One possibility is a Linux system issue with multicast handling. It is well-known that using the same multicast address / distinct ports for distinct streams is looking for troubles depending on the operating system. Similarly, having the same port / distinct addresses may confuse clients if they need to receive both streams. Since I work with Linux, macOS and Windows interchangeably and all of them have their own specific issues and limitations regarding multicast, I cannot really remember what does not work well where.

From my experience, reading/writing in Linux sharing the same _multicast address_ even with different ports has an unexpected behaviour: the processes receive _all_ packets addressed to this address despite the port used.

Please @moving-digital could you test this?

- From an external device, write a valid MPEG-TS stream to the multicast address 239.1.1.1:4001 
- From a different external device, write a second MPEG-TS to the multicast address  239.1.1.1:4002
- From a third Linux device, read the stream 239.1.1.1:4001 with VLC and/or FFmpeg.
- Do the same with a fourth Linux device with the stream 239.1.1.1:4002.

Until here, if your network is right configured, you will see the streams without any error. Then:

- Check another time in the same Linux device, running two instances of VLC or FFmpeg reading one from 239.1.1.1:4001 and the other from 239.1.1.1:4002. You see errors?

moving-digital, 2021-11-23:
Hi @lelegard  @lars18th 

Thank you both for your comments! So I have tried the below, with a total of 3 equipment:

1. MPEG-TS streamer to the multicast address 239.1.1.1:4001
2. MPEG-TS streamer to the multicast address 239.1.1.1:4002
3. Player on Windows 10 Laptop, using VLC, Dektec StreamXpert and TS Reader

The 3 equipment are connected to the same switch, so are on the same network.

When I run two instances of VLC, Dektec StreamXpert or TS Reader, not a single CC error is detected. Yet, the same multicast address is used. Only the port is different.

So, I'm still confused as the issue (CC Errors skyrocketing) only happens when the same multicast address is used on the very same streaming server.

> From my experience, reading/writing in Linux sharing the same multicast address even with different ports has an unexpected behaviour: the processes receive all packets addressed to this address despite the port used.

This seems to be in-line with what Thierry is sharing too. However in the test I just performed, there is no issue on the receiving end, as the players are able to play each stream properly without being confused by the fact that two different equipment are sending two different MPEG2-TS stream in the same network, with the same address.

moving-digital, 2021-11-23:
Hi

I investigated a bit more, and something is still puzzling me.

My streamer 1 uses the below TSP command:
`tsp --verbose --bitrate 10000000 --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7 -I file file1.ts -i -P regulate -O ip 225.1.1.1:11111 -l 10.168.1.127 --packet-burst 7 --enforce-burst`

My streamer 2 uses the below one:
`tsp --verbose --bitrate 10000000 --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7 -I file file2.ts -i -P regulate -O ip 225.1.1.1:11111 -l 172.168.1.129 --packet-burst 7 --enforce-burst`

In this case, both streamers are using different netmasks:
- Streamer 1: 10.168.1.127
- Streamer 2: 172.168.1.129

Then, I've set up a receiving instance of TSP on one server able to listen to both streams, with the below command:

When listening to 239.1.1.1:4001 on 10.168.1.0 network:
`tsp --verbose --bitrate 30000000 --add-input-stuffing 1/10 --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7 -I ip 225.1.1.1:11111 -l 10.168.1.153 --buffer-size 212992 -P zap 1 -P filter --negate --pid 0x1FFF -O ip 239.1.1.1:4001 -l 10.168.1.153 --packet-burst 7 --enforce-burst`

But I will get warnings and most importantly the zap plugin will never stop toggling between the two streams, meaning the `zap` plugin is not able to correctly listen on one given physical network interface only:

* ip: joining multicast group 225.1.1.1:11111 from local address 10.168.1.153
* ip: initial input bitrate is 33,000,000 b/s
* Warning: ip: detected multiple sources for the same destination 225.1.1.1:11111 with potentially distinct streams
* Warning: ip: detected source: 10.168.1.129:55751
* Warning: ip: detected source: 172.168.1.127:60473
* pmt: found service id 0x0001 (1), PMT PID is 0x1000 (4096)
* pmt: PMT version 0 modified
* spliceinject: found service id 0x0001 (1), PMT PID is 0x1000 (4096)
* zap: found service id 0x0001, PMT PID is 0x1000
* zap: found service id 0x0001, PMT PID is 0x0020
* zap: found service id 0x0001, PMT PID is 0x1000
* zap: found service id 0x0001, PMT PID is 0x0020
* zap: found service id 0x0001, PMT PID is 0x1000
* zap: found service id 0x0001, PMT PID is 0x0020
* zap: found service id 0x0001, PMT PID is 0x1000

lelegard, 2021-11-23:
Windows and Linux work differently with multicast. Could you try to receive with VLC on Linux?

On Linux, when receiving multicast, the local socket is bound to the multicast address, not to the local interface address. The local interface address is just used to send the join multicast group request. This is different on Windows, the socket is bound to the local interface. 

To compare TSDuck vs. another receiver, please run the two applications one after the other on the same system.

moving-digital, 2021-11-24:
Hi @lelegard @lars18th 

You guys just rock ! I've done many experiments today, and I can confirm that the issue was due to the fact that I was trying to read the very same multicast address:port from two independent physical Ethernet ports (in my case 225.1.1.1:11111).

As soon as I changed the port, everything got fixed.

So what can be concluded is that using the very same multicast address on different physical Ethernet ports does not cause any issue on Linux if the port is different. 
However, as soon as both the address and port are identical and used in at least two physical interfaces, it will create a conflict, even if these two interfaces do not even have the same subnet at all.

Once again, thank you so much !!!

lars18th, 2021-11-24:
Hi @moving-digital ,

> So what can be concluded is that using the very same multicast address on different physical Ethernet ports does not cause any issue on Linux if the port is different. However, as soon as both the [...]

That's because the binding is done to an address, and not to an interface. In case you bind to the interface _and_ the multicast address, then perhaps the problem doesn't appears.

I'm glad you've fixed your problem.
Regards.

lelegard, 2021-11-24:
> In case you bind to the interface and the multicast address, 

How would you do that ?

moving-digital, 2021-11-25:
Hi @lars18th 

> That's because the binding is done to an address, and not to an interface. In case you bind to the interface _and_ the multicast address, then perhaps the problem doesn't appears.
> 
> I'm glad you've fixed your problem. Regards.

I wouldn't say "fixed but "found a workaround"

moving-digital, 2021-11-25:
> > In case you bind to the interface and the multicast address,
> 
> How would you do that ?

Hi @lelegard 

I'm sure you already have, but I found these 2 or 3 interesting links on Google :

https://www.reddit.com/r/networking/comments/7nketv/proper_use_of_bind_for_multicast_receive_on_linux/

https://stackoverflow.com/questions/10692956/what-does-it-mean-to-bind-a-multicast-udp-socket/14613620

lelegard, 2021-11-25:
> I'm sure you already have, but I found these two interesting links on Google which should help you find out how to use bind() for our matter

They just repeat what I already mentioned and what is already implemented in TSDuck: on Linux, the socket shall be bound to the multicast address, not the local address. And, if needed, the join group is sent on the local interface.

The first link also mentions:

> Having multiple multicast groups that use the same port is a suboptimal design for the exact situation you've found yourself in: it makes it hard to filter the traffic to just what you want if another application on the same machine is also listening to traffic that shares the same port. It's generally best to avoid sharing either multicast group or port across different logical channels.

Which is what you did and what we advised to not do...

closed 2021-12-09 by moving-digital