tsduck/tsduck #901: [FEAT] Generic binding subset

lars18th, 2021-11-25:
#### Is your feature request related to a problem?
I want to use the TSDuck toolkit from Golang, so I need to call to the library from it.

#### Proposed solution:
This is more or less the request asked at #734 . So let me to explain what I request and what not:
- I don't need a full API of TSDuck.
- I only want to instantiate an AsyncReport and one TSProcessor. More or less was it's done in some examples (i.e. `sample-tsp.py`).
- The current Java and Python binding are well designed, but in order to expand them to _new_ bindings they require a huge effort.
- Based on the idea of a _simple_ API, it can be provided _without_ an object oriented approach. So pure "function calls" with basic fixed types could be used.
- This _simple_ API can be described as a "C" API, and then we can use it from other languages.
- For example with the tool `cgo` we can call to any "C" exported function: https://pkg.go.dev/cmd/cgo

#### Alternatives:
The only alternative is that someone provides a new full binding, like the stalled support for nodejs provided in #875 .

#### Additional information:
Please, don't say (with all due respect) that C is an "old" language. This is not the focus of the discussion. The question is if a "C" API could be used to provide a binding to any language. So the actual question is if a _simple_ API could be provided _without_ using objects. And if this could be, then the logical option is to provide it described in C, only to provide an _universal_ interaction, and not because it will preferable to be in C or to use it with this language.

Please @lelegard , comment about this.
Thank you!

lelegard, 2021-11-25:
This is a valid and everlasting subject of discussion. I have quite a lot of experience in mixing languages, back to the 80's, calling C from Ada, Prolog from Ada, C++ from Java, Pascal from C, even weirdnesses such as calling Pascal from Fortran, etc. The basic line is that this is a mess and you need to be _very_ careful. First, the idea of a _fully generic_ binding is an utopia. Forget about it, full stop. Only ad-hoc or low-level solutions are reliable.

Then, you have two cases:

1. Some language provides a well-defined API, ABI or framework to call or to be called from other languages. This is the case for Java with JNI. This is somewhat the case with Python but in a very awkward way which does no make it independent from the running version of Python (and this is why this method is not used in TSDuck).
2. Use low-level mechanisms from the two languages to converge to a very basic low-level ABI which is deterministically defined. This is what TSDuck uses with Python. The low-level mechanism of C++ is C and the low-level mechanism of Python is ctypes. Note that any other language can plug into that common ABI using its own low-level mechanisms.

In the second case, some adaptation layer is necessary in each language to hide the dirt of the low-level ABI. This is the file `tsduck.py` in the TSDuck Python interface. It exposes Python classes to Python applications and internally calls the low-level ABI.

I do not know Go and I cannot really advise. If there is no JNI-like native interface and if there is some deterministic way to call low-level ABI (using integers and buffer addresses), I would recommend to reuse the `tspy` low-level ABI. You will need to write a Go module (or whatever you call it) to expose Go services (or classes, or functions or whatever) and calls the low-level ABI. If some adaptations or a couple of new functions are necessary (the `tspy` ABI relies on the fact that the calling language can encode/decode UTF-16 strings), they could be integrated in TSDuck.

lars18th, 2021-11-25:
Hi @lelegard ,

Thank you for your experienced and thoughtful response. It is not my objective to maintain an everlasting discussion about a topic. I only want to achieve my objective. So thank you for pointing in the right direction.

From what I understand, then the best solution at time to use TSDuck  from another language is to "mimic" the ABI of the Python binding. So then the idea is more or less this: write a wrapper similar to `tsduck.py` that it has all the calls and callbacks from the external language to the TSDuck libary. That's all?

lelegard, 2021-11-25:
> So then the idea is more or less this: write a wrapper similar to tsduck.py that it has all the calls and callbacks from the external language to the TSDuck libary. That's all?

Basically, yes. Just write the Go wrapper calling the same low-level `tspy` functions as `tsduck.py`.

lars18th, 2021-11-25:
> > So then the idea is more or less this: write a wrapper similar to tsduck.py that it has all the calls and callbacks from the external language to the TSDuck libary. That's all?
> 
> Basically, yes. Just write the Go wrapper calling the same low-level `tspy` functions as `tsduck.py`.

OK. Thank you! 👍

TVforME, 2024-08-05:
Hey @lars18th ,
Did you eventually get cgo bindings for TSduck? I'd be interested to help out.  Using Go and exec and responding to stdio and stderr sucks and a binding be a great way forward for both complexity and performance.  The latter is not much of an issue as TSduck is spectacular fast.

closed 2026-04-26 by github-actions[bot]