tsduck/tsduck #902: [BUG] sed syntax error in creating tsduck.pc pkgconfig file with NOCURL=true

barsnick, 2021-12-07:
#### Bug description:
If I build tsduck with ```NOCURL=true```, the "```make install```" step cannot create the ```tsduck.pc``` pkgconfig file, due to an sed syntax error.

#### How to reproduce:
~~~
make install DESTDIR=/mybuildinstallroot INSTALL="/usr/bin/install -p" install-devel NOTEST=true NODEKTEC=true NOCURL=true NOEDITLINE=true NOPCSC=true NORIST=true NOSRT=true SYSPREFIX=/usr ETCDIR=/etc USRLIBDIR=/usr/lib64 UDEVDIR=/usr/lib/udev/rules.d SYSROOT=/mybuildinstallroot
~~~

#### Expected behavior:
tsduck.pc is created and build succeeds.

#### Errors and logs:
~~~
[...]
install -d -m 755 /mybuildinstallroot/usr/share/pkgconfig
sed -e 's|{{VERSION}}|3.29.2651|g' \
    -e 's|{{PREFIX}}|/usr|g' \
    -e 's|{{LIBDIR}}|/usr/lib64|g' \
    -e 's|srt||g' \
    -e 's|librist||g' \
    -e 's|libpcsclite||g' \
    -e 's|libcurl||d' \
    -e 's|:[ ,]*|: |' -e 's|,[ ,]*|, |' -e 's|[ ,]*$||' \
    tsduck.pc >/mybuildinstallroot/usr/share/pkgconfig/tsduck.pc
sed: -e expression #7, char 12: unknown option to `s'
~~~

#### Environment:
 - OS: Linux
 - OS version: Fedora 35 x86_64
 - TSDuck full version: 3.29-2651
 - Installation type: I'm building from source, obviously

#### Additional information:
Obviously, in ```src/libtsduck/config/Makefile```, the line
~~~
            $(if $(NOCURL),-e 's|libcurl||g',) \
~~~
should be
~~~
            $(if $(NOCURL),-e 's|libcurl||d',) \
~~~

Just BTW: I'm creating a "more modern" RPM SPEC file, and adding SPEC options to disable each feature separately. Partly because I can, partly because librist isn't available on Fedora mainline. (Yes, I'm aware of https://github.com/tsduck/rist-installer .)

lelegard, 2021-12-07:
Fixed in commit a0ad9310bfd346515aef1aef6e96758e3bdf2a5c

> Obviously, in src/libtsduck/config/Makefile, the line
>            $(if $(NOCURL),-e 's|libcurl||g',) \
> should be
>            $(if $(NOCURL),-e 's|libcurl||d',) \

I assume you mean the opposite. It was "d" and it should be "g".

lelegard, 2021-12-07:
> Just BTW: I'm creating a "more modern" RPM SPEC file, and adding SPEC options to disable each feature separately. 

I am curious of what you mean by "more modern spec file".

Here, the update of the .pc file shall be done during `make install` since the installation shall be independent from RPM.

> Partly because I can, partly because librist isn't available on Fedora mainline. 

The absence of librist is already automatically detected, `NORIST` is automatically set and the dependency to librist is automatically removed from the spec file. Indeed, the presence of "Requires: librist" in the spec file is provisional since librist is not present in any Linux distro for now. I do not even know if it will be named like that if/when it is added.

closed 2022-09-23 by lelegard