tsduck/tsduck #903: [BUG] tsduck.pc contains libedit dependency even when building with NOEDITLINE=true

barsnick, 2021-12-07:
#### Bug description:
If I build with ```NOEDITLINE=true```, the libedit dependency is not removed from the ```tsduck.pc``` pkgconfig file.

#### How to reproduce:
~~~
make install DESTDIR=/mybuildinstallroot INSTALL="/usr/bin/install -p" install-devel NOTEST=true NODEKTEC=true NOCURL=true NOEDITLINE=true NOPCSC=true NORIST=true NOSRT=true SYSPREFIX=/usr ETCDIR=/etc USRLIBDIR=/usr/lib64 UDEVDIR=/usr/lib/udev/rules.d SYSROOT=/mybuildinstallroot
~~~

#### Expected behavior:
No ```libedit``` in ```Libs.private```

#### Errors and logs:
```libedit``` is still in there in /mybuildinstallroot/usr/lib64/pkgconfig/tsduck.pc
~~~
prefix=/usr
libdir=/usr/lib64

Name: TSDuck
Description: MPEG Transport Stream Toolkit
URL: https://tsduck.io/
Version: 3.29.2651
Requires:
Requires.private: libedit
Libs: -L${libdir} -ltsduck
Libs.private: -lpthread -lrt -ldl -lm -lstdc++
Cflags: -I${prefix}/include/tsduck --std=c++11
~~~

#### Environment:
 - OS: Linux
 - OS version: Fedora 35 x86_64
 - TSDuck full version: 3.29-2651
 - Installation type: I'm building from source, obviously

#### Additional information:
The Makefile for tsduck.pc requires an sed removal rule such as the other dependencies have.

lelegard, 2021-12-07:
Fixed in commit a0ad9310bfd346515aef1aef6e96758e3bdf2a5c

lelegard, 2021-12-07:
And BTW, building the .rpm and .deb packages with NOEDITLINE=1 was also incorrect. Just fixed it in commit 3105f5736519061dffa08f32b58bb692e0f47ca8. The use of libedit was recently added and I must admit I was not thorough enough with all the implications.

barsnick, 2021-12-07:
Ah, good catch, I was exclusively looking at tsduck.pc.

BTW, the sed line removing librist has no effect, as librist hasn't made it into tsduck.pcm which is probably an incorrect omission.

lelegard, 2021-12-07:
> BTW, the sed line removing librist has no effect, as librist hasn't made it into tsduck.pc, which is probably an incorrect omission.

Ah, ah, good catch, your turn 😄 There is a librist.pc file indeed. Added in  commit beac7078ad4ad0c43b56d0989987df3217339b28. Lots of small and annoying fixes today...

closed 2022-09-23 by lelegard