tsduck/tsduck #906: [BUG] Live HLS output includes X-PLAYLIST-TYPE:EVENT

nhjm449, 2021-12-10:
#### Bug description:
The HLS output plugin is including a `#EXT-X-PLAYLIST-TYPE:EVENT` tag in "live" playlists when it shouldn't. The [spec](https://datatracker.ietf.org/doc/html/draft-pantos-hls-rfc8216bis#section-6.2.2) says
> A Live Playlist MUST NOT contain the EXT-X-PLAYLIST-TYPE tag, as no value of that tag allows Media Segments to be removed.

See https://github.com/tsduck/tsduck/blob/e439a50705550076481581e10e73a120fa080640/src/libtsduck/plugins/plugins/tshlsOutputPlugin.cpp#L216

#### How to reproduce:
```
tsp -I ... -O hls --duration 5 --intra-close --live 10 --max-extra-duration 6 --playlist stream.m3u8 --start-media-sequence 1 stream.m3u8
```

#### Expected behavior:
No `#EXT-X-PLAYLIST-TYPE` tag should appear in the m3u8 output.

#### Errors and logs:
~~~
#EXTM3U
#EXT-X-VERSION:3
#EXT-X-TARGETDURATION:5
#EXT-X-MEDIA-SEQUENCE:1
#EXT-X-PLAYLIST-TYPE:EVENT
#EXTINF:5.279,
#EXT-X-BITRATE:2296
stream_000000.ts
~~~

#### Environment:
 - OS: Linux
 - OS version: Arch
 - TSDuck full version: 3.25-2237
 - Installation type: self-packaged

lelegard, 2021-12-10:
Hi @nhjm449,

Interesting. This was an interpretation error on my side. I incorrectly assumed that "live" and "event" were the same, which is incorrect when carefully reading the spec.

Please correct me if I am wrong: in fact, there are _three_ types of playlists, "vod" (static), "event" (increasing only) and "live" (sliding window, adding new and removing old segments). The confusing part is that the keyword `EXT-X-PLAYLIST-TYPE` accepts only _two_ values, `VOD` and `EVENT`. The third type is indicated by the absence of `EXT-X-PLAYLIST-TYPE`. Quite confusing and a poor specification choice in my opinion...

I will fix this but this is more than just a simple matter of keyword generation. This has implications on the plugin command line (how to specify any of the three types) and the way segments are generated and the playlist updated.

nhjm449, 2021-12-10:
Correct, there are three types, and it was pretty silly of them to not add an explicit, default `LIVE` type.

> 3.29

Wondered if you'd mention that. :grin:

Thanks!

lelegard, 2021-12-10:
> 3.29
> Wondered if you'd mention that. 😁

Sure 😄 That is the usual support response: upgrade to the latest version.
Next step response, after you upgrade will be: reboot.

lelegard, 2021-12-11:
Hi @nhjm449 

I fixed this in commit da1d8593ed7e2f6c103bf0ae6897e678d493ef15. The option `--event` has been added to generate a playlist of type event. With option `--live`, the tag `EXT-X-PLAYLIST-TYPE` is no longer present. The default playlist type remains `VOD`.

nhjm449, 2021-12-11:
Thanks!! I should be able to test it next week.

You may also want to update
https://github.com/tsduck/tsduck/blob/da1d8593ed7e2f6c103bf0ae6897e678d493ef15/src/libtsduck/dtv/hls/tshlsPlayList.cpp#L846-L864
as it appears to:
- Assume a `?query#fragment` suffix won't exist
- Assume that only certain file extensions are valid
- Try to override the playlist type `getTag()` likely already set

Any URI that does not have an `EXT-X-STREAM-INF` or `EXTINF` associated with it should probably be considered an error.

lelegard, 2021-12-12:
Here, you refer to the `hls` _input_ plugin, right?

> Assume a ?query#fragment suffix won't exist

Yes, it should be done. I will fix this.

> Assume that only certain file extensions are valid

This is more questionable. First, we must guess the type of content. Is this a media playlist or a media segment? We check `.m3u8` or `.m3u` extensions for media playlist criteria. What should we do with an URI ending in `.foo`? Second, HLS accepts at least two formats of contents: TS and fragmented MP4. TSDuck exclusively manipulates TS and considers HLS as just one more TS source. So, we must be able to filter that ahead.

Do you have any experience of HLS / TS content with extensions other than `.m3u8`, `.m3u` and `.ts`?

> Try to override the playlist type getTag() likely already set

No, it does not. The methods `setType()` and `setTypeMedia()` do not blindly override the playlist type. If the playlist type is unknown, it does set the type as indicated. If the playlist type is already known, it checks that the requested type is compatible with the already established type and generates an error if it is not.

> Any URI that does not have an EXT-X-STREAM-INF or EXTINF associated with it should probably be considered an error.

To be very strict, maybe. But having a segment without information on it does not prevent to read TS packets from that segment. Generally speaking, TSDuck tries to be robust to partial or incorrect input, get what we can get and reject what it so incorrect that we cannot get anything from it (and avoid crashing to protect against fuzzing attacks). In the DVB domain, there are so many incorrect or incomplete signalisation everywhere that TSDuck has learnt to be flexible. This may be less true for HLS, I do not know...

In the `parse()` method, we accumulate data for the next playlist or segment. When an URI is found, we push the URI with the previously accumulated characteristics and reset the next characteristics. We accept to push an URI with empty / default characteristics. This is a design choice.

lelegard, 2021-12-12:
> Assume a ?query#fragment suffix won't exist
> Yes, it should be done. I will fix this.

Done in commit 1a362a6b211f8c08dc98670028a5791106e76e84

> HLS accepts at least two formats of contents: TS and fragmented MP4. 

I just realised that it supports 5 different formats in the latest version of the standard. It is even more important to filter TS content ahead.

nhjm449, 2021-12-16:
Hi @lelegard,

> Here, you refer to the `hls` _input_ plugin, right?

Yes; sorry for mixing that into an output plugin discussion. Thank you!

> This is more questionable. First, we must guess the type of content. Is this a media playlist or a media segment? We check `.m3u8` or `.m3u` extensions for media playlist criteria. What should we do [...]

I've never seen non-standard extensions used before, but, while unwise, it is both valid and possible. I assume non-`.m3u8`/`.m3u` playlists do exist in the wild.

Ran some experiments with reference players on macOS 11.5.2:

- Safari 15.0:
    - Master playlist:
        - Via address bar: `Content-Type` header *required*, regardless of extension
        - Via `<video>` tag: `Content-Type` header *or* `.m3u8` extension *required*
    - Media playlist (via master playlist reference): empty `Content-Type` header + unusual extension allowed
    - TS segment: empty `Content-Type` header + unusual extension allowed
- QuickTime 10.5:
    - Master playlist: `Content-Type` header *or* `.m3u8` extension *required*
    - Media playlist (via master playlist reference): empty `Content-Type` header + unusual extension allowed
    - TS segment: empty `Content-Type` header + unusual extension allowed

(`.m3u` not tested.)

> No, it does not. The methods `setType()` and `setTypeMedia()` do not blindly override the playlist type. If the playlist type is unknown, it does set the type as indicated. If the playlist type is [...]

By this, I specifically mean that the playlist type implied via tags (when they exist) should take precedence over file extensions. If you see `#EXTINF:1,\nmedia.m3u8` or `#EXT-X-STREAM-INF:...\nplaylist.ts`, then it's reasonable emit a warning questioning the user's sanity, but it should not emit a _wrong-playlist-type_ error.

Separately, rejecting bad files is certainly important, but that should be a different error class, and you don't know that they're bad until you actually load them.

> To be very strict, maybe. But having a segment without information on it does not prevent to read TS packets from that segment. Generally speaking, TSDuck tries to be robust to partial or incorrect [...]

The spec declares `EXTINF` as "REQUIRED" for each media segment. (And if this plugin ever gains support for adaptively switching between media playlists, the duration info will be very important.) The spec doesn't appear to *explicitly* say `EXT-X-STREAM-INF` is required, but it is strongly/indirectly implied.

But, yes, I imagine not requiring them is safer, and there's probably a use case for super-minimal playlist files that look more like non-HLS M3Us. I'd suggest emitting a warning when they're missing, at least.