tsduck/tsduck #907: [Q] HLS to UDP MPEG TS. Discontinuous stream

Ulfila80, 2021-12-10:
I'm trying to convert HLS stream to UDP MPEG_TS with tsp utility.

tsp -v --debug --bitrate 4000000 -I hls https://nbcnews2.akamaized.net/hls/live/723426/NBCNewsPlaymaker24x7Linear99a3a827-ua/VIDEO_2_3096000.m3u8 -O ip 239.230.3.3:3000

But output stream is very unstable. It appears in parts. You may see bitrate charts of output stream on screenshot from Dektec analyzer.

[image]

Is possible to convert HLS to continuous UDP MPEG_TS stream with tsp? May be I'm doing something wrong?

Here is detailed output:

```
~$ tsp -v --debug --bitrate 4000000 -I hls https://nbcnews2.akamaized.net/hls/live/723426/NBCNewsPlaymaker24x7Linear99a3a827-ua/VIDEO_2_3096000.m3u8 -O ip 239.230.3.3:3000
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp -v --debug --bitrate 4000000
* Debug: debug level set to 1
* Debug: hls: debug level set to 1
* Debug: hls: ====> tsp -I hls https://nbcnews2.akamaized.net/hls/live/723426/NBCNewsPlaymaker24x7Linear99a3a827-ua/VIDEO_2_3096000.m3u8
* Debug: ip: debug level set to 1
* Debug: ip: ====> tsp -O ip 239.230.3.3:3000
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: hls: downloading https://nbcnews2.akamaized.net/hls/live/723426/NBCNewsPlaymaker24x7Linear99a3a827-ua/VIDEO_2_3096000.m3u8
* Debug: hls: HTTP header: HTTP/1.1 200 OK
* Debug: hls: HTTP header: Akamai-Path-Timestamp: i=1639171212.514;xi=1639171212.537;xo=1639171216.507;s=1639171216.543;
* Debug: hls: HTTP header: Last-Modified: Fri, 10 Dec 2021 21:20:12 GMT
* Debug: hls: HTTP header: X-Akamai-Li
![hls2mpeg-ts](https://user-images.githubusercontent.com/62598847/145643820-3bb2775a-3cd7-415a-b46d-fcb8a16a063b.jpg)
ve-Origin-QoS: d=6000;t=1639171212.524
* Debug: hls: HTTP header: X-Akamai-Server: Akamai-SMT
* Debug: hls: HTTP header: Akamai-Mon-Iucid-Ing: 723426
* Debug: hls: HTTP header: Expires: Fri, 10 Dec 2021 21:20:21 GMT
* Debug: hls: HTTP header: Cache-Control: max-age=0, no-cache, no-store
* Debug: hls: HTTP header: Pragma: no-cache
* Debug: hls: HTTP header: Date: Fri, 10 Dec 2021 21:20:21 GMT
* Debug: hls: HTTP header: Alt-Svc: h3-Q050=":443"; ma=93600,quic=":443"; ma=93600; v="46,43"
* Debug: hls: HTTP header: Transfer-Encoding:  chunked
* Debug: hls: HTTP header: Connection: keep-alive
* Debug: hls: HTTP header: Connection: Transfer-Encoding
* Debug: hls: HTTP header: Akamai-Mon-Iucid-Del: 761199
* Debug: hls: HTTP header: Content-Type: application/x-mpegURL
* Debug: hls: HTTP header: Access-Control-Allow-Origin: *
* Debug: hls: HTTP header: Access-Control-Max-Age: 86400
* Debug: hls: HTTP header: Access-Control-Allow-Credentials: true
* Debug: hls: HTTP header: Access-Control-Expose-Headers: Server,range,hdntl,hdnts,Akamai-Mon-Iucid-Ing,Akamai-Mon-Iucid-Del,Akamai-Request-BC
* Debug: hls: HTTP header: Access-Control-Allow-Headers: origin,range,hdntl,hdnts
* Debug: hls: HTTP header: Access-Control-Allow-Methods: GET,POST,OPTIONS
* Debug: hls: curl: end of transfer
* Debug: hls: MIME type: application/x-mpegurl
* Debug: hls: unsupported HLS tag: EXT-X-PUBLISHED-TIME
* Debug: hls: unsupported HLS tag: EXT-X-CUE
* Debug: hls: unsupported HLS tag: EXT-X-SCTE35
* Debug: hls: unsupported HLS tag: EXT-X-SCTE35
* Debug: hls: unsupported HLS tag: EXT-X-CUE
* Debug: hls: unsupported HLS tag: EXT-X-CUE
* Debug: hls: unsupported HLS tag: EXT-X-CUE
* Debug: hls: unsupported HLS tag: EXT-X-CUE
* Debug: hls: unsupported HLS tag: EXT-X-CUE
* Debug: hls: unsupported HLS tag: EXT-X-CUE
* Debug: hls: unsupported HLS tag: EXT-X-CUE
* Debug: hls: unsupported HLS tag: EXT-X-CUE
* Debug: hls: unsupported HLS tag: EXT-X-CUE
* Debug: hls: unsupported HLS tag: EXT-X-CUE
* Debug: hls: unsupported HLS tag: EXT-X-CUE
* Debug: hls: unsupported HLS tag: EXT-X-CUE
* Debug: hls: unsupported HLS tag: EXT-X-CUE
* Debug: hls: unsupported HLS tag: EXT-X-CUE
* Debug: hls: unsupported HLS tag: EXT-X-CUE
* Debug: hls: unsupported HLS tag: EXT-X-CUE
* Debug: hls: unsupported HLS tag: EXT-X-CUE
[... 93 lines cut ...]
* Debug: hls: HTTP header: HTTP/1.1 200 OK
* Debug: hls: HTTP header: Akamai-Path-Timestamp: i=1639169431.491;xi=1639169431.506;xo=1639169436.686;s=1639169438.053;
* Debug: hls: HTTP header: Content-Length: 2691784
* Debug: hls: HTTP header: Last-Modified: Fri, 10 Dec 2021 20:50:31 GMT
* Debug: hls: HTTP header: X-Akamai-Live-Origin-QoS: d=6000;t=1639169431.496
* Debug: hls: HTTP header: X-Akamai-Server: Akamai-SMT
* Debug: hls: HTTP header: Pragma: no-cache
* Debug: hls: HTTP header: Akamai-Mon-Iucid-Ing: 723426
* Debug: hls: HTTP header: Cache-Control: max-age=603068
* Debug: hls: HTTP header: Expires: Fri, 17 Dec 2021 20:51:46 GMT
* Debug: hls: HTTP header: Date: Fri, 10 Dec 2021 21:20:38 GMT
* Debug: hls: HTTP header: Alt-Svc: h3-Q050=":443"; ma=93600,quic=":443"; ma=93600; v="46,43"
* Debug: hls: HTTP header: Connection: keep-alive
* Debug: hls: HTTP header: Akamai-Mon-Iucid-Del: 761199
* Debug: hls: HTTP header: Content-Type: video/MP2T
* Debug: hls: HTTP header: Access-Control-Allow-Origin: *
* Debug: hls: HTTP header: Access-Control-Max-Age: 86400
* Debug: hls: HTTP header: Access-Control-Allow-Credentials: true
* Debug: hls: HTTP header: Access-Control-Expose-Headers: Server,range,hdntl,hdnts,Akamai-Mon-Iucid-Ing,Akamai-Mon-Iucid-Del,Akamai-Request-BC
* Debug: hls: HTTP header: Access-Control-Allow-Headers: origin,range,hdntl,hdnts
* Debug: hls: HTTP header: Access-Control-Allow-Methods: GET,POST,OPTIONS
* hls: downloading from https://nbcnews2.akamaized.net/hls/live/723426/NBCNewsPlaymaker24x7Linear99a3a827-ua/1638854946_video_576p-30fps-3000/video_52599.ts
* hls: MIME type: video/mp2t, expected size: 2691784 bytes
* Debug: hls: curl: end of transfer
```

lelegard, 2021-12-11:
The tsp command is a processing tool, not specifically a streaming tool. By default, it processes packets as soon as it can. With HLS input, once a new segment is available, it is read (at full download speed) and packets are immediately passed to the next plugin. Then it waits for the next segment, and so on. This is absolutely fine if the output is a file, not if this is IP streaming where you observe the burst of each segment.

In short, HLS input is bursty by design while IP output shall be smooth. You need to regulate the flow between the input and output to absorb the bursts, for instance:
~~~
tsp -I hls ... -P regulate -O ip ...
~~~

TSDuck is a very versatile toolbox. It can work in many different environments. It won't work out of the box for any given context because there is no "always appropriate default". You need to understand the context, understand how tsp works and take appropriate actions.

I strongly recommend to read the reference section for tsp in the user's guide, especially the paragraph about "rendering and transmission speed" and option `--max-flushed-packets`. See also the documentation of the `regulate` plugin (you may try its `--pcr-synchronous` option).

Ulfila80, 2021-12-11:
Thank you very much for your answer.  I will take your advice and try.  I will write if I succeed or not.