tsduck/tsduck #910: [BUG] SpliceInject plugin not generating the expected splice duration if CUE-IN preceded by two CUE-OUT

moving-digital, 2021-12-13:
Hi @lelegard , I would need your help for the below bug:

#### Bug description:
- Working use-case: When starting TSP process, injecting a CUE-OUT then a CUE-IN, TSP/Spliceinject plugin will generate the expected break duration such as duration = delta between CUE-IN and CUE-OUT.
- Failing use-case: However, when starting TSP process, injecting a first CUE-OUT, then a second CUE-OUT, then a CUE-IN, TSP/Spliceinject plugin will not behave correctly and will simply not be able to generate the expected splice duration which should be the delta between the CUE-IN and the last CUE-OUT received.

#### How to reproduce:
- Working use-case: Start TSP process. Inject a CUE-OUT then a CUE-IN. The break duration shall be = delta between CUE-IN and CUE-OUT.
- Failing use-case: Start TSP process. Inject a first CUE-OUT, then a second one few seconds later. Finally, inject a CUE-IN. The break duration between the second CUE-OUT and the CUE-IN will not be generated correctly.

#### Expected behavior:
When injecting many CUE-OUT without any CUE-IN, and suddenly injecting a CUE-IN, this must result in creating an ad break with the duration being equal to the delta between the CUE-IN and the last CUE-OUT sent.

#### Errors and logs:
Not applicable.

#### Environment:
 - OS: Linux
 - OS version: CentOS 7.9
 - TSDuck full version: 3.30-2663
 - Installation type: official binary 

#### Additional information:
Not applicable.

moving-digital, 2021-12-14:
Hi @lelegard 

Some more info from my side that shall be pretty handy in order to troubleshoot:

Working test: 2 cues sent only
1. One CUE-OUT
2. One CUE-IN

[image]

Failing test: 3 cues sent
1. One CUE-OUT
2. Another CUE-OUT
3. One CUE-IN

[image]

The two TS captures can be downloaded here:
https://we.tl/t-q9V4NRMYCO

lelegard, 2021-12-14:
Hi @moving-digital 

I am afraid I do not understand the actual nature of the problem. Is this is a problem with the `spliceinject` or `splicemonitor` plugin? Which duration are you referring to? The splice_insert commands are immediate, so there is no duration involved.

moving-digital, 2021-12-14:
> Hi @moving-digital
> 
> I am afraid I do not understand the actual nature of the problem. Is this is a problem with the `spliceinject` or `splicemonitor` plugin? Which duration are you referring to? The splice_insert commands are immediate, so there is no duration involved.

Hi @lelegard, the problem is according to me with `spliceinject`.

When sending a cue-out (start break), then a cue-in (stop break) later, this should basically create a single break with a duration = delta from CUE-IN - CUE-OUT.

However, there is an issue when two cue-out are sent first, followed by one cue-in. In this case, CMA is not able to get the proper break information, I believe due to an issue in spliceinject.

lelegard, 2021-12-14:
If the problem is that the CMA analyzer is not able to compute a duration, then it seems a problem in the CMA analyzer, isn't it?

All your commands are immediate. So, `spliceinject` has little added value, it just injects the command immediately. That's all. There is no duration information in the splice table. If the analyzer is lost when it get two "splice out" and one "splice in" later, this seems a problem of the analyzer.

closed 2021-12-15 by moving-digital