tsduck/tsduck #911: [Q] Splicemonitor plugin on filtered ump multicast stream

odev1952, 2021-12-13:
Hello,

First of all thank you a lot for the work done so far on TSDuck. I'm new to mpegts and TSDuck has been a lot of help in my mpegts journey.
I am trying to use the splicemonitor plugin on a UDP multicast stream. This stream is a muxed stream containing several UDP multicast streams each with a specific service ID.
My objective is  to monitor SCTE 35 signals on a specific service id:
`tsp -I ip 184.179.30.38@239.127.111.38:9000 -P filter --service 1 -P splicemonitor -o scte_log.log -O drop`

The command above wouldn't generate any logs with the filter plugin. I can confirm that I do see the SCTE 35 splices when the filter plugin is removed.

Any ideas what I might be missing here?

Thank you!

lelegard, 2021-12-14:
Hi @odev1952 

Use `-P zap 1` instead of `-P filter --service 1`.

The `filter` plugin is a low-level tool which filters individual PID's based on selection criteria, regardless of the structure of the TS. `-P filter --service 1` means filter all PID's which are components of service 1. All other PID's are removed, including the PAT, As a result, the structure of the TS is broken. Without PAT, the next plugins downstream cannot locate services and cannot locate SCTE-35 PID's.

The `zap` plugin, on the other hand, rebuilds a single program transport stream (SPTS) with the selected services, removing all other services.

odev1952, 2021-12-15:
Thank you for the thorough explanation! This worked like a charm:)

closed 2021-12-15 by odev1952