tsduck/tsduck #912: [Q] DVB-SSU

nikpakar, 2021-12-15:
Can i use tsduck to implement dvb-ssu for a stb test bed.

Basically input from a binary file and output ts ssu to a udp multicast address.

Thanks in advance.

lelegard, 2021-12-15:
Hi @nikpakar 

An SSU is usually a DSM-CC carousel. Building the relevant DSM-CC sections (DSI, DII and DDB) is often vendor-specific. They should have the tools for that. It depends on how the boot-loader of the STB works and how the flash images are handled.

Once you have the DSM-CC sections, TSDuck can update the signalization and inject the sections. See section 5.2.15 in the user's guide. This was written from an actual use case I had to solve. But it was more than 10/12 years ago...

nikpakar, 2021-12-15:
Thanks a lot for the reply.

I already have a .ts file as the update from the STB vendor. How can i just stream that to a udp multicast for the mux.

nikpakar, 2021-12-18:
Sorry for that dumb question as tsp can do that. Im trying the following but getting few warnings and the receiver showing a lots of cc errors.

./tsp -v -a 14/24 -I file OTA-update.ts --infinite -P limit --bitrate 1024000000 -O ip 239.0.0.1:1234
* tsp: buffer failed to lock into physical memory (12: Cannot allocate memory), risk of real-time issue
* file: unknown initial input bitrate

Is there anything im doing wrong here. I just need to stream the .ts file to the ip multicast destruction at a desired bitrate.

No matter what i put in bit rate, it always goes at around 200-400mbps.

nikpakar, 2022-01-05:
Would highly appreciate if some one could shed some light for me on this. Thanks in advance

lelegard, 2022-01-05:
Did you check the structure of the TS from the vendor? Is it a full TS with all appropriate signalization, ready for a standalone injection, or is it the packetized DSM-CC carousel which must be muxed inside an existing TS?

Please read the documentation to understand how tsp and the various plugins work.

You read the file at the disk read speed and send it at the same speed over IP. 400 Mb/s is plausible, the bottleneck being either the disk or the network. In all cases, this is way too fast for the STB to get the SSU.

Read the documentation of the `limit` plugin. It does not regulate the playout speed. And anyway, if it did, you specify 1 Gb/s as bitrate ("giga", not "mega"). Replace this plugin with `-P regulate --bitrate 1,000,000`.

You should add the `-I file --infinite` option to make sure the TS is cycled and not sent only once. There is no guarantee that the STB will get the complete update in the first cycle.

The message "failed to lock into physical memory" is harmless. Since some users found it alarming, it has been removed in recent versions. This means that you are using an older version of TSDuck. You should upgrade.

nikpakar, 2022-01-12:
@lelegard Thierry, Many thanks for your valuable feedback as always.

Use of -P regulate make it works nicely and i can now stream the file at that bitrate.

How ever the .ts file sent by the stb manufacture does not recognised by the MUX as a DSM-CC carousel which im trying to multiplex it with the final transport stream as a component of a data broadcasting service.

Thank you so much again for your help.