tsduck/tsduck #913: [BUG] tstables --japan mis-detects CDT tables (pid 0x29) as ATSC TVCT table

trollcop, 2021-12-16:
#### Bug description:
I was doing some work with CDT tables (logo descriptor) and extracted several tables for analysis.
I recorded some ts, and extracted just pid 0x29 (the CDT tables)
I was hoping to dump the table contents with tstables and did
`tstables --japan cdt.ts`
But it didn't work.

#### How to reproduce:
See attached cdt.ts and run the command above

#### Expected behavior:
I expect to see some CDT tables getting dumped, or at least unknown table w/data shown, not ATSC table.

#### Errors and logs:

~~~
[be sure to surround verbatim log lines with "~~~" lines]
~~~

#### Environment:
 - OS: Windows
 - OS version: 11
 - TSDuck full version: 3.29-2651

 - Installation type: Official

#### Additional information:
Looking at the tstables --japan dump of the entire TS (without extracting just PID 0x29), it looks like maybe CDT tables are not handled at all?

[cdt.zip](https://github.com/tsduck/tsduck/files/7724910/cdt.zip)

lelegard, 2021-12-16:
Hi @trollcop 

You are right, the ARIB/ISDB-defined CDT is not yet implemented in TSDuck. This is still in the [TODO list](https://github.com/tsduck/tsduck/blob/master/TODO.txt).

I just did a quick check. Currently, 37 tables and 243 descriptors are implemented from various standards (MPEG, DVB, ATSC, SCTE, ARIB/ISDB). The [TODO list](https://github.com/tsduck/tsduck/blob/master/TODO.txt) still contains 5 tables and 29 descriptors to implement, waiting for time or volunteers.

In your case, the ATSC-defined TVCT and ISDB-defined CDT share the same table id: 0xC8. The lack of coordination between standards organizations is very unfortunate.

TSDuck handles unknown or conflicting table ids (and descriptor ids equally) the following way.
- If a table id is unknown, the section payloads are uninterpreted and displayed in hexadecimal (`<generic_long_table>` in XML).
- When two or more tables are implemented with conflicting ids, the "current" standard is used to select the one to interpret. The current standard may come from the context (for instance, encountering an unambiguous ATSC table defines the current standard as ATSC) or from explicit command line options (in your case `--japan` is a shortcut which includes `--isdb`).

Here, the problem is that the ATSC-defined TVCT is the only table which is implemented with table id 0xC8, hence the display. Sorry about that. Any volunteer to implement the ISDB-defined CDT?

lelegard, 2021-12-16:
>  Any volunteer to implement the ISDB-defined CDT?

I would add that ARIB standards are not generally available in English. In most cases, only obsolete versions are available in English, recent versions are in Japanese only. Some documents such as [ARIB STD-B16 (ISDB Standard Digital Receiver for Satellite)](https://www.arib.or.jp/english/std_tr/broadcasting/std-b16.html) are not available at all in English. This is why TSDuck needs Japanese-speaking volunteer developers...

trollcop, 2021-12-16:
Hey,

> Here, the problem is that the ATSC-defined TVCT is the only table which
is implemented with table id 0xC8, hence the display.

At the moment at least while its not implemented, would it make sense to
treat it as "unknown" table ID when used with --japan or --isdb, and t hus
at least display it in hex?

> If a table id is unknown, the section payloads are uninterpreted and
displayed in hexadecimal (<generic_long_table> in XML).
How do I make that show during tstables file.ts output? Or it doens't work
because TVCT decoder takes over?

> Any volunteer to implement the ISDB-defined CDT?
I've implemented CDT table generator so I should have no problem writing a
parser for it as well, tho I'm not sure how useful it would be coding-style
wise. If you point me in the direction of TVCT decoder i could write some
pseudocode/something to parse the resulting stuff from the data thrown
around the specs.

Would there be a way to select using TVCT or CDT when --japan option is
used?

lelegard, 2021-12-16:
This is not as simple as that. There are so many tables and descriptors (currently 37 tables and 243 descriptors) that this requires some robust and generic structure. There is no room for a hack such as "if table_id == 0xC8 and standard == ISDB then display as unknown".

Implementing a table is a bit more than parsing pseudo-code. All tables and descriptors shall be read/write, in binary, XML and C++ forms. Each table or descriptor is implemented as a C++ class which implements
- Binary deserialization
- Binary serialization
- Textual display (`tstables` default output for instance)
- XML deserialization
- XML serialization

Fortunately, the JSON implementations are done using automated XML <=> JSON translation. Otherwise, that would have made two more methods per class.

It cannot be half-done. Imagine the nightmare of dealing with 37 tables and 243 descriptors types and remembering this one is only displayable, this one has no XML support, this one has deserialization only, etc.

I have developed most of them. A few tables and descriptors have been contributed by users/developers who needed them.

lelegard, 2021-12-16:
OK, this CDT has a rather trivial structure. I just implemented it (80% copy/paste from other basic tables). Have a look at commit 07cf74560e4c11c92e0822a7e7c4971789adcd8a if you are interested in the amount of work to fully implement a new table.

trollcop, 2021-12-16:
Nice, yeah it's not a very complex table. Thanks for getting it done so quickly <3

trollcop, 2021-12-17:
I can confirm that 2670 prerelease correctly renders CDT table with --japan and the cdt.ts attached in first post.
Thank you!

closed 2022-01-07 by lelegard