tsduck/tsduck #915: [Q] bitrate_monitor usage with alarm_command and periodic-bitrate params

duncanbarnes, 2021-12-29:
I don't believe this is possible at the moment but thought I would ask in case there is a trick to make it work!

I'm trying to use the bitrate_monitor to monitor input and output bitrates, both for alarms and for status monitoring. I have a script that works to fire the alarm_command variables at an external service gathering data from tsswitch and other processes. 

Is it possible for the alarm_command to be triggered at the specified period?

I can only make it trigger when the signal is outside min and/or max thresholds so this may be an enhancement on top of #31

lelegard, 2021-12-30:
I do not really understand your need. Could you precisely specify what you would like to achieve?

duncanbarnes, 2021-12-30:
My apologies. I'm trying to monitor the bitrate of the transport streams entering and leaving tsswitch in a script which updates a database and monitoring dashboard.

If the bitrate_monitor plugin is started with --alarm-command and --periodic-bitrate arguments, it would be great if the script being called by --alarm-command received values at the --periodic-bitrate interval.

Does that make more sense?

lelegard, 2021-12-31:
OK, I see. Instead of the alarm command being run when there is a change of state, you would like it to be triggered at periodic intervals, reporting the current state, regardless of the state change. Do I get it correctly?

In that case, instead of changing the semantics of the current options, I would rather add some new `--periodic-command` option.  It would get a duration in seconds, just like `--periodic-bitrate`, and forces the `--alarm-command` to be run at periodic intervals, in addition to state changes.

I prefer to keep `--periodic-bitrate` outside of this because reporting a bitrate is a cheap action while running a command (process creation, etc) is more expensive. We may like to have a cheap periodic bitrate reporting and an expensive command launch on alarms only.

duncanbarnes, 2021-12-31:
Yes thats correct. It's purely confidence monitoring to assist with fault finding when there are problems. Noted your comment about cheap and expensive actions, will try not to have too low an interval for reporting but your suggestion sounds excellent, thank you!

lelegard, 2021-12-31:
OK, option `--periodic-command` has just been added. Please try it and let us know if it works as expected

duncanbarnes, 2022-01-03:
Apologies for the delay. Just compiled and tested (on Mac M1 Max), works as expected. I really didn't expect that to be done so soon, thank you so much!

closed 2022-01-03 by duncanbarnes