tsduck/tsduck #916: [FEAT] Minos recipe for static compiling

lars18th, 2021-12-30:
#### Is your feature request related to a problem?
Similar to the Windows PORTABLE version, it will be desirable to have a _full_ Linux STATIC version.

#### Proposed solution:
I suggest to provide a recipe to compile TSDuck for the Ninos-static project (https://github.com/minos-org/minos-static) and make it available in his static-get repository (http://s.minos.io/). Then we will very easy to get it: `static-get tsduck` and run it in any Linux machine without compiling it.

#### Alternatives:
The current release method includes pre-compiled packages for several platforms. And this is nice. However, a full static version will be useful. Mainly to run it for testing on servers and/or containers.

#### Additional information:
I think the effort to provide a full static recipe for TSDuck will be easy for the author. And if hi maintains it, then it will be more easy to have new releases of TSduck supported. Please note that I'm not affiliated with this Minos project, but I have recently discovered it and find it interesting.

lelegard, 2021-12-30:
Building a static version of TSDuck has always been possible using `make static`. I haven't tried it for a while. I hope nothing was broken in the meantime. Please let us now if it still works.

I wasn't aware of that minos project. I tried to understand the rationale behind that and I can't help considering it as rather mistaken...

Building a static TSDuck on Linux is probably the worst idea on Earth. The only context into which this could be justified is a very small and specialized embedded system using a tuned version of `tsp` and nothing else. Any other use seems pretty silly for many reasons, including:

- Only basic IP networking is possible, using bare IP addresses: no host names, no DNS, no HTTP, no HLS, no SRT, no RIST, just the Internet of the late 70's.
- No smartcard support, no Dektec support and more generally no support for any feature which is available in shared library form only.
- TSDuck is made of several tools. Each of them is very small, they all use the huge `libtsduck.so` shared library. With static linking, you get many huge executables. Not a very clever result.
- All plugins must be statically linked inside `tsp` or `tsswitch`. They can no longer be shared libraries.
- There could be no "standard static release" of TSDuck, only specialized builds for specialized needs. Simply because static linking disables many features.

This is really stepping back in history. I do not want to invest in that. Using `make static` and making sure it still works is the most I can accept about static linking. If anyone wants to contribute somehow to a "minos recipe" for TSDuck based on `make static`, fair enough, but I do not want to be involved in that.

Please note that more modern OS such as macOS have completely removed the capability of static build because it created more problems than it solved. Static linking is a remain of the past, nothing else.

lars18th, 2021-12-31:
Hi @lelegard ,

Ok, don't worry. However, please continue with your support of the `make static`.

In any case, without the idea of opening any out-of-the-scope discussion, let me to add one personal comment: Static linking support is a must-have for containers, distributed computing and IoT. From my point of view the current MacOS can't be used as an example of any _server side_ modern operative system.

Regards.

closed 2021-12-31 by lars18th

lelegard, 2021-12-31:
> Static linking support is a must-have for containers, distributed computing and IoT. 

I can just repeat what I wrote: this is OK for a very small specialised embedded system (IoT) but _definitely not_ on a server-side container because many TSDuck features are _unavailable_ with static linking.