tsduck/tsduck #917: [Q] How to generate a pure carrier with Dektec?

szliucm, 2021-12-30:
I need a pure carrier for test , that the output S2 signal is replaced by a simple sine wave at the RF frequency. 
The level of the unmodulated carrier corresponds to the output power. This mode allows, among others, phase noise measurement without requirement of an external amplifier.
How can I do it with Tsduck?

lelegard, 2021-12-30:
Which type of Dektec modulator do you use? Did you check that this device can do that?

TSDuck just drives the Dektec devices. If some software API can do that, TSDuck can be enhanced to use it.

However, please note that the `dektec` output plugin sends the transport stream from `tsp` to a Dektec device. What you want is just an empty signal into which there is no transport stream. So, using `tsp` is a bit useless. If the Dektec modulator can produce that type of empty signal, did you check if some Dektec-provided software can do that already?

szliucm, 2021-12-31:
lelegard, thanks lot for your help.
I got the solution from Dektec just now.
To set RF Mode to CW-channel only, you can use the DTAPI_UPCONV_CW option for the DtOutpChannel::SetRfMode, as described on page 438.

lelegard, 2021-12-31:
The option `--carrier-only` has been added to the `dektec` output plugin. Please try it because I do not have any Dektec modulator here today to test. I am not 100% sure of the result because there is some inconsistency here: we set the modulator output to "carrier only" and we sent a TS at the same time. Please let us know what is the behaviour, either on the software side (any error when sending the TS?) and on the signal side (is this really an empty carrier?)

szliucm, 2022-01-06:
@lelegard You concern is right. The manual does not make it clear. It should set the option but not send out the TS to get a "carrier   only". Otherwise we can only get normal wave.

lelegard, 2022-01-06:
> I modified the "DtPlay" to get the result.

Do you mean that you added `DTAPI_UPCONV_CW` to `SetRfMode`, sent the TS, analyzed the signal and saw the modulated TS on the analyzer? I mean not only the carrier but also the modulation over it.

> Perhaps we should use tsdektec alternatively instead of tsp.

Yes, that makes sense in theory. In practice, I wonder what kind of options we need to add. Normally, the frequency should be enough. But will it work in terms of software sequence? Do we need to set at least the modulation type ? Even though there is no modulation over the carrier. There are tests to be done and potentially a lot of code to add. Working from home because of the pandemic, I have no access to any Dektec modulator or signal analyzer.

I have added option `--carrier-only` to the `dektec` output plugin. Could you please test it with a signal analyzer and check if the modulated TS is present or not? If the modulation is still present, a compromise could be to drop output packets when `--carrier-only` is set. Not fully satisfactory, intellectually speaking, but this is a very specific and rare use case.

szliucm, 2022-01-07:
When I start tsp with --carrier-only , the modulated TS is still present. It need to drop all output packets to get a pure carrier as I did with DtPlay.  It said tsp need at most 1 output plugins .Can you give me some idea about how to drop packtes with tsp command?

lelegard, 2022-01-07:
OK, just I added two things

- In output plugin `dektec`, when `--carrier-only` is present, all packets are dropped instead of being sent. This seems consistent with "carrier only", meaning no TS.
- In command `tsp`, I added the option `--final-wait` to add a wait time after getting the end of input. Zero means infinite. This may be useful is other cases also.

Thus, you can try the following command:
~~~
tsp --final-wait 0 -I null 1 -O dektec --carrier-only --frequency ...
~~~

In fact, any input TS is fine with `-O dektec --carrier-only` but using `-I null` alone would saturate the CPU, generating and dropping null packets. With `tsp --final-wait 0 -I null 1`, we generate one single null packet (required to start everyhting) and then wait forever, maintaining the output carrier. Could you please test with the latest repo?

I admit this is a bit awkward but this is a very specific use case.

szliucm, 2022-01-07:
The output wave is just as expected, "carrier only",  when I use  tsp --final-wait 0 ... with the latest repo. 
By the way, when I terminated the command , the carrier is still there . Not like the normal tsp command's behavior.

lelegard, 2022-01-07:
> By the way, when I terminated the command , the carrier is still there . Not like the normal tsp command's behavior.

I assume that this is because the output channel of the Dektec device is not properly closed after the interrupt. When tsp terminates or is "properly" interrupted, all plugins are explicitly closed. In the case of the `dektec` output plugin, the channel is explicitly detached. I assume that this mutes the carrier.

In the current case, you probably have to hit Ctrl-C twice. The first time is a user interrupt, what I call "properly interrupted". This interrupt is handled by tsp which notifies all plugins to terminate. However, the plugins take that interrupt request synchronously, when they process the next packet. This is immediate in the usual case since hundredths or thousands of packets are processed each second. However, in our case, all plugins remain asleep, no packet is ever processed, no plugin can synchronously process the interrupt request and tsp does not terminate. This is why you have to hit Ctrl-C a second time. In this case, this is a hard interrupt and command is killed without executing any code, meaning not explicitly detaching the Dektec channels.

To validate this hypothesis, you may try tsp `--final-wait 1000 ....`. In that case, tsp waits only 1 second instead of forever and all plugins are properly terminated. 

Which model of modulator do you use?

szliucm, 2022-01-07:
I'm using Dta-2115B. 
No matter pressing Ctrl-C twice or tsp --final-wait 1000 ...., the carrier remains. 
When I use normal tsp  without --carrier-only , the modulated carrier will disapear immediatly after the command  been terminated

lelegard, 2022-01-07:
OK, so that means that the channel detach operations treats "carrier only" and modulated TS in different manners.

I have made a modification to explicitly mute the carrier on close on all modulators. This code was rather old. In the first Dektec modulators, RF muting was not supported. When the DTA-115 was introduced (~12 years ago), it became possible. Code from that time explicitly sets RF mute on close on DTA-115 only. I just change it to set it on all modulators, except the oldest ones, before DTA-115 (DTA-107 and DTA-110T). Please check if the carrier disappears after a normal close (`tsp --final-wait 1000 -I null 1 -O dektec --carrier-only ...`).

Note that this won't change in the case of hard interrupt (Ctrl-C twice) since there is no proper close.

szliucm, 2022-01-08:
The result as below:
1、tsp --final-wait 1000 ... wait forever instead of 1 second. Neet  hard interrupt. 
2、 --final-wait 1000 ... with      --carrier-only ， Ctrl-C once , unmodulated carrier disappers immediately
3、 --final-wait 1000 ...without --carrier-only ， Ctrl-C twice , modulated carrier disappers after about 30 seconds.

lelegard, 2022-01-08:
> tsp --final-wait 1000 ... wait forever instead of 1 second.

This does not make sense. Did you actually specify `-I null 1` (1 packet and ends) and not `-I null` (infinite packet generation)?

szliucm, 2022-01-09:
You are right. When I specfiy -I null 1, everything runs as you predicted.
Thank you so much.

closed 2022-09-23 by lelegard