tsduck/tsduck #918: [Q] How we can convert dvb-t to dvb-s by transmodulation related to the TSDuck ?

haddi21fr, 2022-01-17:


lelegard, 2022-01-17:
You need a DVB-T tuner (any brand, USB or PCI) and a Dektec modulator (USB or PCI).

Then use a command such as:
~~~
tsp -I dvb ... -O dektec ... --stuffing
~~~

I let you read the user's guide (reference sections and examples) for the possible options for the `dvb` input plugin and the `dektec` output plugin. Please note that the option `--stuffing` is required in the `dektec` output plugin because the input DVB-T stream is mostly likely slower than the output DVB-S stream and this option tells the modulator to generate stuffing when the input stream is not fast enough for the selected modulation options.

haddi21fr, 2022-01-17:
Dear Thierry Lelegard

Thank you very much for your kind respond.

Best regards
Hadi

haddi21fr, 2022-01-18:
How we can watch satellite tv by only internet on android phone or pc(windows) with app so that this app is uses internet very very few and without need vpn ?