tsduck/tsduck #922: [Q] How to make use (or set) userAgent on web/hls requests

dutchboontje, 2022-01-20:
Hi @lelegard ,

At first. I would like to mention:  Really good work of you on this streaming solution  !

I am working on a specific application of TSDuck.
One of the parts is to download HLS streams.

On an earlier version of this project, I was using ffmpeg, but I dropped this traject. 
Because of the many problems on handling stream conversions.
You product is just perfect for my use case. Nice to understand, super simple to implement.

For HLS download, I have a simple question:

- How to make use (or set) userAgent on web/hls requests ?

I have been looking a bit in the sources, but can not find out how to set it from HLS(input) plugin.
I guess it will be a simple use, which I overlooked.

Please, can you send a hint ?    ;)

lelegard, 2022-01-21:
Hi @dutchboontje 

Currently, the user agent is hard-coded to "tsduck" because no reason emerged to use any specific value. It should be trivial to make it a command line option to all plugins and commands which use web requests. Let me have a look at it.

lelegard, 2022-01-21:
Option `--user-agent` has been added. Please test it.

dutchboontje, 2022-01-21:
Thanks, that's really quick.
At this moment, it is running.  I can confirm it OK.

I got some compile errors, but after ignoring those the bin. is working.

My compile machine is a bit dated:
`...tsduck$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.6 LTS
Release:        18.04
Codename:       bionic

...tsduck$ cpp --version
cpp (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
`
For your info:
`In file included from tseit.cpp:36:0:                                                                                                                                                                              
/media/truenas-development/compile/programs/tsduck/src/libtsduck/base/app/tsCommandLine.h: In member function ‘void ts::CommandLine::setCommandLineHandler(HANDLER*, ts::CommandStatus (HANDLER::*)(const ts::UStri
ng&, ts::Args&), const ts::UString&)’:                                                                                                                                                                             
/media/truenas-development/compile/programs/tsduck/src/libtsduck/base/app/tsCommandLine.h:113:9: warning: declaration of ‘command’ shadows a member of ‘ts::CommandLine’ [-Wshadow]                                
         {                                                                                                                                                                                                         
         ^                                                                                                                                                                                                         
/media/truenas-development/compile/programs/tsduck/src/libtsduck/base/app/tsCommandLine.h:94:15: note: shadowed declaration is here                                                                                
         Args* command(HANDLER* handler,                                                                                                                                                                           
               ^~~~~~~                                                                                                                                                                                             
/media/truenas-development/compile/programs/tsduck/src/libtsduck/base/app/tsCommandLine.h: In instantiation of ‘void ts::CommandLine::setCommandLineHandler(HANDLER*, ts::CommandStatus (HANDLER::*)(const ts::UStr
ing&, ts::Args&), const ts::UString&) [with HANDLER = ts::EITCommand; typename std::enable_if<std::is_base_of<ts::CommandLineHandler, HANDLER>::value, int>::type <anonymous> = 0]’:                               
tseit.cpp:278:72:   required from here                                                                                                                                                                             
/media/truenas-development/compile/programs/tsduck/src/libtsduck/base/app/tsCommandLine.h:112:14: warning: declaration of ‘command’ shadows a member of ‘ts::CommandLine’ [-Wshadow]                               
         void setCommandLineHandler(HANDLER* handler, CommandStatus (HANDLER::*method)(const UString&, Args&), const UString& command = UString())                                                                 
              ^~~~~~~~~~~~~~~~~~~~~                                                                                                                                                                                
/media/truenas-development/compile/programs/tsduck/src/libtsduck/base/app/tsCommandLine.h:94:15: note: shadowed declaration is here                                                                                
         Args* command(HANDLER* handler,                                                                                                                                                                           
               ^~~~~~~
`

dutchboontje, 2022-01-21:
As I am not able to help on the compile errors.
I am willing to test for you, when any help is needed.

;)

lelegard, 2022-01-22:
GCC 7.5.0 is somewhat old. Recent distros have GCC 10 or 11. I know by experience that GCC 7 introduced a number of bugs and excessively pedantic warnings. The ones you see are part of them. Some programs compile in GCC 4 and GCC 8 but not GCC 7. This clearly means that GCC 7 incorrect behaviour was reverted in GCC 8 onwards.

TSDuck used to compile on GCC 7 when it was current but more recent part of the code may trigger warnings as you see. I just updated the code so that it now compiles with GCC 7 (I have a few old versions of GCC, including 7.5.0, running on an Alma Linux 8 VM).

You may update your repo and retry the build. It should compile.

dutchboontje, 2022-01-31:
I am able to compile on Ubunto 18 again now,

Thanks again.

closed 2022-01-31 by dutchboontje