tsduck/tsduck #925: [Q] DTAPI_E_INVALID_SYMRATE (DTAPI status 4213)

zjaffesyna, 2022-01-23:
Hello i had TSDuck version 3.19-1520, and when i run commands like the bellow, it worked fine:

tsp -v -I ip 239.254.253.31:1234 -O dektec -d 0 --frequency 1528000000 --symbol-rate=13000000 -r 8/9 --modulation=DVB-S2-QPSK

i have upgraded now to the new TSDuck , 3.29-2651, and since the upgrade i get the following error:
" Error: dektec: output device start send error: DTAPI_E_INVALID_SYMRATE (DTAPI status 4213)"

i tired to change the " tsdektec  --power-mode high-quality" , but it didnt help.

Please adivce.

lelegard, 2022-01-23:
Hi,

If the device is a DTU-315, try to set power in the output plugin itself (`-O dektec --power-mode high-quality ...`) See #453.

Otherwise, please try the latest nightly build (or rebuild from the repo) for version 3.30. There was a similar recent issue which was fixed.

zjaffesyna, 2022-01-23:
thanks for your reply.
i tired that:
tsp -v -I ip 239.254.253.31:1234 -O dektec -d 0 --frequency 1528000000 --symbol-rate=28800000 -r 8/9 --modulation=DVB-S2-QPSK --power-mode high-quality

it didnt help.
i went back to my old version 3.19-1520.

will wait to see if it was fixed. 

thanks

lelegard, 2022-01-23:
> will wait to see if it was fixed.

Precisely, did you try with 3.30 nightly builds ?

zjaffesyna, 2022-01-24:
no, i'm not sure how to do that.

zjaffesyna, 2022-01-24:
no, i cant build versions. will wait for a release.

lelegard, 2022-01-24:
If you use Windows or Ubuntu, the latest nightly builds are here: https://tsduck.io/download/prerelease/

zjaffesyna, 2022-01-24:
it now it is working, thanks a lot @lelegard

closed 2022-01-25 by lelegard