tsduck/tsduck #927: [BUG] merge plugin doesn't seem to work too well with files

bstanden, 2022-01-24:
#### Bug description:
Section 5.2.23 of the manual gives an example of merging live transport streams.

Could an example be given of an equivalent scenario using captured transport stream files?

#### How to reproduce:
Capture a main stream, say a.ts - ensure it has lots of stuffing available.
a.ts is CBR stream with valid PCRs

"zap" a SPTS from another stream, resulting in b.ts
b.ts may be a VBR stream.

Object is to insert the service from b.ts into a.ts and merge SI, resulting in c.ts

$ tsp -I file a.ts -P merge "tsp -I file b.ts" -O file c.ts

#### Expected behavior:
c.ts should include all the services from a.ts as well as the single service from b.ts

#### Errors and logs:
b.ts is inserted into a.ts using all available stuffing at the bitrate of a.ts - it is as though the insertion is not respecting the bitrate (by PCRs) of b.ts (this is clearly visible in a tool like DVBInspector)

c.ts is unplayable in respect of the service from b.ts.  Services from a.ts seem ok.

~~~
[be sure to surround verbatim log lines with "~~~" lines]
~~~

#### Environment:
MacOS: 3.30-2690 (homebrew)
Debian: 3.29-2640 (built from source)

#### Additional information:
I apologies in advance if the problem is more my lack of understanding.

lelegard, 2022-01-25:
Hi @bstanden 

The `merge` plugin inserts TS packets as they arrive over time. They are not placed according to the PCR from the merged stream. On the contrary, PCR from the merged stream are updated according to their placement in the final stream (unless `--no-pcr-restamp` is used).

It typically works fine when streams are played at rendering speed. When inputs are files, the input TS are read at full disk speed and the merge is completely different, probably completely irrelevant.

So, the merge process won't work at file speed. You need to recreate the rendering speed using a `regulate` plugin. In both `tsp` commands in your case. The drawback is that the processing is slower (rendering speed instead of file speed).

bstanden, 2022-01-25:
Hi @lelegard 

Thank you for the detailed response - this is understood.  I'll close the issue as it appears that what happens is 'by design' and as you suggest, the "merging" of files in this way is probably not a relevant process.

I've had some success using the 'regulate' plugin in the way you suggest - of course, this means merge processes take time, particularly if multiple services are to be merged.

I'll raise a feature request instead :)

Bob