tsduck/tsduck #931: [BUG]  Cross compile problem

j3dd4h, 2022-01-28:
#### Bug description:
Unable to cross compile tsduck

#### How to reproduce:

make -j16 ARFLAGS=rc MAIN_ARCH=arm  CROSS=true CROSS_TARGET=arm-linux-gnueabi \
	CPP=/usr/bin/arm-linux-gnueabi-cpp CC=/usr/bin/arm-linux-gnueabi-gcc \
	CXX=/usr/bin/arm-linux-gnueabi-g++ GCC=/usr/bin/arm-linux-gnueabi-gcc \
       LD=/usr/bin/arm-linux-gnueabi-ld	AR=LD=/usr/bin/arm-linux-gnueabi-ar  \
       NOPCSC=1 NOCURL=1 NODEKTEC=1 NOTEST=1 NOCURL=1 NOPCSC=1 NOEDITLINE=1 NOSRT=1 NORIST=1 NOTELETEXT=1 NODTAPI=1 static

#### Errors and logs:
~~~
  [LD] /home/builder/tsduck-master/bin/release-arm-tsduck-static/objs-libtsduck/alltables.o
  [LD] /home/builder/tsduck-master/bin/release-arm-tsduck-static/objs-libtsduck/alldescriptors.o
  [LD] /home/builder/tsduck-master/bin/release-arm-tsduck-static/objs-libtsduck/allcharsets.o
  [LD] /home/builder/tsduck-master/bin/release-arm-tsduck-static/objs-libtsduck/allplugins.o
  [AR] /home/builder/tsduck-master/bin/release-arm-tsduck-static/libtsduck.a
rc: /home/builder/tsduck-master/bin/release-arm-tsduck-static/libtsduck.a: No such file or directory
make[3]: *** [Makefile:176: /home/builder/tsduck-master/bin/release-arm-tsduck-static/libtsduck.a] Error 1
make[2]: *** [Makefile:37: default] Error 2
make[1]: *** [Makefile:59: default] Error 2
make: *** [Makefile.inc:527: static] Error 2

~~~

#### Environment:
 - OS:  Ubuntu
 - OS version: 20.04

lelegard, 2022-01-28:
Hi @j3dd4h 

The message `rc: .../libtsduck.a: No such file or directory` comes from command `rc`. WTF?

Look at your command:
~~~
make ... AR=LD=/usr/bin/arm-linux-gnueabi-ar
            ^^^
~~~
Thus, `AR` is empty and the command `$(AR) rc` becomes command `rc`...

j3dd4h, 2022-01-28:
Thank you !!
I need to change my glasses :)

closed 2022-01-28 by j3dd4h