tsduck/tsduck #932: [BUG] PES ENCAP documentation and PCR

lars18th, 2022-01-28:
#### Bug description:
In the current documentation the example at **5.2.28 Preserving EIT’s through ffmpeg processing** has a bug. The line
```
-P encap --pes-mode fixed --pid 0x12 --output-pid 0x900 \
```
Requires to indicate the PCR pid. Without this parameter the tsp process doesn't start:
```
-P encap --pes-mode fixed --pcr-pid 0x1e1 --pid 0x1f1 --pes-offset -250 --output-pid 0x900 \
```
Futhermore I recomend to add a `pes-offset` with a small negative value to overcome the drift when doing the decapsulation.

#### Additional information:

Another question is that: If the `encap` plugin with the PES mode enabled incorporates PTS timestamps referencing another pid with the PCR, then why the resulting encapsulated pid incorporates PCR marks? Perhaps these new PCR marks could have sense on some cases, but normally the best is to not add them. This will reduce the space consumed by the encapsulation. And are more consistent. So I suggest to add a new parameter (`--remove-pcr`) to remove such PCR. Or do it as reverse and not add PCR marks with an option to enable them.

I hope @lelegard you want to fix this soon.
Regards.

lars18th, 2022-01-28:
Hi @lelegard ,

Also the example has incorrect stream identification of the stream (it's Synch, so 0x15) in the PMT:
https://github.com/tsduck/tsduck/blob/7ff0b99115cc1919f302ce85af6ff3abd707bb59/src/libtsduck/dtv/encap/tsPacketEncapsulation.h#L129-L130

So, the example (complete) requires to be:
```
tsp --add-input-stuffing 1/10 \
 -I dvb --uhf-channel 30 \
 -P zap arte --audio fra --no-subtitles --eit --stuffing \
 -P encap --pes-mode fixed --pcr-pid 0x1e1 --pid 0x12 --output-pid 0x900 \
 -P pmt --add-pid 0x900/0x15 --add-pid-registration 0x900/0x4B4C5641 \
 -P filter --negate --pid 0x1FFF | \
...
```

Regards.

lelegard, 2022-01-31:
Hi @lars18th 

The example in the user's guide is an encapsulation in asynchronous mode. It does not need a PCR PID and `tsp` does start and produces the outer PID. The correct stream type is 0x06. I just tried.

You would like to modify the example to synchronous mode (needs `--pcr-pid`, `--pes-offset` and stream type is 0x15). Why not. But this does not mean that the example is incorrect, it is just not the same example.

> Another question is that: If the encap plugin with the PES mode enabled incorporates PTS timestamps referencing another pid with the PCR, then why the resulting encapsulated pid incorporates PCR marks? 

What do you mean? The `encap` output PID does not contain any PCR. In synchronous mode, with `--pcr-pid` and `--pes-offset`, it contains PTS, not PCR.  I just tried.

And BTW, you probably understand PES mode encapsulation better than me: @lars18th, you are the one who implemented it, remember ? 😄