tsduck/tsduck #934: [Q] analyze multicast stream

hygortc, 2022-02-02:
I'm trying to parse the ts by the command 

tsp -m -I ip 236.0.0.1:3004 --local-address 100.64.93.246 \     
    -P until --seconds 10 \
    -P analyze \
    -O drop

the command output does not return anything

rminating...
% tsp -m -I ip 236.0.0.1:3004 --local-address 100.64.93.246 \
    -P until --seconds 10 \
    -P analyze \
    -O drop

* [MON] 2022/02/02 17:08, resource monitoring started
* [MON] 2022/02/02 17:08, VM: 32 GB (+17 MB), CPU:0.09% (average:0.09%)
* [MON] 2022/02/02 17:08, VM: 32 GB (stabilizing), CPU:0.00% (average:0.05%)
* [MON] 2022/02/02 17:08, VM: 32 GB (stabilizing), CPU:0.00% (average:0.03%)

i am using mac os x monterey intel processor

lelegard, 2022-02-02:
First, use `tsp -I ... -P until --seconds 10 -O file test.ts`

If there is no input packet, the command will probably never terminates because `-P until --seconds 10` means "terminate when a TS packet is passing 10 seconds after the first one". Meaning no packet = no end.

Ultimately, you may try `tsp -I ... | tsdump`. If it doesn't display anything, there is not even one single packet.

On the other hand, if the first command produces a non-empty `test.ts` file, you may analyse it later.

hygortc, 2022-02-03:
I performed the test with the commands and observed that the multicast stream did not arrive. when I test with tsreader and the stream arrives...

I verified that when I use tsduck the multicast stream does not arrive and through the tsreader the stream arrives

so open tsreader and tsduck and tsduck was able to capture the multicast traffic that tsreader requested...

I believe it's something with tsduck's igmp request on OSX

hygortc, 2022-02-03:
Texto de origem
actually tsp is not creating the request for the multicast stream

I am attaching the test performed by tsreader and the test performed by tsduck 

tsreader via virtualbox in ethernet bridge... traffic captured from host
[image]

test tsduck
[image]

hygortc, 2022-02-03:
% tsp --debug -I ip 236.0.0.1:3004 --local-address 100.64.93.246 -P until --seconds 10 -O file test.ts
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp --debug
* Debug: debug level set to 1
* Debug: ip: debug level set to 1
* Debug: ip: ====> tsp -I ip 236.0.0.1:3004 --local-address 100.64.93.246
* Debug: file: debug level set to 1
* Debug: file: ====> tsp -O file test.ts
* Debug: trying to load "/usr/local/Cellar/tsduck/3.29-2651/bin/tsplugin_until.dylib"
* Debug: dlopen(/usr/local/Cellar/tsduck/3.29-2651/bin/tsplugin_until.dylib, 0x000A): tried: '/usr/local/Cellar/tsduck/3.29-2651/bin/tsplugin_until.dylib' (no such file), '/usr/local/lib/tsplugin_until.dylib' (no such file), '/usr/lib/tsplugin_until.dylib' (no such file)
* Debug: trying to load "/usr/local/Cellar/tsduck/3.29-2651/bin/until.dylib"
* Debug: dlopen(/usr/local/Cellar/tsduck/3.29-2651/bin/until.dylib, 0x000A): tried: '/usr/local/Cellar/tsduck/3.29-2651/bin/until.dylib' (no such file), '/usr/local/lib/until.dylib' (no such file), '/usr/lib/until.dylib' (no such file)
* Debug: trying to load "/usr/local/Cellar/tsduck/3.29-2651/lib/tsduck/tsplugin_until.dylib"
* Debug: registering processor plugin "until", status: ok
* Debug: until: debug level set to 1
* Debug: until: ====> tsp -P until --seconds 10
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: ip: setting socket reuse address to 1
* Debug: ip: binding socket to 236.0.0.1:3004
* ip: joining multicast group 236.0.0.1:3004 from local address 100.64.93.246

lelegard, 2022-02-04:
When you write ` -I ip ... --local-address 100.64.93.246`, `tsp` sends an IGMP multicast request from the local interface 100.64.93.246 _only_. By default, on a multi-hosted machine, without further knowledge of where the multicast will come from, the IGMP request is sent on all local interfaces.

On your wireshark screenshot, we see that the request comes from 100.64.88.24. It this the address of your en4 interface? In which case, this would mean that you specified the wrong interface.. However, I may misinterpret the wireshark screenshot.

I would first try without `--local-address` option, just to try on all interfaces.

hygortc, 2022-02-04:
already tested with --local-address the interface that the traffic comes from is en4

ip 100.64.93.246 is on en4 

en4: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
	options=50b<RXCSUM,TXCSUM,VLAN_HWTAGGING,AV,CHANNEL_IO>
	ether 38:c9:86:2a:52:af 
	inet6 fe80::1842:f8d5:962f:1b85%en4 prefixlen 64 secured scopeid 0x4 
	inet 100.64.93.246 netmask 0xffffff00 broadcast 100.64.93.255
	nd6 options=201<PERFORMNUD,DAD>
	media: autoselect (1000baseT <full-duplex,energy-efficient-ethernet>)
	status: active

lelegard, 2022-02-04:
It used to work on my Mac. I can't remember when I did the last multicast reception test but is was probably macOS 10.x. Now, with macOS Monterey 12.1, it no longer works on my Mac either.

I have seen multiple reports on the net saying that the conditions of multicast reception have changed starting with macOS Big Sur 11. Multiple "solutions" are provided but, as usual with that kind of reports, none of them works.

I will try to find a solution but I haven't any right now. Anyone having the _correct_ way of receiving multicast on macOS Big Sur onwards?

hygortc, 2022-02-05:
i am use osx 12

see this 
https://developer.apple.com/documentation/network/nwconnectiongroup

lelegard, 2022-02-05:
More on that multicast issue with macOS Monterey. I started VLC on the same Mac and same UDP stream. VLC does not receive the stream either. Using Wireshark on the same Mac, we see the incoming multicast packets on the right interface. But we do not see any outgoing IGMP membership request, neither from tsp nor VLC.

In your case, TSReader runs on a Windows VM running in the Mac host, right? In that case, the outgoing IGMP request was sent by the Windows VM and is transparently propagated to the outside network through the bridge. What I see is that no IGMP request is generated from macOS, neither from tsp nor VLC.

This is a bit worrying because, usually, "VLC is always right". VLC is very robust and benefits from a community of hundredths of developers. This is why VLC always works, there is always someone to make a specific case work. If VLC does not work either, then we are in trouble.

Does anyone know a native macOS application correctly receiving UDP multicast on macOS Big Sur or Monterey? I mean with open source code which can be examined.

lelegard, 2022-02-05:
> see this https://developer.apple.com/documentation/network/nwconnectiongroup

Hum, this is a high-level Swift API. TSDuck is developed in C++ and uses low-level system calls. I cannot really transpose this documentation into useful information.

lelegard, 2022-02-06:
This really looks like a macOS 11 & 12 issue with multicast, not a TSDuck specific issue.

I did some tests, using a [simple C program sending or receiving multicast](https://github.com/tsduck/tsduck/files/8010182/mctest.zip) that I just wrote and which recreates what the `ip` plugin uses. The same code can be compiled on Linux and macOS. I did some tests between the two and got variable results.

Initial configuration: In the macOS system settings -> security and privacy -> firewall -> advanced options, the options "block all incoming connections" and "stealth mode" are __permanently disabled__. I also tried to temporarily completely deactivate the macOS firewall and the results are the same. So, let's avoid mentioning firewall issues from now on.

I run Wireshark on macOS and on the Linux system. In all cases, the two instances of Wireshark see exactly the same thing. The blocage occurs on macOS, between the application and the probe point of Wireshark. Again, _this is not a firewall issue_ (completely disabled in some tests).

When the multicast receiver runs on Linux, we see the outgoing IGMP "join group" requests. When the receiver runs on macOS, we never see them, even on the Wireshark instance running on macOS.

First observation: the outgoing IGMP "join group" requests are blocked on macOS in all configurations I tested.

When the multicast sender runs on Linux, we see the incoming packets on the Wireshark instance running on macOS.

Second observation: the multicast data packets always reach the macOS interface.

Then, we have two cases:

1. When the Linux system is a physical machine on the same network (a Raspberry Pi in my case), the multicast receiver application running on macOS receives the packets (even though the IGMP "join group" requests were not sent on the network).
2. When the Linux system is a virtual machine on the Mac (Ubuntu 20.10 guest on Parallels Desktop), the multicast receiver application running on macOS __does not receive__ the packets.

I haven't seen any option in the Parallels Desktop settings to alter the multicast on the virtual network. Moreover, multicast between two Linux virtual machines on this virtual LAN works. So, the blocage is not due to the virtual network itself.

This is weird. Any help or suggestion is welcome.

dutchboontje, 2022-05-21:
Hi,
In linux there exists reverse path filtering.
( https://sysctl-explorer.net/net/ipv4/rp_filter/ )

... somebody checked that one already ?