tsduck/tsduck #935: [Q] Error: input plugin: invalid ELF header

ggregoire, 2022-02-04:
Hi!

Opening this issue as a question since it's probably not a bug.

I've been trying to add a scte 35 marker to a file. I've tried different files and always get the same error:

```bash
tsduck  | * Error: test.ts: invalid ELF header
tsduck  | * Error: input plugin test.ts not found
```

I first tried TSDuck through the deb installer, then tried to build it from the sources, but I get the same error.

Any help would be appreciated. :)

Adding the 2 Dockerfiles below and the command I'm passing to TSDuck.

From deb:

```Dockerfile
FROM debian:bullseye

RUN apt-get update && apt-get -y install \
        wget \
        libsrt1.4-openssl \
        libedit2 \
        libpcsclite1 \
        libcurl4

RUN wget -O /tmp/tsduck.deb 'https://github.com/tsduck/tsduck/releases/download/v3.29-2651/tsduck_3.29-2651.debian11_amd64.deb' && \
    dpkg -i /tmp/tsduck.deb

ENTRYPOINT ["tsp"]
```

From sources:

```Dockerfile
FROM debian:bullseye

RUN apt-get update && apt-get install -y \
    git \
    make \
    g++ \
    libedit-dev \
    locales \
    python

RUN sed -i '/en_US.UTF-8/s/^# //g' /etc/locale.gen \
    && locale-gen

ENV LANG en_US.UTF-8
ENV LANGUAGE en_US:en
ENV LC_ALL en_US.UTF-8

RUN git clone https://github.com/tsduck/tsduck.git \
    && cd tsduck \
    && ./scripts/install-prerequisites.sh \
    && make -j10

ENV PATH=/tsduck/bin/release-x86_64-buildkitsandbox:$PATH

ENTRYPOINT ["tsp"]
```

The command (copy/pasted from the doc):

```bash
    command: [
      "--add-input-stuffing",
      "1/10",
      "-I",
      "test.ts",
      "-P",
      "pmt",
      "--service",
      "1010",
      "--add-programinfo-id",
      "0x43554549",
      "--add-pid",
      "600/0x86",
      "-P",
      "spliceinject",
      "--service",
      "1010",
      "--files",
      "splice.xml",
      "--wait-first-batch",
      "-P",
      "filter",
      "--negate",
      "--pid",
      "0x1FFF",
      "-O",
      "file",
      "output.ts"
    ]
```

lelegard, 2022-02-04:
The solution is often in the error message: `Error: input plugin test.ts not found`

Do you specify a plugin named `test.ts` ?

Yes you do:
~~~
      "-I",
      "test.ts",
~~~

You most probably wanted the `file` input plugin, meaning `-I file test.tst`

ggregoire, 2022-02-04:
@lelegard thanks, it works!

Not sure how I missed that, it was indeed in the doc page 431, `tsp … -I file spts.ts …`.

closed 2022-02-04 by ggregoire