tsduck/tsduck #936: [Q] How to inject a SCTE-35 on a specific frame in a TS File?

moving-digital, 2022-02-10:
Hi @lelegard 

Is it possible to inject a SCTE-35 on the first or last frame of a TS File?

Unlike relying on splice insert immediate or on the PTS for Live TS, one would like to inject the SCTE-35 on a particular frame for a TS file.

Is TSP / spliceinject plugin able to do so ?

lelegard, 2022-02-10:
Hi @moving-digital 

This is not possible for several reasons:

- There are various types of SCTE-35 sections on input, immediate or on PTS. Each SCTE-35 section is processed according to its type. There is no frame-synchronous type. If we add some command line option for that, to which input section should we apply the frame-synchronous insertion? All? Immediate only? Why?
- In the `tsp` processing, no new packet can be added. "Insertion" is done by replacing null packets. There is no guarantee that a free null packet will be present righ before a start of frame.
- Frankly speaking, I do not see the use case. Inserted data are as good as they can be processed by a receiver. In most receivers, the a/v demux and data sections demux are separate and work asynchronously. When a data demux processes a section, the immediately following packets (the start of frame in that case) are usually already gone in the hardware a/v decoding pipeline. So, whatever action you would like to trigger on the immediately following frame, it is already too late.

lelegard, 2022-02-10:
For this type of usage, you must send the SCTE-35 markers in advance, and probably several times, to make sure the player is ready to switch. The SCTE-35 markers shall contain the PTS of the switch time. As explained in my previous response, having an immediate SCTE-35 only once, just before the switch frame is probably too late for most receivers.

moving-digital, 2022-02-10:
Hi @lelegard 

Thank you very much. So basically, I got no choice but start to read the PTS contained in the TS file, and as soon as a valid one can be extracted, I must use it to insert the SCTE-35 in the file. Is that correct ?

lelegard, 2022-02-10:
For the start of ads in VOD: read the main program TS, locate the frame where the ad should start, get its PTS, build the splice out command with that PTS. Rewind the file and use `spliceinsert` to insert it (make sure null packets are present or  artificially add/remove them).

For a live TS, compute the hypothetical PTS of the add start (current PTS + time to ad start).

For the end of ad, it depends whether you realign PCR and PTS of the ad video stream or if you switch the player on another input stream. In the first case, knowing the exact duration of the ad, you may generate the splice in at the same time as the splice out. In the second case, the PTS from the ad file will remain the same and you may generate and insert the splice in in advance in the ad file using the same technique as the computation of the splice out command.

moving-digital, 2022-02-17:
Thanks @lelegard 

Is the PTS decimal value accepted by TSDuck, or the hexa value needs to be computed first ?

closed 2022-02-17 by moving-digital