tsduck/tsduck #937: [Q] Small changes to improve the Android support

lars18th, 2022-02-15:
Hi @lelegard ,

I want to point you to this patch for the Termux App: https://github.com/termux/termux-packages/pull/9011/files

FYI, the Termux project is a plain Linux environment for Android smartphones. It not uses root or something different that the regular Android environment, but enables to run Linux binaries with the restricted Android security enforcement. These binaries are packages compiled for it as a Linux distribution. The significant changes are the prefix of the binaries (pointing to the directory of the App) and the small necessary patches to run the binaries using the Android kernel instead of the stock Linux kernel.

Now, the Termux has an official package of TSDuck. And it works without troubles. Hower, some small patches are applied to the source code. These patches are targeting two main areas: the compilation environment and the source code. The reason to open this issue is to ask if you want to incorporate in the main code some of these changes. The idea is to maintain the compatibility of future versions with Termux without huge efforts. I'm not suggesting that you incorporate this platform to the official supported environments. That's not necessary. So I'm only suggesting to try to reduce the potencial necessary changes in the future.

For example, perhaps you want to incorporate the changes that uses the `__ANDROID__` and `__ANDROID_API__` flags. Or update some of the compiling scripts to limit the number of changes.

In any case, it's not my objective to disturb your work. Only to comment that the TSDuck package is now included in this exotic distribution, and if you want then it will more easy to maintain it.

And if you are wondering: The reason to support this environment is not only for fun (great to run `tsp -I ... -O analyze -O drop` in your smartphone in your hand to check anything live!). It's because with it we could develop Android tools that uses TSDuck as a native library. Once we have the Termux library compiled, it's trivial to incorporate the recipe into any Android project. So it's a good way to provide this indirect support for Android. Don't you think so?

Regards.

lelegard, 2022-02-16:
Hi @lars18th 

Thanks for pointing this. I had never heard of this "termux" project before.

Of course, when you write "smartphone", we all know that the target is Android TV set-top boxes. I never liked the idea of TSDuck running on STB because, despite all promises of vertue, this is rarely for legitimate reasons. Anyway, letting this aside, let's concentrate on the technical side.

First, the name of the project is "tsduck", not "libtsduck", considering that you include everything, not only the library. I mention this point because distro maintainers are usually very careful at keeping a consistent naming of projects across package managers. For instance, the two projects "srt" and "librist" were named as such by their respective authors, even though they are all libraries for specific protocols, SRT and RIST, with some basic tools. Homebrew already had an "srt" package but nothing about RIST. When I proposed a package (or "formula" in their jargon) for RIST, I proposed it under the name "rist", just to be consistent with "srt". However, the Homebrew mantainers checked what was that package and they imposed the name "librist" because that was the original name, used everywhere else. The termux maintainers were not that strict, as it seems.

Concerning what seems to be termux-specific build files such as `build.sh` or `*.subpackage.sh`, they are independent from the main TSDuck code. They could be inserted in the main source tree (very much like files to build `.exe`, `.rpm` or `.deb`), but certainly not hanging out at top level. Or you may prefer to provide them independently.

Concerning modifications to Makefiles, this could be integrated but of course not as plain replacement as in the patch. They must be conditioned to some test which triggers the termux environment. What could this be? Some command line option such as `make TERMUX=true`? Or automatically determined from the environment?

There is only one modification I will always refuse here: the removal of compilation option `-Werror`. Warnings are never gratuitous. They say something. Most of the time, they indicate that there is a potential bug under some specific environment or condition. A warning must always be analyzed and understood, the underlying problem must be fixed. If a new compiler or new version of a compiler triggers a new warning, this means that we have more information on a potential issue in the code. TSDuck compiles everywhere with the highest paranoid level of warnings, on all systems, all compilers, and we must continue. No exception.

Modifications in the C++ code can be integrated, maybe with the detection and definition of `TS_ANDROID` in `tsPlatform.h`.

There is one modification which makes me skeptical: the definition of `ioctl_request_t` in `tsPlatform.h`. The template trick was designed to define `ioctl_request_t` as "the type of the second parameter of system call `ioctl()`, whatever it is". This was done to avoid a list of `#ifdef`, to be updated on each new system. Why do you need to force it to `int` here?

lelegard, 2022-02-16:
I wonder what `::execl("@TERMUX_PREFIX@/bin/sh", ...)` means. Do they replace all file system calls with wrappers which dynamically replace the string `"@TERMUX_PREFIX@"`?

Concerning `ioctl_request_t`, there should be no reason to use that trick. The system call `ioctl()` must be part of their restricted set of calls. Otherwise, some other parts of TSDuck wouldn't compile. And if `ioctl()`  is defined in a system header file, the template trick will work. It should not be changed.

I will provide some changes to the C++ code to try to avoid having to patch it.

lars18th, 2022-02-16:
Hi @lelegard ,

> I wonder what ::execl("@TERMUX_PREFIX@/bin/sh", ...) means. Do they replace all file system calls with wrappers which dynamically replace the string "@TERMUX_PREFIX@"?

Yes, because inside the Termux environment only the address `/data/com.termux/` and subdirectories are available for "linux" binaries.

> Concerning ioctl_request_t, there should be no reason to use that trick. The system call ioctl() must be part of their restricted set of calls. Otherwise, some other parts of TSDuck wouldn't [...]

OK. I'll try to discuss about this with the maintainers of Termux.

> I will provide some changes to the C++ code to try to avoid having to patch it.

That sounds very good! 👍