tsduck/tsduck #939: [BUG] TsFile cannot open an existing file

heweisheng, 2022-02-18:
#### Bug description:

I can't use tsfile to open an existing file. I added a line to the source code to try again. I successfully opened this file

#### How to reproduce:

just use TSFile::open(filename,ts::TSFile::WRITE,*report)

#### Expected behavior:

The file can be opened

#### Errors and logs:

~~~
C:\Users\Administrator>G:\TSDuck\bin\tsanalyze.exe "F:\问题视频\第10期下 心儿遭
亲哥送婚纱-蓝光4K.ts"
cannot open F:\ÎŠÌ¢Ê“Æµ\Ú±0ÆšÏ‚ Ð„Ô¢Ç—ç‹é‰´-6â´‹.ts: ç³»ç»Ÿæ‰¾ä¸åˆ°
æŒ‡å®šçš„è·¯å¾„ã€‚
~~~

#### Environment:
 - OS: Win7
 - OS version: sp1
 - TSDuck full version: 3.29.2651.0
 - Installation type: I compile it myself

#### Additional information:

I try to Use CreateFileW api open it success. add line In TSFile::openInternal
~~~
    if (!_filename.empty()) {
        // Non-empty file name, open it.
        _handle = ::CreateFileA(_filename.toUTF8().c_str(), access, shared, NULL, winflags, attrib, NULL);
        if (_handle == INVALID_HANDLE_VALUE) {
            //retry use CreatFileW
            _handle=::CreateFileW((LPCWSTR)_filename.c_str(), access, shared, NULL, winflags, attrib, NULL);
            if (_handle == INVALID_HANDLE_VALUE) {
                const SysErrorCode err = LastSysErrorCode();
                report.log(_severity, u"cannot open %s: %s", { getDisplayFileName(), SysErrorCodeMessage(err) });
                return false;
            }
        }
        // Move to end of file if --append
        if (append_access && ::SetFilePointer(_handle, 0, NULL, FILE_END) == INVALID_SET_FILE_POINTER) {
            const SysErrorCode err = LastSysErrorCode();
            report.log(_severity, u"cannot append to %s: %s", {getDisplayFileName(), SysErrorCodeMessage(err)});
            ::CloseHandle(_handle);
            return false;
        }
    }
~~~

lelegard, 2022-02-18:
Hi @heweisheng 

Thanks for reporting this. It seems that the "A" variants of the Windows file system calls cannot open UTF-8 file names (unlike Linux and macOS) and we need to use the "W" variants (UTF-16). This one is old code. I need to check all potential remaining usages of the "A" variants and fix them.

Directly using the "W" variant is the right way, no need to try the "A" and then the "W" in case of failure.

I will do the cleanup.

heweisheng, 2022-02-21:
Thanks,The TSFile::open, I can use now.

closed 2022-02-21 by lelegard