tsduck/tsduck #940: [FEAT] JSON standard output for SRT plugin statistics

George-, 2022-02-19:
#### Is your feature request related to a problem?
Consuming the statistics using the flags available on the SRT input is difficult in other applications or scripts, e.g. python

#### Proposed solution:
Add a —json flag to the SRT plug-in for TSDuck similar to the flag provided on other plugins. If possible make this available for all stats, including the separate bitrate reporting interval stats.

lelegard, 2022-02-20:
The following option has been added to the `srt` input and output plugins:
~~~
--json-line[='prefix']
    With --statistics-interval or --final-statistics, report the statistics
    as one single line in JSON format. The optional string parameter
    specifies a prefix to prepend on the log line before the JSON text to
    locate the appropriate line in the logs.
~~~

George-, 2022-02-21:
That works perfectly, thanks @lelegard 👍

closed 2022-02-21 by George-