tsduck/tsduck #941: [Q] DTMF signal in stream

dvbshka, 2022-02-22:
Hi! Does tsduck allow to analyze an audio for the appearance of a DTMF code in it? if so, can you give an example pls?

lelegard, 2022-02-22:
Hi @dvbshka 

TSDuck is a transport stream framework. It processes the TS structure, not the inner audio & video encoding and decoding. Detecting DTMF in an audio stream involves some signal processing which is out of scope for TSDuck. So, the answer is no, unfortunately.

Maybe ffmpeg can do that but you need to check.

dvbshka, 2022-02-22:
OK thank you for quick answer.

rowan-eland, 2022-03-16:
> Hi! Does tsduck allow to analyze an audio for the appearance of a DTMF code in it? if so, can you give an example pls?

Hi @dvbshka,

You can check skittyinject from dream-stream here: https://dream-stream.io/skittyinject-gpi-dtmf-scte-35-scte35/

They got an All-In-One-Box DTMF to SCTE35 converter.

dvbshka, 2022-03-16:
hi @rowan-eland 
this solution for inject target. i asked for receive from stream this targets..

rowan-eland, 2022-03-16:
Hi @dvbshka 

Not sure what you mean by "i asked for receive from stream this targets", can you clarify?

DTMF is analog audio cue tone. To analyse such thing, you need to decode the audio first.

dvbshka, 2022-03-16:
hi @rowan-eland 
i need get labels from stream. not insert..

dream-stream-io, 2022-06-06:
Hi @dvbshka 

At dream-stream.io, we have developed skittyinject-dtmf, a state-of-the-art DTMF to SCTE-35 converters, so that OTT Service Providers can implement DAI and SSAI fairly easily. 

It comes in four different versions:
1) Pocket Hardware DTMF to SCTE-35 (External audio module)
2) Pocket Software DTMF to SCTE-35 (Internal audio de-embedder from TSoIP)
3) Industry-Grade Hardware DTMF to SCTE-35 (External audio module)
4) Industry-Grade Software DTMF to SCTE-35 (Internal audio de-embedder from TSoIP)

Link to skittyinject-dtmf: https://dream-stream.io/skittyinject-dtmf-analog-audio-cue-tone-scte35-scte-35/

The Internal audio de-embedder from TSoIP may be what you are looking for, contact us for more info !

SonnyWalkman, 2022-09-30:
I'd suggest looking into Gstreamer for what you want
The  mpegtsdemux plugin is what you need to demux the TS stream into separate Video and Audio streams.

https://gstreamer.freedesktop.org/documentation/mpegtsdemux/index.html?gi-language=c

The simply pipe the audio sink through the dtmfdetect plugin into a fakeaudiosink.

https://gstreamer.freedesktop.org/documentation/spandsp/dtmfdetect.html?gi-language=c
https://gstreamer.freedesktop.org/documentation/debugutilsbad/fakeaudiosink.html?gi-language=c

which effectively emits message 'dtmf-event' on pipeline bus.    Basically each letter detected is emitted in order 1234567890#*