tsduck/tsduck #942: [Q] Develop production level CAS using TSDuck

Hakai-Shin, 2022-02-22:
1. Is it possible to develop production level ecmg, emmg, scrambler, scs, descrambler and other cas related modules using TSDuck development environment and existing plugins as code base ?.
2. Is it possible to make a production level cas by using real ecmg, real emmg and other cas modules as plugins with tsp?

lelegard, 2022-02-22:
Hi @Hakai-Shin 

The answer is "of course not". Developing a CAS is primarily a matter of security, logical security, physical security, mitigating software and hardware attacks and all sorts of threats that CAS vendors have learnt to fight in the last three decades.

TSDuck is a lab tool. It is typically dedicated to test the environment and integration of production CAS systems. I have worked 20 years in the CAS industry. I have primarily developed TSDuck for my own usage to test real CAS products, not to replace them.

TSDuck focuses on the functional part of CAS systems for debug or demo purposes. The value of a real CAS product is in it security. It is quite possible to build a fully functional CAS system using TSDuck out of the box. It works, functionally speaking. It is compatible with all DVB SimulCrypt standards, functionally. But it has zero security and its value as a CAS product is consequently zero.

That being said, using TSDuck as code example for the functional aspects of TS processing for CAS is still possible.

Hakai-Shin, 2022-02-22:
Hello @lelegard 
Thank you for giving your valuable time. I understood that CAS built using TSDuck would lack security aspects and it would be the responsibility of the product developer to look after the security aspect of their CAS. Keeping aside the security aspect, a CAS also needs to be DVB Simulcrypt compliant and I lack finer understanding of DVB Simulcrypt protocols. So my goal is to use TSDuck code base inorder to make DVB Simulcrypt compliant modules in CAS. I have been following ETSI TS 103 197 V1.5.1 (2008-10)[ Digital Video Broadcasting (DVB); Head-end implementation of DVB SimulCrypt] as reference but I don't want to write everything from scratch  for my CAS and I want to use the TSDuck code for interfaces, networking, memory management, etc. Basically what you have said in your last line(i.e.) using TSDuck as code example for the functional aspects of TS processing for CAS.