tsduck/tsduck #945: [BUG] rmorphan removes ISDB global PIDs

APTX, 2022-02-26:
#### Bug description:
The `rmorphan` processing plugin seems to remove global PIDs that are unique to ISDB.

#### How to reproduce:
Using `AAB.ts` from the [stream repository](https://tsduck.io/streams/?name=japan-dttv):
```
$ tsp --verbose --japan -I file AAB.ts -P rmorphan -O file AAB_rmorphan.ts   
* rmorphan: PID 0 (0x0000) is referenced
* rmorphan: PID 1 (0x0001) is referenced
* rmorphan: PID 2 (0x0002) is referenced
* rmorphan: PID 8191 (0x1FFF) is referenced
* rmorphan: PID 16 (0x0010) is referenced
* rmorphan: PID 17 (0x0011) is referenced
* rmorphan: PID 18 (0x0012) is referenced
* rmorphan: PID 19 (0x0013) is referenced
* rmorphan: PID 20 (0x0014) is referenced
* rmorphan: PID 21 (0x0015) is referenced
* rmorphan: PID 22 (0x0016) is referenced
* rmorphan: PID 28 (0x001C) is referenced
* rmorphan: PID 29 (0x001D) is referenced
* rmorphan: PID 30 (0x001E) is referenced
* rmorphan: PID 31 (0x001F) is referenced
* file: initial input bitrate is 17,267,628 b/s
* rmorphan: PID 257 (0x0101) is referenced
* rmorphan: PID 258 (0x0102) is referenced
* rmorphan: PID 259 (0x0103) is referenced
* rmorphan: PID 8136 (0x1FC8) is referenced
* rmorphan: PID 768 (0x0300) is referenced
* rmorphan: PID 785 (0x0311) is referenced
* rmorphan: PID 786 (0x0312) is referenced
* rmorphan: PID 788 (0x0314) is referenced
* rmorphan: PID 6960 (0x1B30) is referenced
* rmorphan: PID 6969 (0x1B39) is referenced
* rmorphan: PID 6970 (0x1B3A) is referenced
* rmorphan: PID 6971 (0x1B3B) is referenced
* rmorphan: PID 96 (0x0060) is referenced
* rmorphan: PID 256 (0x0100) is referenced
* rmorphan: PID 273 (0x0111) is referenced
* rmorphan: PID 274 (0x0112) is referenced
* rmorphan: PID 276 (0x0114) is referenced
* rmorphan: PID 277 (0x0115) is referenced
* rmorphan: PID 6912 (0x1B00) is referenced
* rmorphan: PID 6913 (0x1B01) is referenced
* rmorphan: PID 6914 (0x1B02) is referenced
* rmorphan: PID 6915 (0x1B03) is referenced
* rmorphan: PID 6916 (0x1B04) is referenced
* rmorphan: PID 6917 (0x1B05) is referenced
* rmorphan: PID 6918 (0x1B06) is referenced
```

#### Expected behavior:
I don't think these global PIDs should be removed. Other Global PIDs are not removed.

#### Errors and logs:
[`tsanalyze` of original file from stream repository](https://tsduck.io/streams/japan-dttv/AAB.analyze.txt) (global PIDs section only):
```
|=============================================================================|
|  Global PID's                                                               |
|  TS packets: 35,743, PID's: 13 (clear: 13, scrambled: 0)                    |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Global PID's ................................. C      842,105 b/s  |
|   Subt.  Global PSI/SI PID's (0x00-0x1F) .............. C      104,394 b/s  |
|  0x0000  PAT .......................................... C       15,054 b/s  |
|  0x0001  CAT .......................................... C          141 b/s  |
|  0x0010  NIT .......................................... C        1,507 b/s  |
|  0x0011  SDT/BAT ...................................... C          753 b/s  |
|  0x0012  EIT .......................................... C       86,630 b/s  |
|  0x0014  TDT/TOT ...................................... C          306 b/s  |
|  0x0023  ISDB SDTT .................................... C          259 b/s  |
|  0x0024  ISDB BIT ..................................... C        1,484 b/s  |
|  0x0027  ISDB EIT ..................................... C        9,047 b/s  |
|  0x0028  ISDB SDTT .................................... C           47 b/s  |
|  0x0029  ISDB CDT ..................................... C           23 b/s  |
|  0x0071  ARIB Content Protection EMM (ISDB) ........... C       55,978 b/s  |
|  0x1FFF  Stuffing ..................................... C      670,871 b/s  |
|=============================================================================|
```

Processed file (global PIDs section only):
```
|=============================================================================|
|  Global PID's                                                               |
|  TS packets: 32,906, PID's: 7 (clear: 7, scrambled: 0)                      |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Global PID's ................................. C      775,719 b/s  |
|   Subt.  Global PSI/SI PID's (0x00-0x1F) .............. C      104,455 b/s  |
|  0x0000  PAT .......................................... C       15,064 b/s  |
|  0x0001  CAT .......................................... C          141 b/s  |
|  0x0010  NIT .......................................... C        1,509 b/s  |
|  0x0011  SDT/BAT ...................................... C          754 b/s  |
|  0x0012  EIT .......................................... C       86,681 b/s  |
|  0x0014  TDT/TOT ...................................... C          306 b/s  |
|  0x0071  ARIB Content Protection EMM (ISDB) ........... C            0 b/s  |
|  0x1FFF  Stuffing ..................................... C      671,263 b/s  |
|=============================================================================|
```

#### Environment:
 - OS: Linux
 - OS version: Gentoo
 - TSDuck full version: 3.30-2700
 - Installation type: built from source

lelegard, 2022-03-06:
Hi @APTX 

This is fixed. ISDB and ATSC predefined PID's are now correctly considered as referenced when the corresponding options such as `--isdb`, `--japan`, `--atsc`, `--usa` and the like are specified, either as `tsp` level or at `rmorphan` level.

APTX, 2022-03-07:
Hi @lelegard 

I can confirm the issue is fixed. The ISDB global PID's are no longer removed.

Current `tsanalyze` output for global PID's:
```
|=============================================================================|
|  Global PID's                                                               |
|  TS packets: 33,367, PID's: 12 (clear: 12, scrambled: 0)                    |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Global PID's ................................. C      786,586 b/s  |
|   Subt.  Global PSI/SI PID's (0x00-0x1F) .............. C      104,455 b/s  |
|  0x0000  PAT .......................................... C       15,064 b/s  |
|  0x0001  CAT .......................................... C          141 b/s  |
|  0x0010  NIT .......................................... C        1,509 b/s  |
|  0x0011  SDT/BAT ...................................... C          754 b/s  |
|  0x0012  EIT .......................................... C       86,681 b/s  |
|  0x0014  TDT/TOT ...................................... C          306 b/s  |
|  0x0023  ISDB SDTT .................................... C          259 b/s  |
|  0x0024  ISDB BIT ..................................... C        1,485 b/s  |
|  0x0027  ISDB EIT ..................................... C        9,052 b/s  |
|  0x0028  ISDB SDTT .................................... C           47 b/s  |
|  0x0029  ISDB CDT ..................................... C           24 b/s  |
|  0x0071  ARIB Content Protection EMM (ISDB) ........... C            0 b/s  |
|  0x1FFF  Stuffing ..................................... C      671,263 b/s  |
|=============================================================================|
```

closed 2022-03-07 by APTX