tsduck/tsduck #946: [Q] Regulating packet flow with http input plugin

APTX, 2022-02-26:
When reading from a file, packet flow can be regulated with:
```
tsp -I file -i /path/to/file.ts -P regulate --pcr-synchronous -O <output that needs regulating>
```
The same command with http input does not seem to work:
```
tsp -I http -i http://localhost/file.ts -P regulate --pcr-synchronous -O <output that needs regulating>
```
As it is a local server on the same machine, using HTTP is as fast as reading the file. For some reason this doesn't seem to work and packets are flow a lot faster (as if `regulate` did nothing, but I don't know how to verify this).

In the comments of #71 two options are mentioned, `http --max-queue` (which does not seem to exist for the `http` input plugin but exists for other plugins) and `tsp --buffer-size-mb`. I don't know if they can actually help with this problem.

Is using the `regulate` plugin the correct way to adjust packet flow when the `http` input plugin is used? How could the two mentioned options help?

lelegard, 2022-02-28:
Hi @APTX 

In the two commands, do you point to exactly the same file, just changing the access method ?

If this is the case, this is surprising. Inside the `tsp` plugin execution framework, plugins are independent from each other. The `regulate` plugin, or any other packet processing plugin, works on packets without any knowledge of previous plugins, including input plugins. If the `file` and `http` plugins read exactly the same file, the `regulate` plugin is called with the exact same packets, at equivalent speeds and there should be no difference in behaviour.

APTX, 2022-02-28:
Thank you for your reply.

I tried to verify if the flow is really regulated using [`pv`](http://www.ivarch.com/programs/pv.shtml):
```
tsp -I http -i http://localhost/file.ts -P regulate --pcr-synchronous | pv > /dev/null
```
The reported throughput is the same with both the `http` and `file` plugins. I am not completely certain the file was the same one before, but I made sure of it for this test. I thought I only changed the input plugin, but I will have to check if something else didn't change as well.

It seems that the command is indeed correct, so I will consider the question answered.

closed 2022-02-28 by APTX