tsduck/tsduck #948: [BUG] http input plugin loses data

APTX, 2022-03-01:
#### Bug description:
When the `http` input plugin is used not all data might end up in the `tsp` pipeline.

#### How to reproduce:
Using [`482000000.ts`](https://tsduck.io/streams/uk-freeview/482000000.ts) from the [streams repository](https://tsduck.io/streams/?name=uk-freeview):
```bash
# 1. Download the TS file to $PWD
# 2. Start HTTP server. This will start serving files in $PWD over http on port 9001
python -m http.server 9001

# Open a new shell
# 3. First run tsp with file input
tsp --verbose -I file -i 482000000.ts -O play # The complete TS plays in a loop
tsp --verbose -I file -i 482000000.ts -P regulate --pcr-synchronous -O play # Same as above

# 4. Compare to running with http input
tsp --verbose -I http -i http://localhost:9001/482000000.ts -O play # Only a small part of the ts plays in a loop
tsp --verbose -I http -i http://localhost:9001/482000000.ts -P regulate --pcr-synchronous -O play # Same as above
```
Initially I only saw issues with `regulate`, but when trying to find a reproduction example I found that there are also problems without `regulate`.
The `play` output plugin does not need regulation, but should still work with it.
The `play` output plugin is used as it is easy to see that not all of the TS data makes it to the output.

It's also possible to see this without repeat mode (`-i` option):
```bash
tsp --verbose -I file -i 482000000.ts -P regulate --pcr-synchronous -O play # The complete TS plays
tsp --verbose -I http http://localhost:9001/482000000.ts -P regulate --pcr-synchronous -O play # tsp exits before the complete the TS plays
tsp --verbose -I fork 'curl -s http://localhost:9001/482000000.ts' -P regulate --pcr-synchronous -O play # The complete TS plays
```

#### Expected behavior:
All commands should behave the same way.

#### Errors and logs:
```
$ tsp --verbose -I http -i http://localhost:9001/482000000.ts -P regulate --pcr-synchronous -O play
* regulate: minimum wait: 10,000,000 nano-seconds, using 50,000,000 ns
* http: downloading from http://localhost:9001/482000000.ts
* http: MIME type: unknown, expected size: 202762700 bytes
* http: unknown initial input bitrate
* regulate: using PID 0x0065 (101) for PCR reference
* http: downloading from http://localhost:9001/482000000.ts
* http: MIME type: unknown, expected size: 202762700 bytes
* http: downloading from http://localhost:9001/482000000.ts
* http: MIME type: unknown, expected size: 202762700 bytes
* http: downloading from http://localhost:9001/482000000.ts
* http: MIME type: unknown, expected size: 202762700 bytes
* Warning: regulate: out of sequence PCR, maybe source was cycling, restarting regulation
* http: downloading from http://localhost:9001/482000000.ts
* http: MIME type: unknown, expected size: 202762700 bytes
* Warning: regulate: out of sequence PCR, maybe source was cycling, restarting regulation
* Warning: regulate: out of sequence PCR, maybe source was cycling, restarting regulation
* Warning: regulate: out of sequence PCR, maybe source was cycling, restarting regulation
* http: downloading from http://localhost:9001/482000000.ts
* http: MIME type: unknown, expected size: 202762700 bytes
* Warning: regulate: out of sequence PCR, maybe source was cycling, restarting regulation
* http: downloading from http://localhost:9001/482000000.ts
* http: MIME type: unknown, expected size: 202762700 bytes
* http: downloading from http://localhost:9001/482000000.ts
* http: MIME type: unknown, expected size: 202762700 bytes
* http: downloading from http://localhost:9001/482000000.ts
* http: MIME type: unknown, expected size: 202762700 bytes
* Warning: regulate: out of sequence PCR, maybe source was cycling, restarting regulation
* Warning: regulate: out of sequence PCR, maybe source was cycling, restarting regulation
* Warning: regulate: out of sequence PCR, maybe source was cycling, restarting regulation
* http: downloading from http://localhost:9001/482000000.ts
* http: MIME type: unknown, expected size: 202762700 bytes
* Warning: regulate: out of sequence PCR, maybe source was cycling, restarting regulation
* http: downloading from http://localhost:9001/482000000.ts
* http: MIME type: unknown, expected size: 202762700 bytes
* http: downloading from http://localhost:9001/482000000.ts
* http: MIME type: unknown, expected size: 202762700 bytes
* http: downloading from http://localhost:9001/482000000.ts
* http: MIME type: unknown, expected size: 202762700 bytes
* Warning: regulate: out of sequence PCR, maybe source was cycling, restarting regulation
* tsp: user interrupt, terminating...
```
VLC output was removed

#### Environment:
 - OS: Linux
 - OS version: Gentoo 
 - TSDuck full version: 3.30-2700
 - curl version: 7.81.0 (x86_64-pc-linux-gnu) libcurl/7.81.0 OpenSSL/1.1.1m zlib/1.2.11 nghttp2/1.47.0
 - Installation type: built from source

#### Additional information:
I initially thought this was a problem with regulation (#946)

APTX, 2022-03-02:
Hi @lelegard,

I ran the test:
```bash
# The requested test
tsp -I http http://localhost:9001/482000000.ts -O file tmp.ts && cmp tmp.ts 482000000.ts ; echo $?
# Prints:
# cmp: EOF on tmp.ts after byte 16777120, in line 85646
# 1
# Note: 16777120 bytes (exactly 16MiB) is 89240 packets, around 8% of 482000000.ts

# Check if the server serves the file correctly
curl -s http://localhost:9001/482000000.ts > tmp.ts && cmp tmp.ts 482000000.ts ; echo $?
# Prints:
# 0

# Previous test, but inside tsp pipeline
tsp -I fork 'curl -s http://localhost:9001/482000000.ts' -O file tmp.ts && cmp tmp.ts 482000000.ts ; echo $?
# Prints:
# 0
```

I checked the server output.
The first command:
```
127.0.0.1 - - [02/Mar/2022 19:54:36] "GET /482000000.ts HTTP/1.1" 200 -
----------------------------------------
Exception occurred during processing of request from ('127.0.0.1', 60336)
Traceback (most recent call last):
  File "/usr/lib/python3.9/socketserver.py", line 683, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/lib/python3.9/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python3.9/http/server.py", line 653, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/lib/python3.9/socketserver.py", line 747, in __init__
    self.handle()
  File "/usr/lib/python3.9/http/server.py", line 427, in handle
    self.handle_one_request()
  File "/usr/lib/python3.9/http/server.py", line 415, in handle_one_request
    method()
  File "/usr/lib/python3.9/http/server.py", line 660, in do_GET
    self.copyfile(f, self.wfile)
  File "/usr/lib/python3.9/http/server.py", line 860, in copyfile
    shutil.copyfileobj(source, outputfile)
  File "/usr/lib/python3.9/shutil.py", line 208, in copyfileobj
    fdst_write(buf)
  File "/usr/lib/python3.9/socketserver.py", line 826, in write
    self._sock.sendall(b)
ConnectionResetError: [Errno 104] Connection reset by peer
----------------------------------------
```
The other two commands:
```
127.0.0.1 - - [02/Mar/2022 19:56:08] "GET /482000000.ts HTTP/1.1" 200 -
127.0.0.1 - - [02/Mar/2022 19:56:11] "GET /482000000.ts HTTP/1.1" 200 -
```

When I encountered the problem initially, I was using [nginx](https://nginx.org/), but testing with the python server is much easier to set up.

lelegard, 2022-03-02:
Hi @APTX 

I did the same test as you and I ran into a very strange situation. All tests are done on a Ubuntu 21.10 VM. All files are local (HTTP server on same VM and using http://localhost to access the files).

First test: I used the same python server as you and got the same result as you: the output file is truncated at 16 MB. Since this is the default size of the global `tsp` packet buffer, I tried various other, larger or smaller, buffer sizes. The output file is truncated at the buffer size. Approximately. The only difference is that I do not have any Python stack trace and no "connection reset by peer", just the normal single log line.

Second test: I used Apache as server. Then, the same `tsp` command works fine. All packets are read, regardless of the global buffer size.

In practice, most servers are Apache or IIS (Windows), sometimes nginx. This kind of issue was never raised.

From a client perspective, before the buffer is full, the input function is repeatedly called in the input thread. Once the buffer is full, the input thread is suspended until there is some free space in the buffer. At that point, the input function is called again, etc. It seems that the small wait time makes a difference. When the server is Apache, the wait time is transparent. When the server is Python, the connection terminates.

I will try with Wireshark to see which side resets the connection.

APTX, 2022-03-06:
Hi @lelegard 

I tried the first test with wireshark and it looks like the `http` input plugin is closing the connection:
```
745	0.026944095	127.0.0.1	127.0.0.1	TCP	66	51086 → 9001 [RST, ACK] Seq=86 Ack=22404195 Win=2270720 Len=0 TSval=1867951988 TSecr=1867951987
```
A similar reset also happens with nginx (the server is set up with HTTPS and http2 leading to more complex wireshark logs). I do not have any Apache or IIS servers in my environment to test. I also do not want to have a dedicated HTTP server just for TSDuck.

On the client side, I tried a few commands that can download a file over HTTP. There is no reset with any of them and the file always matches the source. I used the python HTTP server for all the tests:
```bash
# Curl command (built from the same libcurl TSDuck uses). This is the same as the example in my previous comment
tsp -I fork 'curl -s http://localhost:9001/482000000.ts' -O file tmp.ts && cmp tmp.ts 482000000.ts ; echo $?
# Prints:
# 0

# wget
tsp -I fork 'wget -q -O - http://localhost:9001/482000000.ts' -O file tmp.ts && cmp tmp.ts 482000000.ts ; echo $?
# Prints:
# 0

# dog. The sed command removes the HTTP response headers
tsp -I fork 'dog http://localhost:9001/482000000.ts | sed "1,/^\r$/d"' -O file tmp.ts && cmp tmp.ts 482000000.ts ; echo $?
# Prints:
# 0

# netcat. Raw HTTP request, same sed command as above
tsp -I fork 'printf "GET /482000000.ts HTTP/1.1\r\nHost: 127.0.0.1\r\n\r\n" | nc 127.0.0.1 9001 | sed "1,/^\r$/d"' -O file tmp.ts && cmp tmp.ts 482000000.ts ; echo $?
# Prints:
# 0
```
I also tried with `-P regulate --pcr-synchronous` to test if suspending the input thread affects the result. Wireshark shows that TCP retransmission can happen, but it does not change the result. The commands work as before, but take a minute to complete.

Downloading the file with a web browser (chrome, firefox) works too.

Playing the file over HTTP works as well:
```bash
vlc http://localhost:9001/482000000.ts
mpv http://localhost:9001/482000000.ts
```

Except for the TSDuck `http` input plugin, I could not find any other HTTP client that would not work correctly with the python HTTP server.

lelegard, 2022-03-06:
Hi @APTX 

I have seen the same thing on my side. The problem is not really in the `http` input plugin. It just calls libcurl. All interactions, including terminating the session, are performed by libcurl. There is some issue in the way libcurl is used ("multi" mode) and how it reacts to the Python HTTP server. On Windows, using the WinInet native library instead of libcurl, the plugin works fine, even with the Python HTTP server. On Linux, the same plugins which fails when receiving from the Python server succeeds when receiving from Apache. 

Installing Apache is not really complicated. Just install package `httpd` or `apache2`, depending on the distro, and that's all. It took me just a minute to get the test ready on Ubuntu. Maybe it is more difficult on Gentoo. I have 10 different Linux distros on 10 different VM's and Gentoo is by far the one where everything is more complicated...

I need to investigate, it may take some time.

lelegard, 2022-03-06:
OK, it was faster than expected. I think I have found and fixed the issue. At least, it works on my side now with the Python server. The problem was seen when the server sent very small chunks of data in the middle of the stream. Production servers such as Apache don't do that. This is why it never happened. The Python server being a test tool (it is documented as such), small chunks were possible.

Please test with the latest repo and let us know if it works for you now.

closed 2022-03-07 by APTX