tsduck/tsduck #950: [BUG] t2mi plugin incorrectly handles streams with duplicate packets

deeptho, 2022-03-02:
#### Bug description:

The t2mi plugin (tsp -P t2m) incorrectly handles duplicate transport stream packets.
This results in a corrupt output stream. 

#### How to reproduce:

I have captured a stream from satellite (13W or 14W - not sure) which contains 
a t2mi stream with PID=4096. This stream contains such duplicate packets. 
I can provide that stream if needed.
Decoding the stream as follows:
```cat test.ts|tsp -P t2mi --pid 4096 --plp > out.ts```
leads to a corrupt out.ts video stream (video completely broken)

#### Expected behavior:

Correct stream 

#### Errors and logs:

~~~
[be sure to surround verbatim log lines with "~~~" lines]
~~~

#### Environment:
 - OS:  Linux
 - OS version:  Fedora 34
 - TSDuck full version:  latest master:  commit 80548f09822845107d5f9d0bc41b53f236b3a868 
 - Installation type: 
 git

#### Additional information:

The problem occurs in  ts::T2MIDemux::feedPacket
The test "Check if we loose synchronization." is wrong. It checks the discontinuity indicator
and the continuity counter, but fails to take into account that the mpeg standard allows for
repeating a transport packet exactly once (original + one copy). This leads to two successive
packets having the same continuity counter value.

Replacing the test by 
~~~
    // Check if we loose synchronization.
    if (pc->sync && (pkt.getDiscontinuityIndicator() || (
										 pkt.getCC() != ((pc->continuity + 1) & CC_MASK)
										 && pkt.getCC() != (pc->continuity & CC_MASK)
											 )
					)) {
			pc->lostSync();
    } 
~~~
solves the problem. Technicaly, this fix is also incorrect, because it would accept 
three or more consecutive identical transport stream packets, whereas that would  indicated an invalid stream. In practice, this is good enough.

lelegard, 2022-03-03:
Hi @deeptho 

Thank you for this deep analysis. The problem seems real and I just fixed it. Could you please test the latest state of the repo with your TS?

It would be interesting to get this TS. May I add it in the stream repository to add a TS with a new behaviour (T2MI with duplicated packets)? Since this is a capture from a satellite, the TS must be public.

deeptho, 2022-03-03:
Where should I put the TS?

deeptho, 2022-03-03:
I have tested the latest version on the stream and can confirm that it works.

lelegard, 2022-03-03:
> Where should I put the TS?

Any file transfer site such as wetransfer.com should be ok.

deeptho, 2022-03-04:
I have uploaded the stream here:
https://0x0.st/oPeq.ts

Note that this file will auto-erase after some time.

lelegard, 2022-03-05:
Hi @deeptho 

Thanks for the file. In fact, the problem does not come from "duplicated packets". It comes from packets without a payload. 

Example, using `tsdump` on the file:

~~~
* Packet 2,025
  ---- TS Header ----
  PID: 4096 (0x1000), header size: 4, sync: 0x47
  Error: 0, unit start: 0, priority: 0
  Scrambling: 0, continuity counter: 7
  Adaptation field: no (0 bytes), payload: yes (184 bytes)
  ---- Full TS Packet Content ----
  47 10 00 17 2D 1E 07 5C F9 5D EE ED 4F 95 8C 6B CC 2C 6B B1 41 F4 EC 00
  F7 2D FB DD 87 14 28 BA CD 16 74 0C D9 ED 4D 55 9F 67 0F E8 E2 AA 79 06
  9F 7D 58 1D 1A 01 70 33 45 F8 E0 58 23 A7 68 2C 5A 46 B3 0A 40 4A 2B 72
  A3 FE BF 5A 17 0A 80 53 93 72 F7 F1 76 06 5C 00 0C EB CA 4E 20 D5 D4 06
  0D 34 04 C9 A0 04 A5 E4 54 AF 31 E6 C8 8D AB AE 9F D0 B0 A7 4B 81 0B 0F
  62 25 B7 79 BB AB 9A B3 86 EC 57 96 04 28 12 1D 2B C8 95 C2 9B 6A AF 85
  6C EB A7 32 F5 48 D2 BA 7F 8B 75 A4 D5 C9 CA CE 15 4F 96 93 1F FF 10 DF
  E0 E1 E2 E3 E4 E5 E6 E7 E8 E9 EA EB EC ED EE EF F0 F1 F2 F3

* Packet 2,026
  ---- TS Header ----
  PID: 4096 (0x1000), header size: 188, sync: 0x47
  Error: 0, unit start: 0, priority: 0
  Scrambling: 0, continuity counter: 7
  Adaptation field: yes (184 bytes), payload: no (0 bytes)
  Discontinuity: 0, random access: 0, ES priority: 0
  PCR: 0x076CEF4329C
  ---- Full TS Packet Content ----
  47 10 00 27 B7 10 32 B1 0C 07 81 08 FF FF FF FF FF FF FF FF FF FF FF FF
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
~~~

The second packet is not a duplicated of the first one, it is a packet without payload.

Instead of inserting PCR's in regular packets of the PID, the broadcaster inserts empty packets (no payload) containing only the PCR. Since there is no payload, the continuity counter shall not be incremented according to ISO 13818-1.

This method is slightly inefficient but perfectly valid.

In practice, the fix which ignores packets with a duplicated CC works but is not really sufficient. I will add a explicit test for payload presence.

lelegard, 2022-03-05:
In the TSDuck test suite, test-109 has been added from a very small extract of your file as non-regression test for this issue.

closed 2022-09-23 by lelegard