tsduck/tsduck #951: [BUG] Suspicous code in pointer field handling in t2mi plugin

deeptho, 2022-03-02:
#### Bug description:

While investigating another bug, I discovered some suspicius code in tsT2MIDemux.cpp.
I cannot be sure that it is incorrect, and I do not have a stream that indicates that the code 
is incorrect

~~~
  if (!pc->sync) {
            pc->sync = true;
            // Skip end of previous packet, before retrieving synchronization.
            data += pf;
            size -= pf;
        }
~~~

However, I see two potential issues
1. if a pointer field is present, then the current TS packet contains the start of a t2mi packet, but could theoretically
also contain the end of the previous t2mi packet (instead of stuffing). I do not know t2mi well enough to be sure that this can actually occur, but if it does, then the current code will forget to append the first part of the TS packet to last t2mi output packet.

2. skipping to the start f the packet after a non-zero pointer field pf is only done when snc has been lost (!pc->sync), but shouldn't this be done always?

lelegard, 2022-03-03:
Hi @deeptho 

Thanks for your analysis of the code. However, I thnk that the current code is correct.

> if a pointer field is present, then the current TS packet contains the start of a t2mi packet, but could theoretically also contain the end of the previous t2mi packet (instead of stuffing).

Yes, indeed. But we are in a branch where synchronization is lost (` if (!pc->sync)`), the previous T2MI packet is broken, truncated or inconsistent. This means that the end of this packet shall be discarded. The fact that the TS packet has a PUSI indicator is a chance for the demux to resynchronize at the beginning of a new T2MI packet. This is why we must skip the data before the pointer field for a fresh restart.

>  skipping to the start f the packet after a non-zero pointer field pf is only done when snc has been lost (!pc->sync), but shouldn't this be done always?

No, it shouldn't. When synchronization is not lost, we must keep the end of the previous T2MI packet. Note that if this specific part of the code was wrong, most T2MI demux would fail. Which is not the case.

deeptho, 2022-03-03:
It seems you are right. Thanks for explaining.

closed 2022-03-03 by deeptho