tsduck/tsduck #952: [BUG] Unable to compile on raspberry

lucdig, 2022-03-07:
#### Bug description:
Unable to compile on raspberry

#### How to reproduce: as root
apt install git g++ dpkg-dev doxygen dos2unix graphviz curl pcscd libpcsclite-dev libcurl4 libcurl4-openssl-dev
git clone https://github.com/tsduck/tsduck.git
cd tsduck
make install
source src/tstools/release-arm/setenv.sh

#### Expected behavior:
installation complete

#### Errors and logs:

~~~
base/app/tsEditLine.cpp:40:10: fatal error: editline/readline.h: No such file or directory
   40 | #include <editline/readline.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [../../Makefile.inc:506: /root/tsduck/bin/release-arm-raspetta/objs-libtsduck/tsEditLine.o] Error 1
make[1]: *** [Makefile:41: install] Error 2
make: *** [Makefile:155: install] Error 2

~~~

#### Environment:
uname -a: 

 - OS: 
 - OS version: 
 - TSDuck full version: 
 - Installation type: 

#### Additional information:

lucdig, 2022-03-07:
uname -a:
Linux raspetta 5.10.92+ #1514 Mon Jan 17 17:35:21 GMT 2022 armv6l GNU/Linux

lelegard, 2022-03-07:
Hi @lucdig 

As the error message implies, you need to install the package `libedit-dev` (if you are using a Raspbian distro). Please read the [TSDuck building instructions here](https://tsduck.io/doxy/building.html). Especially, after cloning or pulling the repo, the script `scripts/install-prerequisites.sh` ensures that all prerequisites are installed. If you do not use any of the 13 distros which are supported by that script, reading it will give you a list of what to install.

Note that `src/tstools/release-arm/setenv.sh` has been removed a long time ago. Please read the updated [building instructions](https://tsduck.io/doxy/building.html).

lucdig, 2022-03-07:
thanks a lot

closed 2022-03-07 by lucdig