tsduck/tsduck #953: [BUG] after running the scripts/install-prerequisites.sh on a Raspberry pi, the build failed

lucdig, 2022-03-07:
#### Bug description:
after running the scripts/install-prerequisites.sh on a Raspberry pi, the build failed

#### How to reproduce:
git clone https://github.com/tsduck/tsduck.git
cd tsduck
scripts/install-prerequisites.sh
make install

#### Expected behavior:

#### Errors and logs:

~~~
  [JAVAC] src/io/tsduck/AbstractAsyncReport.java
Error occurred during initialization of VM
Server VM is only supported on ARMv7+ VFP
make[3]: *** [Makefile:65: /root/tsduck/bin/release-arm-raspetta/java/io/tsduck/AbstractAsyncReport.class] Error 1
make[2]: *** [Makefile:187: install-tools] Error 2
make[1]: *** [Makefile:41: install] Error 2
make: *** [Makefile:155: install] Error 2

~~~

#### Environment:
Linux raspetta 5.10.92+ #1514 Mon Jan 17 17:35:21 GMT 2022 armv6l GNU/Linux
 - OS: 
 - OS version: 
 - TSDuck full version: 
 - Installation type: 

#### Additional information:

lucdig, 2022-03-07:
# tsp -I dvb -a 0 --delivery-system DVB-T --bandwidth 8-MHz --frequency 842000000 --guard-interval 1/32 --hierarchy none --high-priority-fec 2/3 --low-priority-fec none --modulation 64-QAM --spectral-inversion auto --transmission-mode 8K -P analyze --normalize -O drop
tsp: symbol lookup error: tsp: undefined symbol: TSDUCK_LIBRARY_VERSION_3_29_2651

lelegard, 2022-03-07:
Hi @lucdig 

The error message is clear: `Server VM is only supported on ARMv7+ VFP`. Your `uname` outputs `armv6l`. So, this is probably not the right Java package for your system. You need another Java package for Arm v6 systems.

My Raspberry Pi system is:
~~~
$ cat /sys/firmware/devicetree/base/model; echo
Raspberry Pi 3 Model B Rev 1.2
$ uname -a
Linux zozoberry 5.10.52-v7+ #1441 SMP Tue Aug 3 18:10:09 BST 2021 armv7l GNU/Linux
~~~

So, this is an Arm v7 CPU. I have no Arm v6 Raspberry Pi and I have no idea which Java package you need.

You must check which Java package is required on your system, remove the current one (for v7+ only) and install the right one. Please provide the right package name for Arm v6 and I will update the `install-prerequisites.sh` script.

Please also specific which distro you use. Mine is Raspbian but yours is probably different. My system is up to date (no update available) but your kernel is slightly more recent, 5.10.92 instead of 5.10.52. So your distro must be different.

The other error `tsp: symbol lookup error` is simply due to the incomplete build. The TSDuck library was built (version 3.30-xxx) but not the commands such as `tsp`. As a consequence, you use the previously installed `tsp` (version 3.29) with the new library (3.30) and there is a mismatch.

lucdig, 2022-03-08:
Hi, my hardware is:

~~~
# cat /sys/firmware/devicetree/base/model; echo
Raspberry Pi Zero W Rev 1.1
~~~

About 2 years ago I achieved to build tsduck on this hardware

lelegard, 2022-03-08:
> About 2 years ago I achieved to build tsduck on this hardware

You probably built it just before the Java API had been added. TSDuck is developed in C++ but it has Python and Java API's for applications in these languages. These API's were introduced about two years ago.

You have two solutions:
- Find and install the right java packages for Arm v6.
- If you do not need java, completely remove the java packages. In that case, considering that there is no java support on the system, TSDuck will be rebuilt without java support.

lucdig, 2022-03-08:
Hi @lelegard , I don't need java, how can I build without the java packages?

Thanks

lelegard, 2022-03-09:
As I previously wrote, just remove all Java development packages. When the command `javac` is not found (this is the Java compiler), TSDuck builds without Java support. Starts with `dpkg -l '*jdk*'` and removed all installed ones.

lucdig, 2022-03-09:
I removed all the packages listed with `dpkg -l '*jdk*'`, I ran again `make install` and the build now is ok. Thank you very much

closed 2022-03-09 by lucdig