tsduck/tsduck #954: [FEAT] splicemonitor plugin: option to disable/discard "splice_null" on demand

moving-digital, 2022-03-11:
#### Is your feature request related to a problem?
The feature request is related to an improvement of `splicemonitor` plugin.
Not a problem, but performances issue due to `splice_null` command inserted every 1 second in the TS to monitor, causing throttling for the CPU receiving too much information.

#### Proposed solution:
Would it be possible to add an option to disable/discard "splice_null" on demand ?

#### Alternatives:
No alternative.

#### Additional information:

lelegard, 2022-03-12:
Hi @moving-digital 

The option `--select-commands` has been added.

Example: to display insert, schedule, time signal only, use `--select-commands 4-6`.

moving-digital, 2022-03-15:
So fast ! Thanks a lot, @lelegard  !

closed 2022-03-15 by moving-digital