tsduck/tsduck #955: [FEAT] Bind DVB adapter by MAC address, instead of adapterN & device name

morpig, 2022-03-14:
#### Is your feature request related to a problem?

Currently, tsduck only accept DVB adapter values based on adapter number from /dev/dvb/adapterN & device name based on OS.

Not feasible for multiple DVB adapters, especially running the same models. 

If there is restart/reset of OS, then adapterN would be randomized based on "first come first serve"

```
tslsdvb
0: /dev/dvb/adapter0 ("TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T, TBS-Tech TBS 5521 2.00", DVB-T, DVB-T2, ISDB-T, DVB-C, 2)
1: /dev/dvb/adapter0:1:1:1 ("TurboSight TBS 5520SE DVB-S/S2/S2X, TBS-Tech TBS 5521 2.00", DVB-S, DVB-S2, DSS)
2: /dev/dvb/adapter1 ("TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T, TBS-Tech TBS 5521 2.00", DVB-T, DVB-T2, ISDB-T, DVB-C, 2)
3: /dev/dvb/adapter1:1:1:1 ("TurboSight TBS 5520SE DVB-S/S2/S2X, TBS-Tech TBS 5521 2.00", DVB-S, DVB-S2, DSS)
4: /dev/dvb/adapter2 ("TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T, TBS-Tech TBS 5521 2.00", DVB-T, DVB-T2, ISDB-T, DVB-C, 2)
5: /dev/dvb/adapter2:1:1:1 ("TurboSight TBS 5520SE DVB-S/S2/S2X, TBS-Tech TBS 5521 2.00", DVB-S, DVB-S2, DSS)
6: /dev/dvb/adapter3 ("TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T, TBS-Tech TBS 5521 2.00", DVB-T, DVB-T2, ISDB-T, DVB-C, 2)
7: /dev/dvb/adapter3:1:1:1 ("TurboSight TBS 5520SE DVB-S/S2/S2X, TBS-Tech TBS 5521 2.00", DVB-S, DVB-S2, DSS)
8: /dev/dvb/adapter4 ("TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T, TBS-Tech TBS 5521 2.00", DVB-T, DVB-T2, ISDB-T, DVB-C, 2)
9: /dev/dvb/adapter4:1:1:1 ("TurboSight TBS 5520SE DVB-S/S2/S2X, TBS-Tech TBS 5521 2.00", DVB-S, DVB-S2, DSS)
10: /dev/dvb/adapter5 ("TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T, TBS-Tech TBS 5521 2.00", DVB-T, DVB-T2, ISDB-T, DVB-C, 2)
11: /dev/dvb/adapter5:1:1:1 ("TurboSight TBS 5520SE DVB-S/S2/S2X, TBS-Tech TBS 5521 2.00", DVB-S, DVB-S2, DSS)
```

#### Proposed solution:

DVB Adapters (mostly TBS cards), have unique MAC address attached to them, and dvb-usb is able to identify them.

```
[   13.317656] dvb-usb: TurboSight TBS 5520SE successfully initialized and connected.
[   13.317719] dvb-usb: found a 'TurboSight TBS 5520SE' in cold state, will try to load a firmware
[   13.317721] dvb-usb: downloading firmware from file 'dvb-usb-id5520se.fw'
[   13.474664] dvb-usb: found a 'TurboSight TBS 5520SE' in warm state.
[   13.474812] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[   13.705094] dvb-usb: MAC address: 00:22:ab:c0:63:5e

[   13.754171] dvb-usb: TurboSight TBS 5520SE successfully initialized and connected.
[   13.754191] dvb-usb: found a 'TurboSight TBS 5520SE' in cold state, will try to load a firmware
[   13.754193] dvb-usb: downloading firmware from file 'dvb-usb-id5520se.fw'
[   13.910685] dvb-usb: found a 'TurboSight TBS 5520SE' in warm state.
[   13.910831] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[   14.139594] dvb-usb: MAC address: 00:22:ab:c0:62:a3
[   14.204795] dvb-usb: TurboSight TBS 5520SE successfully initialized and connected.
```

Looking forward for the additional feature to bind adapters by ``--mac 00:22:ab:c0:62:a3``

#### Alternatives:

#### Additional information:

lelegard, 2022-03-15:
Hi @morpig 

DVB adapters usually have no MAC address. A MAC address is used in network adapters because each network adapter needs to be addressed from outside. DVB adapters are broadcast receivers and do not need any MAC address.

You see a MAC address in the `dmesg` output. This means that the driver for this device finds some device-specific MAC address. But this is just very specific to this TBS device. I haven't seen this on any other DVB device (at least the ones I used). This is not a standard property of DVB devices and cannot be used in TSDuck.

You may try to automate the detection of the appropriate device using a script, then determine the corresponding `/dev/dvb/adapterX` and use it in the `tsp` command. You may have a look at the description of the device in `/sys/devices`. Start with:
~~~
find /sys/devices/ -path '*/dvb*.frontend*/device'
~~~
These are symbolic links to directories which contain the device characteristics. In practice, the pseudo-files here are not all standard. The driver may place device-specific information here. In the case of the TBS driver, you may find here the MAC or pseudo-MAC address.

In all cases, this information is not standard, device-specific or driver-specific,  and hard to integrate into TSDuck. However, in your specific case, knowing your devices, you may write a script to locate each device.

morpig, 2022-03-15:
Thanks for the response! 

Yes I do understand that MAC address is used for network adapters, this is also a first for me to see DVB adapters using MAC address. I used MyGica DVB devices before, and it does not have any MAC address attached to it.

I found out that MAC address does indeed being supported by the dvb-usb in main kernel, and being used by other manufacturers too other than TBS (Hauppauge, DVBSky, etc)

Looking at ``/sys/devices``, it's useless too.

```
find /sys/devices/ -path '*/dvb*.frontend*/device'
/sys/devices/pci0000:00/0000:00:01.2/0000:01:00.0/usb1/1-3/dvb/dvb1.frontend1/device
/sys/devices/pci0000:00/0000:00:01.2/0000:01:00.0/usb1/1-3/dvb/dvb1.frontend0/device
/sys/devices/pci0000:00/0000:00:01.2/0000:01:00.0/usb1/1-4/dvb/dvb3.frontend1/device
/sys/devices/pci0000:00/0000:00:01.2/0000:01:00.0/usb1/1-4/dvb/dvb3.frontend0/device
/sys/devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb3/3-4/dvb/dvb5.frontend0/device
/sys/devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb3/3-4/dvb/dvb5.frontend1/device
/sys/devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb3/3-2/dvb/dvb2.frontend0/device
/sys/devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb3/3-2/dvb/dvb2.frontend1/device
/sys/devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb3/3-3/dvb/dvb4.frontend0/device
/sys/devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb3/3-3/dvb/dvb4.frontend1/device
/sys/devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb3/3-1/dvb/dvb0.frontend0/device
/sys/devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb3/3-1/dvb/dvb0.frontend1/device
```

The software (that I know) that can bind DVB adapters based on MAC address is Astra Cesbo, but I prefer using open source software instead of proprietary licenses.

[image]

If you have any ideas on how to bind (OS side) based on mac address to ``/dev/dvb/adapterX``, please let me know..

lelegard, 2022-03-15:
> Looking at /sys/devices, it's useless too.

On the contrary. Each line is a symbolic link to a directory. The content of the directory (and subdirectories) is a set of pseudo files. The list of files depends on the driver. Some are standard (`idVendor`, `idProduct`), some are optional such as `serial`. The latter is where the serial number is read from. Please explore those pseudo files in your case and try to find one containing the MAC address. This will be a non-standard file but at least you may get a place where to look for a known MAC address.

For instance, assuming that the directory contains a pseudo file named `mac` containing the MAC address in the form `nn:nn:nn:nn:nn:nn`, the following command should build the TSDuck device name from a known MAC address `NN:NN:NN:NN:NN:NN`:
~~~
DEV=$(grep -l /sys/devices/*/*/*/usb*/*/dvb/dvb*.frontend*/device/mac NN:NN:NN:NN:NN:NN | sed 's|^.*/dvb\([0-9]*\)\.frontend\([0-9]*\)/.*$|/dev/dvb/adapter\1:\2:\2:\2|')
tsp -I dvb -d "$DEV" ...
~~~
Please test it and adapt it if necessary. I just typed the command here without test, it may contain typos.

morpig, 2022-03-15:
Yeah it's too complicated rather than grabbing adapterN/device name.

In short, I was told that DVB adapters (who have MAC address attached to it) will act as a network adapter:

```
/dev/dvb/adapter0# ls
demux0  dvr0  frontend0  frontend1  net0
```

you will need to open net0, and grab the MAC address similar to a network device.

if anyone is interested on how to grab DVB mac address: https://gitlab.com/berdyansk/astra-sm/blob/ea54f582bc0b5e15708e0685b91050d40803ffc5/modules/dvb/dvbls.c

going to close for now, @lelegard thx for your help!

closed 2022-03-15 by morpig