tsduck/tsduck #957: [FEAT] 'splicemonitor --display-commands' : Adding `packet-index` and `splice-pid` in the logs

moving-digital, 2022-03-15:
Hi @lelegard 

#### Is your feature request related to a problem?
Yes, related to a problem: When using `splicemonitor` plugin with `--display-commands` options, it indeed displays the content of SCTE-35 splice insert commands as expected. But the short event description is still logged into a separate log first, while some of its info are not contained in the "full" log one, as follows:

```
* splicemonitor: { "#name": "event", "event-id": 69, "event-type": "out", "packet-index": 403853, "progress": "immediate", "splice-pid": 256 }
* splicemonitor: { "#name": "splice_information_table", "#nodes": [ { "#name": "splice_insert", "avail_num": 0, "avails_expected": 1, "out_of_network": true, "splice_event_cancel": false, "splice_event_id": 69, "splice_immediate": true, "unique_program_id": 12 } ], "protocol_version": 0, "pts_adjustment": 0, "tier": 0 }

#### Proposed solution:
When `--display-commands` option is enabled, would it be possible to merge the two levels of `splicemonitor` logs under a single one instead, since some information such as `packet-index` and `splice-pid` are not available under the that option?

#### Additional information:

The full command I'm using is:

`tsp --realtime=on --bitrate 1000000 --add-input-stuffing 1/10 --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7 -I ip 225.1.1.1:11111 -l 10.168.1.153 --buffer-size 5000000 -P zap 1 -P history --ignore-stream-id-change -P splicemonitor --display-commands --json-line -O drop`

moving-digital, 2022-03-16:
Hi @lelegard 

I've just tested the 3.31-2711. This is perfect, the SCTE-35 PID is now present in the metadata node. However, the packet index is not. Could you check on your end?

lelegard, 2022-03-16:
> However, the packet index is not. Could you check on your end?

It is when the option `--packet-index` is specified.

See this sample output in the non-regression test suite: https://github.com/tsduck/tsduck-test/blob/master/reference/test-110.6.log
~~~
* splicemonitor: JSON:{ "#name": "splice_information_table", "#nodes": [ { "#name": "metadata", "first_ts_packet": 2000, "last_ts_packet": 2000, "pid": 1280 }, { "#name": "splice_null" } ], "protocol_version": 0, "pts_adjustment": 0, "tier": 4095 }
~~~
Generally speaking, anything that is in the non-regression test suite is guaranteed to be tested after each commit (or more precisely after each push of a bunch of commits).

moving-digital, 2022-03-16:
Thanks a lot, @lelegard !

closed 2022-03-16 by moving-digital