tsduck/tsduck #958: [FEAT] 'splicemonitor --display-commands' to log PIDs in decimal instead of hexa

moving-digital, 2022-03-15:
Hi @lelegard 

#### Is your feature request related to a problem?
We can consider it is. Related to the feature enhancement required here https://github.com/tsduck/tsduck/issues/957 , I actually found out some discrepancies between `splicemonitor --display-commands` and `splicemonitor --all-commands` :

1. `splicemonitor --all-commands` works as expected, meaning: 
- All logs from the short event description are available in the 'long' event description
- All PIDs are displayed in decimal

2. `splicemonitor --display-commands` does not entirely work as expected, or at least, does not have the same behavior than `--all-commands` (But I guess it should) : 
- Two logs from the short event description are not available in the 'long' event description: `packet-index` and `splice-pid`
- All PIDs are displayed in hexadecimal

#### Proposed solution:
To make `splicemonitor --display-commands` behave exactly like `splicemonitor --all-commands`:
- All logs from the short event description should be available in the 'long' event description
- All PIDs shoud be displayed in decimal

#### Alternatives:
No altermatives.

#### Additional information:

**For --all-commands:
Command:**
`tsp --realtime=on --bitrate 1000000 --add-input-stuffing 1/10 --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7 -I ip 225.1.1.1:11111 -l 10.168.1.153 --buffer-size 5000000 -P zap 1 -P history --ignore-stream-id-change -P splicemonitor --all-commands --json-line -O drop`
**Result:**
```
* splicemonitor: { "#name": "event", "event-id": 67, "event-type": "out", "packet-index": 169028, "progress": "immediate", "splice-pid": 256 }
* splicemonitor: { "#name": "splice_information_table", "#nodes": [ { "#name": "splice_insert", "avail_num": 0, "avails_expected": 1, "out_of_network": true, "splice_event_cancel": false, "splice_event_id": 67, "splice_immediate": true, "unique_program_id": 12 } ], "protocol_version": 0, "pts_adjustment": 0, "tier": 0 }
```

**For --display-commands:
Command:**
`tsp --realtime=on --bitrate 1000000 --add-input-stuffing 1/10 --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7 -I ip 225.1.1.1:11111 -l 10.168.1.153 --buffer-size 5000000 -P zap 1 -P history --ignore-stream-id-change -P splicemonitor --display-commands --json-line -O drop`
**Result:**
```
* splicemonitor: { "#name": "event", "event-id": 67, "event-type": "out", "packet-index": 122026, "progress": "immediate", "splice-pid": 256 }
* splicemonitor: { "#name": "splice_information_table", "#nodes": [ { "#name": "splice_insert", "avail_num": "0", "avails_expected": "1", "out_of_network": "true", "splice_event_cancel": "false", "splice_event_id": "0x00000043", "splice_immediate": "true", "unique_program_id": "0x000C" } ], "protocol_version": "0", "pts_adjustment": "0", "tier": "0x0000" }
```

moving-digital, 2022-03-15:
Hi @lelegard 

Thanks for the swift response! I'm using 3.30-2709.

I will try with the latest commit and keep you posted.

lelegard, 2022-03-15:
> I will try with the latest commit and keep you posted.

This is more than the latest commit. The official release 3.30 has been released. Adding the `--select-commands` option was the last update in the 3.30 release cycle. Next commit will be a 3.31 pre-release.

moving-digital, 2022-03-16:
Hi @lelegard 

Everything is now working as expected with 3.31-2711, thanks a lot ! 

You are now even able to predict a requested fix before the issue is even raised 👍  Thank you so much once again.

closed 2022-03-16 by moving-digital