tsduck/tsduck #959: [BUG] merge plugin: PID conflict message for EIT PID

basicmaster, 2022-03-16:
#### Bug description:
When the merge plugin is used and PSI/SI merging is enabled (which is the default), a PID conflict error message may appear for the EIT PID 0x12.

#### How to reproduce:
1. Get two different multiplex dumps (e.g. of similar bitrate) that both contain an EIT.
2. Invoke `./tsp -a 1/1 -I file ts1.ts -P merge "./tsp -I file ts2.ts" -O drop`

#### Expected behavior:
I would expect to not get a PID conflict error be shown, as the PSI merger makes sure that both EITs are merged into the output stream of this plugin (e.g. for the PAT PID 0x0, I don't get a message as well).

#### Errors and logs:
~~~
* Error: merge: PID conflict: PID 0x0012 (18) exists in the two streams, dropping from merged stream
~~~

#### Environment:
 - OS: Linux
 - OS version: Ubuntu 18.04
 - TSDuck full version: 3.31-2711
 - Installation type: built from Git

#### Additional information:
PS 1: The documentation of the merge plugin (especially introduction and `--no-psi-merge` param) only refers to PAT/CAT/SDT, but never mentions the EIT.

PS 2: To prevent the need to do EIT merging, I actually expected the `--drop` option to work for any possible PID, as the param documentation adds no further constraints. So to just ignore the complete EIT of the merged stream (while using the PSI/SI merger for PAT/CAT/SDT), I intended to use the option `-d 0x12`. This doesn't work at the moment, as the PIDs related to the PSI/SI merger are [added to the PID whitelist](https://github.com/tsduck/tsduck/blob/5b9e1ba59acb0add119fa4e81afe9ff8e68f518a/src/tsplugins/tsplugin_merge.cpp#L355) after [the `--drop` and `--pass` params have been processed](https://github.com/tsduck/tsduck/blob/5b9e1ba59acb0add119fa4e81afe9ff8e68f518a/src/tsplugins/tsplugin_merge.cpp#L347). I propose to swap both blocks in order to fix this.

lelegard, 2022-03-17:
Hi @basicmaster 

Thanks for reporting this. There are issues indeed but not the one you suspect. The symptoms are misleading.

The EIT sections are actually merged. The user's guide and help text omit to specify this and I fixed them.

The PID conflict error message was also incorrect and is now removed.

The real problem was that the resulting EIT PID had missing packets. The EIT sections were actually merged and, as long as the missing packets did not invalidated EIT sections, we could find EIT sections from both streams in the resulting EIT PID.

If you are interested in the details, the PAT, SDT and CAT are merged as follow: The tables in the corresponding PID's in the main TS are modified to merge services from both TS. The corresponding PID's in the merged TS are nullified. Thus, the resulting bitrates of PAT, SDT and CAT are unchanged but they reference all services from both streams. In the EIT PID, the EIT sections from both TS are interspersed without being modified (each section references one single service). But it is not possible to keep the EIT bitrate of the main TS, there would not be enough space. The output EIT PID bitrate is the sum of the two EIT bitrates. In practice, we reuse the space of all EIT packets from the two TS to rebuild a new larger EIT PID. Now here comes the bug: the space of EIT packets originally coming from the merged TS were nullified after the section merge. This means that the output EIT PID could contain EIT sections from both streams but packets were missing. This is now fixed.

Please note a potential unrelated problem which is due to your command: `./tsp -a 1/1 -I file ts1.ts -P merge "./tsp -I file ts2.ts" -O drop`. The command is valid but has a race condition which may result in simply merging nothing. The `merge` plugin creates a process and merges packets as they are received. If not perfect, this works relatively well with real-time streams. However, with unregulated streams, this cannot work. The main stream is read at the speed of the input file, which can be very fast. In parallel, the other `tsp` process starts and then outputs packets. However, in the meantime, the processing of the main stream may have already completed, with nothing merged.

Please read the `tsp` reference section in the user's guide which explains the consequences of offline (unregulated) TS vs. real-time (regulated) TS.

If you want to merge files, you must regulate the processing on both sides to let the merging process occur, something like:
~~~
./tsp -a 1/1 -I file ts1.ts -P regulate -P merge "./tsp -I file ts2.ts -P regulate " -O ...
~~~
The unfortunate consequence is that the processing is slowed down at playout speed. But it works.

basicmaster, 2022-03-17:
Thank you for your explanations and for the fixes. Merging the two TSs now works without issues.

I actually also haven't been able to properly parse the EIT of the resulting TS (just forgot to mention it); your details explain why this was the case. It is always interesting to see which implications tiny bugs can have.

Also thanks for the hint regarding the race condition. Initially I used tsp with DVB devices and got the mentioned PID conflict message; later I tried with DVB recordings to be able to reproduce the issue. As there are two independent tsp processes, there is indeed the need to synchronize them (kind of). For testing, this is absolutely fine.

Regarding my PS 2, I noticed that the documentation introduction of the plugin (listing the steps done by the plugin) already explains why `-d 0x12` won't have an effect (so I will use the `filter` plugin when creating the merged TS, if needed).

closed 2022-03-17 by basicmaster