tsduck/tsduck #961: [Q] DVB-T2 support for dektec modulators

macczy, 2022-03-17:
Hi,
I have a dektec DTU-315 modulator, which I wanted to use with ts duck to stream DVB-T2 streams. At first started with simple command:

~~~
./tsp -v -I file /path/to/test_stream.ts --infinite -O dektec --modulation DVB-T2 --frequency 642000000
* file: initial input bitrate is 24,128,572 b/s
* dektec: using Dektec device 0, output channel 0 (DTU-315 port 1)
* dektec: setting output TS bitrate to 39,160,146 b/s
* dektec: setting output carrier frequency to 642,000,000 Hz
* dektec: output fifo size: 16,777,216 bytes, max: 16,777,216 bytes, typical: 16,777,216 bytes
* dektec: initial output bitrate: 39,160,146 b/s
* dektec: Will preload FIFO before starting transmission. Preload FIFO size: 13,421,696 bytes.
* dektec: DTU-315 output FIFO load is 13,421,696 bytes, starting transmission
~~~

Then on TV you can tune to, for example, ITV channel.
Unfortunately, the output is not as expected. The video is sped up and, well let's say it's constantly jumping. Please take a look at the attached recording, as it should be clear from it what I mean: 
https://drive.google.com/file/d/1IUh2121fSmAJQhBWKLAddqxR66riA7IK/view?usp=sharing

The stream I've used is a recording from UK. The file was quite big so I've cut a small piece of it. It's enough to reproduce, so I'm also attaching this file: https://drive.google.com/file/d/1BKrarpZYXmeQ9-lgqJ8czVNY58NTypkf/view?usp=sharing

I've tried changing bitrate manually, but that does not seem to do the job, so I wonder if you have any suggestions how we could stream it properly.

For reference - the same stream on same device works properly when using DekTec's StreamXpress instead of ts duck, so I'm positive the problem does not lay in the stream itself, but rather in how we try to use TSduck or some bug in the implementation.

#### Environment:
 - OS: Linux
 - OS version: Ubuntu 20.04
 - TSDuck full version: 3.31-2711
 - Installation type: building from source

lelegard, 2022-03-17:
Hello @macczy 

We can see this:
~~~
* file: initial input bitrate is 24,128,572 b/s
...
* dektec: setting output TS bitrate to 39,160,146 b/s
~~~

The input bitrate is 24 Mb/s based on PCR analysis, probably a DVB-T stream. Based on the default DVB-T2 modulation parameters (you didn't specify any specific value), the computed output bitrate is 39 Mb/s. `tsp` is a transport stream processor, not a specialized streamer such as StreamXpress. So, all packets go straight through all plugins and the modulator is starving most of the time.

I would recommend two things:
- Use `tsp --add-input-stuffing 6/10`. On input, 6 null packets are added every 10 packets, bumping the bitrate to 38.6 Mb/s.
- Use `-O dektec --stuffing` to make sure that the Dektec modulator adds the exact amount of missing stuffing.

macczy, 2022-03-18:
Thank for quick answer.
I tried using stuffing alone earlier but this by itself wouldn't work. But with `--add-input-stuffing` it does the trick.

There comes another question then - is it possibly to automatically detect the value needed for this flag? Or I need to manually calculate it before starting each stream?

lelegard, 2022-03-18:
>  is it possibly to automatically detect the value needed for this flag?

Based on which criteria? I assume you mean something like "up to that bitrate". The problem is that `tsp` is a very generic tool. The input can be anything, CBR, VBR, sporadic, whatever. The concept of bitrate itself is still to be defined in that generic context. Based on wall-clock time or offline files? Based on PCR? Are they reliable, corrupted, cyclic? Etc.

Because `tsp` is designed to be a low level generic tool making no assumption on the structure of the manipulated stream, the user has to know exactly the context and which actions and parameters to apply.

In your case, a CBR input with reliable PCR and a hardware-regulated output, the combination of the two options I mentioned is probably the most straightforward.

In case of a more variable or hazardous input, I would recommend to use `--add-input-stuffing` with higher values to maintain a bitrate always higher than expected. Then, use the plugin `reduce` further downstream using the appropriate options depending on the context.

macczy, 2022-03-18:
Thank you for all your help. Setting higher stuffing and using` -P reduce --target-bitrate` works perfectly for me.

closed 2022-03-18 by macczy